margelo/react-native-vision-camera #2363: Still gives a black screen issue, but it's not happening on all devices, specifically the Redmi Note 11 Pro Plus

majidhidile, 2024-01-09:
### What's happening?

I updated my camera to version 3.7.0. It works well on my Pixel 4a, but on my Redmi Note 11 Pro Plus, the camera shows a black screen. I've attached an image below for reference.
[image]
[image]

### Reproduceable Code

```tsx
const device = useCameraDevice(isFront ? 'front' : 'back');
const format = useCameraFormat(device, [
{ videoResolution: 'max' },
{ photoResolution: 'max' },
{ videoResolution: { width: 1280, height: 720 } },
{ photoResolution: { width: 1280, height: 720 } },
]);
<Camera
orientation="portrait"
format={format}
device={device}
isActive={true}
ref={cameraRef}
onError={(e) => {
console.log('eRROR-->', e);
}}
style={[StyleSheet.absoluteFill]}
photo={true}
video={true}
audio={true}
enableZoomGesture={true}
>
```

### Relevant log output

```shell
14:38:07.629  E  [ERROR][PPROC  ] camxopenode.cpp:4173 UpdateBandwidth() cq2+m_currentrequestID is 63,m_previewrequestID=62,requestDelta=1
14:38:07.630  E  [ERROR][PPROC  ] camxopenode.cpp:8304 PrepareRSStatsMetadata() cq3+sensorWidth is 4000,sensorHeight=3000
14:38:07.662  E  [ERROR][PPROC  ] camxopenode.cpp:4173 UpdateBandwidth() cq2+m_currentrequestID is 64,m_previewrequestID=63,requestDelta=1
14:38:07.662  E  [ERROR][PPROC  ] camxopenode.cpp:8304 PrepareRSStatsMetadata() cq3+sensorWidth is 4000,sensorHeight=3000
14:38:07.698  E  [ERROR][PPROC  ] camxopenode.cpp:4173 UpdateBandwidth() cq2+m_currentrequestID is 65,m_previewrequestID=64,requestDelta=1
14:38:07.698  E  [ERROR][PPROC  ] camxopenode.cpp:8304 PrepareRSStatsMetadata() cq3+sensorWidth is 4000,sensorHeight=3000
14:38:07.728  E  [ERROR][PPROC  ] camxopenode.cpp:4173 UpdateBandwidth() cq2+m_currentrequestID is 66,m_previewrequestID=65,requestDelta=1
14:38:07.728  E  [ERROR][PPROC  ] camxopenode.cpp:8304 PrepareRSStatsMetadata() cq3+sensorWidth is 4000,sensorHeight=3000
14:38:07.763  E  [ERROR][PPROC  ] camxopenode.cpp:4173 UpdateBandwidth() cq2+m_currentrequestID is 67,m_previewrequestID=66,requestDelta=1
14:38:07.763  E  [ERROR][PPROC  ] camxopenode.cpp:8304 PrepareRSStatsMetadata() cq3+sensorWidth is 4000,sensorHeight=3000
14:38:07.796  E  [ERROR][PPROC  ] camxopenode.cpp:4173 UpdateBandwidth() cq2+m_currentrequestID is 68,m_previewrequestID=67,requestDelta=1
14:38:07.796  E  [ERROR][PPROC  ] camxopenode.cpp:8304 PrepareRSStatsMetadata() cq3+sensorWidth is 4000,sensorHeight=3000
14:38:07.829  E  [ERROR][PPROC  ] camxopenode.cpp:4173 UpdateBandwidth() cq2+m_currentrequestID is 69,m_previewrequestID=68,requestDelta=1
14:38:07.829  E  [ERROR][PPROC  ] camxopenode.cpp:8304 PrepareRSStatsMetadata() cq3+sensorWidth is 4000,sensorHeight=3000
14:38:07.862  E  [ERROR][PPROC  ] camxopenode.cpp:4173 UpdateBandwidth() cq2+m_currentrequestID is 70,m_previewrequestID=69,requestDelta=1
14:38:07.862  E  [ERROR][PPROC  ] camxopenode.cpp:8304 PrepareRSStatsMetadata() cq3+sensorWidth is 4000,sensorHeight=3000
14:38:07.897  E  [ERROR][PPROC  ] camxopenode.cpp:4173 UpdateBandwidth() cq2+m_currentrequestID is 71,m_previewrequestID=70,requestDelta=1
14:38:07.897  E  [ERROR][PPROC  ] camxopenode.cpp:8304 PrepareRSStatsMetadata() cq3+sensorWidth is 4000,sensorHeight=3000
14:38:07.928  E  [ERROR][PPROC  ] camxopenode.cpp:4173 UpdateBandwidth() cq2+m_currentrequestID is 72,m_previewrequestID=71,requestDelta=1
14:38:07.928  E  [ERROR][PPROC  ] camxopenode.cpp:8304 PrepareRSStatsMetadata() cq3+sensorWidth is 4000,sensorHeight=3000
14:38:07.963  E  [ERROR][PPROC  ] camxopenode.cpp:4173 UpdateBandwidth() cq2+m_currentrequestID is 73,m_previewrequestID=72,requestDelta=1
14:38:07.963  E  [ERROR][PPROC  ] camxopenode.cpp:8304 PrepareRSStatsMetadata() cq3+sensorWidth is 4000,sensorHeight=3000
14:38:07.996  E  [ERROR][PPROC  ] camxopenode.cpp:4173 UpdateBandwidth() cq2+m_currentrequestID is 74,m_previewrequestID=73,requestDelta=1
14:38:07.996  E  [ERROR][PPROC  ] camxopenode.cpp:8304 PrepareRSStatsMetadata() cq3+sensorWidth is 4000,sensorHeight=3000
14:38:08.030  E  [ERROR][PPROC  ] camxopenode.cpp:4173 UpdateBandwidth() cq2+m_currentrequestID is 75,m_previewrequestID=74,requestDelta=1
14:38:08.030  E  [ERROR][PPROC  ] camxopenode.cpp:8304 PrepareRSStatsMetadata() cq3+sensorWidth is 4000,sensorHeight=3000
14:38:08.063  E  [ERROR][PPROC  ] camxopenode.cpp:4173 UpdateBandwidth() cq2+m_currentrequestID is 76,m_previewrequestID=75,requestDelta=1
14:38:08.063  E  [ERROR][PPROC  ] camxopenode.cpp:8304 PrepareRSStatsMetadata() cq3+sensorWidth is 4000,sensorHeight=3000
14:38:08.096  E  [ERROR][PPROC  ] camxopenode.cpp:4173 UpdateBandwidth() cq2+m_currentrequestID is 77,m_previewrequestID=76,requestDelta=1
14:38:08.096  E  [ERROR][PPROC  ] camxopenode.cpp:8304 PrepareRSStatsMetadata() cq3+sensorWidth is 4000,sensorHeight=3000
14:38:08.131  E  [ERROR][PPROC  ] camxopenode.cpp:4173 UpdateBandwidth() cq2+m_currentrequestID is 78,m_previewrequestID=77,requestDelta=1
14:38:08.131  E  [ERROR][PPROC  ] camxopenode.cpp:8304 PrepareRSStatsMetadata() cq3+sensorWidth is 4000,sensorHeight=3000
14:38:08.164  E  [ERROR][PPROC  ] camxopenode.cpp:4173 UpdateBandwidth() cq2+m_currentrequestID is 79,m_previewrequestID=78,requestDelta=1
14:38:08.164  E  [ERROR][PPROC  ] camxopenode.cpp:8304 PrepareRSStatsMetadata() cq3+sensorWidth is 4000,sensorHeight=3000
```

### Camera Device

```json
LOG  {
  "sensorOrientation": "landscape-right",
  "hardwareLevel": "full",
  "maxZoom": 10,
  "minZoom": 1,
  "maxExposure": 24,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": false,
  "minExposure": -24,
  "name": "BACK (0)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
 LOG  {
  "sensorOrientation": "landscape-right",
  "hardwareLevel": "full",
  "maxZoom": 10,
  "minZoom": 1,
  "maxExposure": 24,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": false,
  "minExposure": -24,
  "name": "BACK (0)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Redmi Note 11 Pro Plus

### VisionCamera Version

3.7.0

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

DidacClaria, 2024-01-09:
If you set the exposure of the Camera component to 0 you should see it fine in all devices.

`<Camera ... exposure={0} />`

majidhidile, 2024-01-09:
@DidacClaria  Thanks for your response, this issue is fixed.

mrousavy, 2024-01-11:
Logs are not relevant. You included logs that have nothing to do with VisionCamera. Please include the full `adb logcat` logs from your application.

mrousavy, 2024-01-11:
Also, setting `exposure` to 0 is not a fix - it's a workaround. We need a real fix.

GolliAch, 2024-01-30:
Some of our users also reported this problem on  Poco x4 pro and other devices.

mrousavy, 2024-02-01:
Working on a fix for this issue here: https://github.com/mrousavy/react-native-vision-camera/pull/2494 👀 

Still in development, so don't test yet. Long way to go, this is a really complicated issue (as you can see in the changes/Files of that PR)...

closed 2024-02-06 by mrousavy

mrousavy, 2024-02-06:
Just released [VisionCamera 3.9.0-beta.1](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v3.9.0-beta.1), which includes the [big `PersistentCaptureSession` PR](https://github.com/mrousavy/react-native-vision-camera/pull/2494) that aims to fix all the blackscreen issues.... :eyes::tada: 

You can try that new beta out yourself and let me know if you still encounter blackscreen issues. I'll continue to test on more devices in the meantime, organize the code a bit better, and then I'll tackle the preview stretching, focus, and flash issues!

If that fixes the blackscreen issues for you, please consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks! ❤️