margelo/react-native-vision-camera #2364: Stretched camera image 🐛

MateusAnderle, 2024-01-09:
### What's happening?

I am experiencing an issue with the camera view in my React Native app where the camera image appears horizontally stretched. I am using version 3.7.0 of the react-native-vision-camera library. 

<div align="center" >
[image]
</div>

However, in development, when I make some changes inside the <Camera /> style and save, it updates the application and corrects the stretching issue. However, I can't track down the error.

<div align="center" >
[image]
</div>

### Reproduceable Code

```tsx
<Box flex={1} bgColor="#000" position="relative">
        <TouchableOpacity
          style={{
            backgroundColor: 'rgba(0,0,0,0.5)',
            borderRadius: 40,
            justifyContent: 'center',
            alignItems: 'center',
            height: 50,
            width: 50,
            top: 50,
            left: 20,
            zIndex: 5,
            position: 'absolute',
          }}
          onPress={onBackButton}
        >
          <Icon as={Ionicons} name="arrow-back" size={6} color="white" />
        </TouchableOpacity>
        <VisionCamera
          ref={cameraRef}
          style={{ flex: 1 }}
          device={device}
          format={format}
          isActive
          orientation="portrait"
          photo
        />
        <View style={{ padding: 16, flexDirection: 'row', gap: 16 }}>
          <Button flex={1} onPress={takePicture} disabled={takingPicture}>
            Tirar foto
          </Button>

          <Button flex={1} onPress={handlePickImage} disabled={takingPicture}>
            Escolher foto
          </Button>
        </View>
      </Box>
```

### Relevant log output

```shell
There are no error logs
```

### Camera Device

```json
Variable format:

{
   "autoFocusSystem":"contrast-detection",
   "fieldOfView":77.50570519625971,
   "maxFps":60,
   "maxISO":3000,
   "maxZoom":4,
   "minFps":1,
   "minISO":40,
   "photoHeight":720,
   "photoWidth":1280,
   "pixelFormats":[
      "yuv",
      "native"
   ],
   "supportsDepthCapture":false,
   "supportsPhotoHdr":false,
   "supportsVideoHdr":false,
   "videoHeight":1080,
   "videoStabilizationModes":[
      "off",
      "off",
      "cinematic-extended"
   ],
   "videoWidth":1920
}
```

### Device

Samsung galaxy A10

### VisionCamera Version

3.7.0

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

iqorlobanov, 2024-01-09:
As a workaround, you can set the width and height to x2 and change them to the required values after the camera is initialized (`onInitialized`). You may still have to change the camera position in `onInitialized` too.

Example: 
```typescript
const device = useCameraDevice(isInitialized || isIOS ? props.position : 'front');
``` 

This is a very nasty bug, but if you need to use this version, you can do as I indicated above

gsagergithub, 2024-01-09:
> As a workaround, you can set the width and height to x2 and change them to the required values after the camera is initialized (`onInitialized`). You may still have to change the camera position in `onInitialized` too.
> 
> Example:
> 
> ```ts
> const device = useCameraDevice(isInitialized || isIOS ? props.position : 'front');
> ```
> 
> This is a very nasty bug, but if you need to use this version, you can do as I indicated above

Same issue + sometimes onInitialized callback does not trigger :(

iqorlobanov, 2024-01-09:
> Same issue + sometimes onInitialized callback does not trigger :(

`onInitialized` fires only if `isActive={true}`

iqorlobanov, 2024-01-09:
> Same issue + sometimes onInitialized callback does not trigger :(

Share your latest code here

gsagergithub, 2024-01-09:
> > Same issue + sometimes onInitialized callback does not trigger :(
> 
> Share your latest code here

Well, that explains my problem. I'm trying to set isActive inside onInitialized, that's what makes sense for me. Ty Igor!

MateusAnderle, 2024-01-09:
> ```ts
> ```ts
> const device = useCameraDevice(isInitialized || isIOS ? props.position : 'front');
> ```
> 
> 
>     
>       
>     
> 
>       
>     
> 
>     
>   
> ```

This didn't work for me. The image is still opening stretched.

iqorlobanov, 2024-01-09:
> >```
> 
> > const device = useCameraDevice(isInitialized || isIOS ? props.position : 'front');
> 
> >```
> 
> 
> 
> This didn't work for me. The image is still opening stretched. 

Did you try to change width and height after initialization?

mrousavy, 2024-01-09:
> > Same issue + sometimes onInitialized callback does not trigger :(
> 
> `onInitialized` fires only if `isActive={true}`

Fixed in https://github.com/mrousavy/react-native-vision-camera/commit/a8b85a30533bd5b7ed5b142b8c2892af11ff65c3

MateusAnderle, 2024-01-09:
> > > 
> > 
> > 
> > > const device = useCameraDevice(isInitialized || isIOS ? props.position : 'front');
> > 
> > 
> > > 
> > 
> > 
> > This didn't work for me. The image is still opening stretched.
> 
> Did you try to change width and height after initialization?

Works for me! It's not the ideal fix for the problem, but resolves my problem right now, thank you!

closed 2024-01-09 by MateusAnderle

CodeReset, 2024-01-11:
I'm experiencing strange behavior; it's also stretched horizontally for me. I'm trying to adjust the styles in onInitialized. If I set the width and height like this:

```
const {width, height} = useWindowDimensions();
<Camera
  style={
    isInitialized
      ? {
          width,
          height,
        }
      : StyleSheet.absoluteFill
  }
  // other camera properties
/>
```

It remains stretched, but if, for example, I do it like this:

```
<Camera
  style={
    isInitialized
      ? {
          width: width,
          height: height - 1,
        }
      : StyleSheet.absoluteFill
  }
  // other camera properties
/>
```
Then it looks normal, but obviously, this is not a solution.

MateusAnderle, 2024-01-11:
> I'm experiencing strange behavior; it's also stretched horizontally for me. I'm trying to adjust the styles in onInitialized. If I set the width and height like this:
>
> ```
> const {width, height} = [...]

I'm using flex: 0 and flex: 1. Works fine

iqorlobanov, 2024-01-11:
> I'm experiencing strange behavior; it's also stretched horizontally for me. I'm trying to adjust the styles in onInitialized. If I set the width and height like this:
>
> ```
> const {width, height} = [...]

Try to use a larger value before initialization. Your solution works because you gave a smaller value of height after initialization

mrousavy, 2024-01-11:
Hey!

After 8 hours of debugging, I finally found the culprit! I fixed the preview stretching issue in this PR: https://github.com/mrousavy/react-native-vision-camera/pull/2377

If you appreciate my dedication and free support, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) so I can keep providing fixes and building out new features for my open-source libraries in my free time.

iaheck, 2024-01-11:
hey @mrousavy ! Thank you so much for everything you do and your fast responses. I've tried the version `3.7.1` and I think it still has the issue (at least on my Galaxy S10e). But version `3.6.16` works fine for what I'm doing so I think I'll stick to that for now. Maybe if more people try `3.7.1` and report if it's fixed for them? thank you!

BarbieriMatheus, 2024-01-11:
I tried the new version 3.7.1 with the fix but the issue still

buitrbao222, 2024-01-12:
Still happening on v3.7.1, i'm using RN 0.72.4, Android 12, Redmi Note 5 Pro

r1cs1, 2024-01-12:
Same issue for me. If I set isActive to true only when camera is initialized then it works for the first time I open the camera. Then it has the same issue when I unmount the camera and remount later in the app. Only workaround that worked is to set isActive true->false->true forcing a rerender on the camera preview fixes the stretch.

snips11, 2024-01-12:
Still see the error on 3.7.1, it seems to work when the code scanner is being used but we also use another camera where it is not being used and it doesnt.

mrousavy, 2024-01-13:
That's weird.
I can't reproduce the bug 🤷‍♂️

Can you try to move this line here up two lines: https://github.com/mrousavy/react-native-vision-camera/blob/d7479338926576f8824f76734f59581d740c06bc/package/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt#L23

(so outside of the UiThreadUtil thingy) and see if that fixes anything?

iqorlobanov, 2024-01-13:
> That's weird.
> 
> I can't reproduce the bug 🤷‍♂️

Try to set aspectRatio 4/3 for example and set width and height of camera based on aspect ratio

alexrififi, 2024-01-13:
@mrousavy I tried your [fix with holder](https://github.com/mrousavy/react-native-vision-camera/issues/2364#issuecomment-1890396204), but unfortunately it did not solve the issue

r1cs1, 2024-01-13:
I’ve also experienced that setting the width and height to match camera format ratio (4:3) the preview is zoomed in compared to the photo taken.

mrousavy, 2024-01-15:
Are y'all using a `format`?

r1cs1, 2024-01-15:
> Are y'all using a `format`?

I do.

```typescript
  const format = useCameraFormat(device, [
    { photoAspectRatio: 4 / 3 },
    { photoResolution: { width: 1280, height: 960 } },
    { photoHdr: true }
  ]);
```

snips11, 2024-01-15:
I wasn’t, just setting the back camera. I was noticing the issue more when in portrait but would work in landscape.

Thanks

James Parsons

Founder

[signatureImage]

T: 020 3693 3769

M: 07969363020

W: parsonsgreenmedia.com<https://parsonsgreenmedia.com/>

[A black and white logo  Description automatically generated]<https://www.crowncommercial.gov.uk/agreements/RM1043.8>

mrousavy, 2024-01-15:
Again, landscape (or orientation in general) is not yet supported. It's the top number one pinned issue in the repo.

As for the stretched preview in portrait @r1cs1 - I don't know, I cannot reproduce it. Please create a new issue with adb logcat logs and screenshots, otherwise I can't help

snips11, 2024-01-15:
It wasn’t just limited to orientation as the issue was in portrait, I just noticed it worked in landscape and thought it may help. The main one for me would be turning on code scanner would also make it work.

Thanks

James Parsons

Founder

[signatureImage]

T: 020 3693 3769

M: 07969363020

W: parsonsgreenmedia.com<https://parsonsgreenmedia.com/>

[A black and white logo  Description automatically generated]<https://www.crowncommercial.gov.uk/agreements/RM1043.8>

Menardi, 2024-01-15:
@mrousavy Thanks for your great work on this plugin. I can reproduce this in the example app just by commenting out the video and frameProcessor props in `CameraPage.tsx`, and not changing anything else:

```
                photo={true}
                // video={true}
                audio={hasMicrophonePermission}
                // frameProcessor={frameProcessor}
              />
```

On a Pixel 7a running Android 14, when I open the example app from cold after making these changes, the preview is distorted, similar to the screenshot at the top of this issue.

Here's a short screencap showing the issue. It's probably not as clear from the video as it is in person, but you can see that the iPhone image on the react-native-vision-camera website is skewed and very narrow.

https://github.com/mrousavy/react-native-vision-camera/assets/2932890/00511530-3b44-4e4e-ba23-b3c2f9c08346

These are all the logs shown in `yarn start`:

```
 LOG  Running "VisionCameraExample" with {"rootTag":11}
 LOG  Loading react-native-worklets-core...
 LOG  Worklets loaded successfully
 LOG  Re-rendering Navigator. Camera: granted | Microphone: granted
 LOG  Camera: BACK (0) | Format: (4000x2000 photo / 2560x1280@30 video @ 30fps)
 LOG  Camera initialized!
 LOG  Camera initialized!
```

If you need me to do any specific debugging, or would prefer I open a new issue, do let me know. Thanks!

mrousavy, 2024-01-15:
Thanks, will try to investigate this!

yvessidina, 2024-01-16:
Hi !
@mrousavy, thanks for this great plugin.

I have reproduced "stretched camera image" on Android on version 3.7.1.
I have tested on : 

- SamSung M13 - Android 13
- OnePlus T8  - Android 13
- SamSung S10+ - Android  12
- Pixel 7 - Android 13

"stretched camera image" is just on the screen, and when I take a photo, the output was normal (no stretch on the output photo)

On iOS, all thing is Ok.

this is my instance (I use Back Camera). 

```
<Reanimated.View style={streamViewPhoto}>
        <PanGestureHandler enabled={isFocussed}>
            <Camera
                key={"CameraViewer"}
                enableHighQualityPhotos={true}
                style={streamViewPhoto}
                pixelFormat="native"
                resizeMode="contain"
                ref={camera}
                device={device}
                isActive={isFocussed}
                format={format}
                accessibilityViewIsModal={true}
                photo={true}
                fps={fps}
                focusable={true}
                hdr={hdr}
            />
        </PanGestureHandler>
</Reanimated.View> 
```

Hope it will help you to investigate
best regard

mrousavy, 2024-01-16:
I'd need the `adb logcat` logs to get more insights but I think I already know how to fix it

yvessidina, 2024-01-16:
Hi @mrousavy here is a part of logcat.
Hope it will help you
[logcat_debug.log](https://github.com/mrousavy/react-native-vision-camera/files/13947906/logcat_debug.log)

maksimLobachevskiy, 2024-01-17:
This behavior is also reproduced on Android with the latest 3.8.0 release. But when I switch to front camera then back to Back camera, the image looks good.

Jigar2311, 2024-01-18:
Hello, i am trying to make landscape recording using orientation locker to lock camera view as landscape but the camera view not perfectly looking .

i have attached camera video screenshot here
Can you please look into it.

[image]

mrousavy, 2024-01-18:
Hey - orientation is not yet supported. See https://github.com/mrousavy/react-native-vision-camera/issues/1891

Jigar2311, 2024-01-18:
I was using version 2.16.7, which was functioning well in all orientations. However, with the new version, I am encountering issues. Is there an estimate of how long it will take to resolve this issue and make it work again?

mrousavy, 2024-01-18:
Read the discussion in #1891, that'll answer your questions.

olegmilan, 2024-01-25:
> Same issue for me. If I set isActive to true only when camera is initialized then it works for the first time I open the camera. Then it has the same issue when I unmount the camera and remount [...]

Same here. Also there's another workaround:
until onInitialised callback is called, pass some format==undefined.
And once camera is initialised just pass a proper format and code looks like this(this code also contains some other UI fixes. I tried to avoid changes to native code with patches):

```
    const device = getCameraDevice('back');
    const format = getCameraFormat(device, [
      { photoResolution: { width: 1280, height: 720 } },
      { videoResolution: { width: 1280, height: 720 } },
    ]);

    const aspectRatio = format ? format.photoHeight / format.photoWidth : 3 / 4;
    const paddingVertical = (Utils.device.screenHeight() - Utils.device.screenWidth() / aspectRatio) / 2;

    // on iOS we don't need that flag
    const isCameraInitialized = isIOS || this.isCameraInitialized;

    // PaddingVertical works properly, when add extra View layer(View -> View -> Camera)
    return (
      <View style={{ flex: 1, paddingVertical }}>
        <View style={{ flex: 1 }}>
          <VisionCamera
            ref={(ref) => { this.camera = ref; }}
            photo
            format={isAndroid && !this.isCameraInitialized ? undefined : format}
            resizeMode={'contain'}
            style={{ aspectRatio }}
            isActive={isCameraVisibleProp && format !== undefined}
            device={device}
            onInitialized={() => {
              // we need to call this mechanism only on initial launch for Android
              if (isIOS || this.isCameraInitialized) {
                return;
              }

              // better use requestAnimationFrame, since updating state immediately doesn't always fixes the issue
              requestAnimationFrame(() => {
                this.isCameraInitialized = true;
                // refresh state to show the camera
                this.setState({});
              });
            }}
          />
        </View>
      </View>
    );
```

Menardi, 2024-02-05:
Fyi for anyone else following this, it seems this might be a duplicate of #2142. There's some more discussion there, including some potential workarounds.

mrousavy, 2024-02-05:
yup 👍

alexrififi, 2024-03-07:
I confirm the problem.
Device `Poco c40`, `Android 11`

erenkulaksiz, 2024-03-10:
I think i have a temporary solution to this.

Make a state called `ìsInitialized`:

```typescript
const [isInitialized, setIsInitialized] = useState<boolean>(false);
```

Put this prop to `<Camera />`:

```jsx
<Camera
  onInitialized={() => {
    setIsInitialized(true);
  }}
  {...yourProps}
/>
```

Now, we need to update style prop whenever isInitialized changes.
I did this to fix the issue:
```jsx
<Camera
  style={
    isInitialized
      ? {
          position: "absolute",
          top: 0,
          left: 0,
          right: 0,
          bottom: 0,
        }
      : {
          width: 0,
          height: 0,
        }
   }
   {...yourProps}
/>
```
Spent my 4 hours debugging this so you dont have to. Dont forget, it is temporary solution and this needs to be fixed.

sberkeo, 2024-03-11:
Thank you for your solution, it really solve it temporarily, but when you try to open the camera with fast login on slow devices or in unexpected situations, the preview still stretched. Even if I tried a little with the module codes, I saw that it was solved in the latest version. I recommend you to use the 4.0.0-beta.5 version.

den1a, 2024-03-11:
> Thank you for your solution, it really solve it temporarily, but when you try to open the camera with fast login on slow devices or in unexpected situations, the preview still stretched. Even if I [...]

My camera doesn't open at all on version 4.0.0-beta.5

` ERROR  Camera.onError(session/invalid-output-configuration): Failed to configure the Camera Session because the output/stream configurations are invalid! [session/invalid-output-configuration: Failed to co
nfigure the Camera Session because the output/stream configurations are invalid!]
`

mrousavy, 2024-03-12:
V4 is not meant for the public yet, that's why it's a beta and not even merged to main.

RaresM7373, 2024-04-30:
So is the camera stretching issue on Android fixed in 4.x ? I am using 4.0.1 and still see the same issue? Thanks

SUBTAIN-ALI, 2024-11-12:
I resolve this issue on Android devices using a specific combination of approaches. I'm unsure if it will work for you, but I have confirmed it on two different projects.

for React Native 0.71.0 and also make sure this
"react-native-reanimated": "3.6.1",

```
"react-native": "0.71.0",
"react-native-reanimated": "3.6.1",
"react-native-vision-camera": "3.6.4",
```

Initialize the camera devices
```
 const [selectedCamera, setSelectedCamera] = useState('front');
  const device = useCameraDevice(selectedCamera, {
    physicalDevices: [
      'ultra-wide-angle-camera',
      'wide-angle-camera',
      'telephoto-camera',
    ],
  });
```
set the format of the video or pitcher
  ```
const camera = useRef(null);
  const isFocused = useIsFocused();
  
  const format = useCameraFormat(device, [
    {videoResolution: {width: 1280, height: 720}},
    {fps: 30},
  ]);

```

```
{device && (
        <Camera
          ref={camera}
          style={StyleSheet.absoluteFill}
          device={device}
          isActive={isFocused}
          video={true}
          audio={true}
          photo={false}
          enableZoomGesture={true}
          format={format}
          onError={onError}
        />
      )}
```