margelo/react-native-vision-camera #2365: 🐛 [Reopen] Unable to find a specification for `react-native-worklets-core` depended upon by `VisionCamera`

zjkuang, 2024-01-09:
### What's happening?

`pod install` failed with
```
[!] Unable to find a specification for `react-native-worklets-core` depended upon by `VisionCamera`
```
I saw a previous closed PR https://github.com/mrousavy/react-native-vision-camera/issues/1519 but the solution suggested there (adding the plugin in the babel.config.js) doesn't work. So I am reopening this issue.

### Reproduceable Code

```tsx
pod install
```

### Relevant log output

```shell
[!] Unable to find a specification for `react-native-worklets-core` depended upon by `VisionCamera`
```

### Camera Device

```json
{
  "reason": "Can't do it since it didn't pass `pod install`."
}
```

### Device

N/A

### VisionCamera Version

3.7.0

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-01-09:
Please create a build issue for this. You created a bug report.

closed 2024-01-09 by mrousavy

hmadJutt, 2024-02-02:
same issue

sudo-AndrewSmith, 2024-08-14:
> same issue

me too(

phazei, 2024-09-11:
Same issue.

Some dev's are getting:
```
[Codegen] Done.
Analyzing dependencies
RNFBAnalytics: Using Firebase/AnalyticsWithoutAdIdSupport pod in place of default Firebase/Analytics
[VisionCamera] Thank you for using VisionCamera ❤️
[VisionCamera] If you enjoy using VisionCamera, please consider sponsoring this project: https://github.com/sponsors/mrousavy
[VisionCamera] $VCEnableLocation is not set, enabling CLLocation APIs by default...
[VisionCamera] $VCEnableFrameProcessors is not set, enabling Frame Processors if Worklets is installed...
[VisionCamera] react-native-worklets-core not found - Frame Processors are disabled!
Downloading dependencies
Generating Pods project
```

Others:

```
[Codegen] Done.
Analyzing dependencies
RNFBAnalytics: Using Firebase/AnalyticsWithoutAdIdSupport pod in place of default Firebase/Analytics
[VisionCamera] Thank you for using VisionCamera ❤️
[VisionCamera] If you enjoy using VisionCamera, please consider sponsoring this project: https://github.com/sponsors/mrousavy
[VisionCamera] $VCEnableLocation is not set, enabling CLLocation APIs by default...
[VisionCamera] $VCEnableFrameProcessors is not set, enabling Frame Processors if Worklets is installed...
[VisionCamera] react-native-worklets-core found at ., Frame Processors are enabled!
[!] Unable to find a specification for `react-native-worklets-core` depended upon by `VisionCamera/FrameProcessors`
```

phazei, 2024-09-12:
@mhsfh  I fixed it here:
https://github.com/mrousavy/react-native-vision-camera/pull/3188#issuecomment-2342555851