margelo/react-native-vision-camera #2371: ✨ QR code scanning using ML Kit not accepted on F-Droid

root-io, 2024-01-10:
### What feature or enhancement are you suggesting?

Apps using ML Kit are not accepted on F-Droid.

### What Platforms whould this feature/enhancement affect?

Android

### Alternatives/Workarounds

[ZXing ?](https://github.com/zxing/zxing)

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2024-01-11:
VisionCamera 3.7.0 does not use `com.google.android.gms:play-services-mlkit-barcode-scanning` anymore if you need a QR Code Scanner - it uses `com.google.mlkit:barcode-scanning`.  This is an embedded model, it does not have the Google Play Services installed afaik. That means it should work on F-Droid.

Also, ZXing is just not as fast and efficient as MLKit.

closed 2024-01-11 by mrousavy

root-io, 2024-01-12:
Log from their fdroid build pipeline: 

```shell
INFO: Removing usual suspect 'com.google.mlkit' at node_modules/react-native-vision-camera/android/build.gradle
```

Well it seems they flag the whole `com.google.mlkit:*` without distinction

mrousavy, 2024-01-15:
Gotcha, yea okay I mean in that case you can easily create a Frame Processor Plugin to use ZXing instead. It's quite easy to implement :)