margelo/react-native-vision-camera #2372: 🐛 takePhoto stuck on step 2/3

dmandrade, 2024-01-10:
### What's happening?

In some situations, the action of taking a photo gets stuck in step 2/3 and does not finish.

### Reproduceable Code

```tsx
const [cameraPosition, setCameraPosition] = useState<'front' | 'back'>('back')
const device = useCameraDevice(cameraPosition,{
    physicalDevices: ['wide-angle-camera']
})

<Camera
    ref={cameraRef}
    device={device}
    format={format}
    hdr={enableHdr}
    isActive={isActive}
    onInitialized={onInitialized}
    onError={onError}
    enableZoomGesture={false}
    orientation="portrait"
    photo={true}
/>
```

### Relevant log output

```shell
14:33:55 PDVMundo: take photo 
14:33:55 CameraSession: Photo capture 0/3 - preparing capture request (640x480)...
14:33:55 CameraSession: Photo capture 1/3 - starting capture...
14:33:56 CameraSession: Photo Captured!
14:33:56 CameraSession: Photo capture 2/3 complete - received metadata with timestamp 35238519064000
14:33:56 CameraSession: Photo captured! 640 x 480
```

### Camera Device

```json
{
  "sensorOrientation": "landscape-right",
  "hardwareLevel": "full",
  "maxZoom": 10,
  "minZoom": 1,
  "maxExposure": 24,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": false,
  "minExposure": -24,
  "name": "BACK (0)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Pocophone M5s (Android 13), Samsung Galaxy A03 (Android 12), Galaxy S22 (Android 13)

### VisionCamera Version

3.7.0

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-01-11:
What logs are coming in after that? You didn't include the logs after.

Maybe you are destroying the Camera session before photo capture has finished? Are you navigating away by any chance?

dmandrade, 2024-01-11:
I didn't navigate away, the camera remained open and with preview working.

I took the photo and it simply stopped at step 2 of the capture, after 1 minute it just generated the logs related to "updating camerasession"

The error happens more frequently on weak devices, but I had to repeat the test several times to occur.
It also happens more frequently when goes background and returns before taking a photo

I can do other tests and update the logs here.

stevengoldberg, 2024-01-14:
I've observed the same behavior for awhile on my iPhone 14 Pro. I have no idea exactly what conditions cause it, but I have also noticed that it tends to happen when trying to take a photo after the app returns from the background.

I call `const photo = await camera.current.takePhoto(photoOptions)` and the promise never gets resolved, which results in an infinite spinner in my app. The only workaround I've found is to wrap the `takePhoto` call in a function that rejects after a timeout. The hangup still happens intermittently, but when it throws after the timeout it generally works as expected on the next attempt.

dolakzdenek, 2024-01-16:
For me the issue was that start taking a photo changed `isActive` (through state value) to `false`. It led to the freezing the camera, that you described. When I removed this state changed, it's alright. So check the `isActive` state.

stevengoldberg, 2024-01-16:
> For me the issue was that start taking a photo changed `isActive` (through state value) to `false`. It led to the process freeze. When I removed this state changed, it's alright. So check the `isActive` state.

FWIW I don’t think we’re talking about the same issue. The issue I’m seeing results in the photo being taken correctly the majority of the time, but intermittently timing out.

mrousavy, 2024-01-30:
I could do `abortCaptures()` before closing a session to make sure we get a cancelation notice for those photos. I think that's the reason you don't get the callbacks, it's probably because you disable the camera when taking a photo

mrousavy, 2024-02-01:
Working on a fix for this issue here: https://github.com/mrousavy/react-native-vision-camera/pull/2494 👀 

Still in development, so don't test yet. Long way to go, this is a really complicated issue (as you can see in the changes/Files of that PR)...

stevengoldberg, 2024-02-01:
Thanks @mrousavy for all your hard work! Is the WIP fix only on the android side? I have this issue on iOS — is that unrelated?

mrousavy, 2024-02-01:
@stevengoldberg iOS doesn't have this exact issue, there is no such thing as "step 2/3" on iOS. 

So you'd need to create a new issue with logs from iOS for me to investigate it, but for now I'm focused on Android.

closed 2024-02-06 by mrousavy

mrousavy, 2024-02-06:
Just released [VisionCamera 3.9.0-beta.1](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v3.9.0-beta.1), which includes the [big `PersistentCaptureSession` PR](https://github.com/mrousavy/react-native-vision-camera/pull/2494) that aims to fix all the blackscreen issues.... :eyes::tada: 

You can try that new beta out yourself and let me know if you still encounter blackscreen issues. I'll continue to test on more devices in the meantime, organize the code a bit better, and then I'll tackle the preview stretching, focus, and flash issues!

If that fixes the blackscreen issues for you, please consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks! ❤️