margelo/react-native-vision-camera #2376: 🐛 OnePlus 8T on 3.7.0 configure_streams() Invalid argument(s) for configure_streams()

mgcrea, 2024-01-11:
### What's happening?

I have encountered issues with this device (OnePlus 8T running Android 13) in the past (see. https://github.com/mrousavy/react-native-vision-camera/issues/1947#issuecomment-1852411928) that used to reboot or freeze when trying to activate the Camera.

With the 3.7.0 release the behaviour changed and it might help find the root issue. What is peculiar is that the camera works fine when using the `codeScanner` prop but fails to start (blacked-out) as soon as I remove the prop. So it looks like the default set of options does not work but somehow the code scanner changes the format to something that is working fine.

I've used `adb logcat '*:S *:E ReactNative:V ReactNativeJS:V VisionCameraProxy:V VisionCameraCodeScanner:V'` to generate relevant logs.

### Reproduceable Code

```tsx
const devices = useCameraDevices();
  const device = devices.find(({position}) => position === 'back');
  const format = useCameraFormat(device, [{videoResolution: {width: 1280, height: 720}}]);

      <Camera
        style={StyleSheet.absoluteFill}
        device={device}
        format={format}
        isActive
      />
```

### Relevant log output

```shell
01-08 17:08:12.510  1041  1041 E netd    : dispachCmd:oplusGetRtt
01-08 17:08:12.511  1041  1041 E netd    : handleGetRttCmd: netId = 100, uid = 10383
01-08 17:08:12.513  1041  1041 E netd    : dispachCmd cmd:oplusGetRtt  result:0
01-08 17:08:14.474 13022 13022 E OplusBracketLog: [OplusViewMirrorManager] updateHostViewRootIfNeeded, not support android.view.ViewRootImpl@b57a62d
01-08 17:08:14.912  1041  1041 E TetherController: property_get sys.radio.data.modem_share_flag return 0, 1!
01-08 17:08:15.089  5044  5086 E horae   : 0: back:33656,frame:34634,front:33843, 5, +(859), 1 ,34800,36990,38430,39404,35493
01-08 17:08:15.106 13022 13022 E unknown:ReactInstanceManager: destroyRootView called
01-08 17:08:15.106 13022 13022 E unknown:ReactInstanceManager: destroyRootView called, unmountReactApplication
01-08 17:08:15.480  5349  5349 E VendorNfcService: changeNfcStateByWirelessCharge: feature WIRELESS_CHARGING = true; mIsWirelessCharging: false; mIsReverseChargingSwitchOpen: false; mNfcWorkMode: 0; isOverSeaRegion() = true
01-08 17:08:15.481  5490  6456 E Battery : ChargeProtectionUtils: mCanlendarTmp.isEmpty()!
01-08 17:08:15.527  1041  1041 E netd    : dispachCmd:oplusGetRtt
01-08 17:08:15.527  1041  1041 E netd    : handleGetRttCmd: netId = 100, uid = 10383
01-08 17:08:15.528  1041  1041 E netd    : dispachCmd cmd:oplusGetRtt  result:0
01-08 17:08:15.628  1041  1041 E TetherController: property_get sys.radio.data.modem_share_flag return 0, 1!
01-08 17:08:15.795 13022 14882 I ReactNativeJS: Running "RgisConnect" with {"rootTag":31}
01-08 17:08:16.413 13022 14882 I ReactNativeJS: Loading react-native-worklets-core...
01-08 17:08:16.417 13022 14882 I ReactNativeJS: Worklets loaded successfully
01-08 17:08:16.424 13022 14882 I VisionCameraProxy: Initializing VisionCameraProxy...
01-08 17:08:16.424 13022 14882 I VisionCameraProxy: Creating Worklet Context...
01-08 17:08:16.426 13022 14882 I VisionCameraProxy: Worklet Context created!
01-08 17:08:16.430 13022 14882 D VisionCameraCodeScanner: CodeScannerProcessorPlugin initialized with options: {}
01-08 17:08:18.549  1041  1041 E netd    : dispachCmd:oplusGetRtt
01-08 17:08:18.549  1041  1041 E netd    : handleGetRttCmd: netId = 100, uid = 10383
01-08 17:08:18.551  1041  1041 E netd    : dispachCmd cmd:oplusGetRtt  result:4
01-08 17:08:19.328 13022 14882 W ReactNativeJS: 'Encountered empty layout: {"width":0,"height":0}', '\n    at TestPage (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.rgisconnect&modulesOnly=false&runModule=true:335286:41)\n    at ActionSheetProvider (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.rgisconnect&modulesOnly=false&runModule=true:329751:24)\n    at BarcodeReaderProvider (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.rgisconnect&modulesOnly=false&runModule=true:332074:24)\n    at BackdropProvider (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.rgisconnect&modulesOnly=false&runModule=true:328990:36)\n    at EnsureSingleNavigator (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.rgisconnect&modulesOnly=false&runModule=true:274453:24)\n    at BaseNavigationContainer (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.rgisconnect&modulesOnly=false&runModule=true:274054:28)\n    at ThemeProvider (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.rgisconnect&modulesOnly=false&runModule=true:279524:21)\n    at NavigationContainerInner (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.rgisconnect&modulesOnly=false&runModule=true:279414:26)\n    at NavigationContainerWithStatePersistence (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.rgisconnect&modulesOnly=false&runModule=true:273384:24)\n    at ThemeProvider (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.rgisconnect&modulesOnly=false&runModule=true:268798:38)\n    at ThemeProvider (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.rgisconnect&modulesOnly=false&runModule=true:233225:24)\n    at TranslationProvider (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.rgisconnect&modulesOnly=false&runModule=true:201296:24)\n    at Provider (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.rgisconnect&modulesOnly=false&runModule=true:198410:21)\n    at App\n    at ReactNativeProfiler (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.rgisconnect&modulesOnly=false&runModule=true:156585:36)\n    at RCTView\n    at View (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.rgisconnect&modulesOnly=false&runModule=true:60450:43)\n    at __Sentry.TouchEventBoundary (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.rgisconnect&modulesOnly=false&runModule=true:158357:36)\n    at RootApp\n    at RCTView\n    at View (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.rgisconnect&modulesOnly=false&runModule=true:60450:43)\n    at RCTView\n    at View (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.rgisconnect&modulesOnly=false&runModule=true:60450:43)\n    at AppContainer (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.rgisconnect&modulesOnly=false&runModule=true:60294:36)\n    at RgisConnect(RootComponent) (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.rgisconnect&modulesOnly=false&runModule=true:108956:28)'
01-08 17:08:19.423 13022 14882 W ReactNativeJS: 'Encountered empty layout: {"width":0,"height":0}', '\n    at TestPage (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.rgisconnect&modulesOnly=false&runModule=true:335286:41)\n    at ActionSheetProvider (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.rgisconnect&modulesOnly=false&runModule=true:329751:24)\n    at BarcodeReaderProvider (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.rgisconnect&modulesOnly=false&runModule=true:332074:24)\n    at BackdropProvider (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.rgisconnect&modulesOnly=false&runModule=true:328990:36)\n    at EnsureSingleNavigator (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.rgisconnect&modulesOnly=false&runModule=true:274453:24)\n    at BaseNavigationContainer (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.rgisconnect&modulesOnly=false&runModule=true:274054:28)\n    at ThemeProvider (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.rgisconnect&modulesOnly=false&runModule=true:279524:21)\n    at NavigationContainerInner (http://localhost:8081/index.bundle//&platform=android&dev=true&minify=false&app=com.rgisconnect&modulesOnly=false&runModule=true:279414:26)\n    at NavigationContainerWithStatePersistence (http://local [... cut ...]
### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mgcrea, 2024-01-18:
@mrousavy I did not but maybe the logcat filters are too restrictive?

`adb logcat '*:S *:E ReactNative:V ReactNativeJS:V VisionCameraProxy:V'`

Do you have a set of filters to recommend?

Isn't this part:

```
01-08 17:08:19.510  1195  1195 E CamX    : [ERROR][HAL    ] camxhal3.cpp:1453 configure_streams() Invalid argument(s) for configure_streams()
01-08 17:08:19.510  1764  7820 E Camera3-Device: Camera 0: configureStreamsLocked: Set of requested inputs/outputs not supported by HAL
01-08 17:08:19.510  1764  7820 E CameraDeviceClient: endConfigure: Camera 0: Unsupported set of inputs/outputs provided
01-08 17:08:19.510 13022 14819 E CameraCaptureSession: Session 0: Failed to create capture session; configuration failed
01-08 17:08:19.511 13022 14819 E CreateCaptureSession: Camera #0: Failed to create CameraCaptureSession https://github.com/mrousavy/react-native-vision-camera/issues/2!
01-08 17:08:19.512 13022 14819 E CameraSession: Failed to configure CameraSession! Error: [session/cannot-create-session] Failed to create a Camera Session for Camera #0!, isRunning: false, Config-Diff: Difference(deviceChanged=true, outputsChanged=true, sidePropsChanged=true, isActiveChanged=true)
01-08 17:08:19.512 13022 14819 E CameraSession: com.mrousavy.camera.core.CameraSessionCannotBeConfiguredError: [session/cannot-create-session] Failed to create a Camera Session for Camera #0!
```

with `Failed to create a Camera Session for Camera #0!,` about vision camera creating a new camera session?

mrousavy, 2024-01-18:
Try setting `photo={true}`, see what happens

mrousavy, 2024-01-30:
Hey @mgcrea - did you try that?

mgcrea, 2024-01-30:
@mrousavy unfortunately I don't have the device on hand (it's not mine), but will try to re-test ASAP when I pair with the dev using it!

mrousavy, 2024-01-31:
Okay yea please let me know then. I don't want this to become a stale issue

mgcrea, 2024-02-01:
@mrousavy it does indeed work properly when setting `photo={true}`, the camera properly starts and the video feed works.

However when trying to use a custom frame processor along `photo={true}` it fails again. 

Fresh dump:
https://gist.github.com/mgcrea/16c00f57d6e9f5b6651d9659175a7739

mrousavy, 2024-02-01:
Working on a fix for this issue here: https://github.com/mrousavy/react-native-vision-camera/pull/2494 👀 

Still in development, so don't test yet. Long way to go, this is a really complicated issue (as you can see in the changes/Files of that PR)...

mgcrea, 2024-02-02:
`3.9.0-beta-0` still does not work (without `photo={true}`).

We get a systematic black screen on first display, but it ends up working when we toggle the `format` props (via fast refresh) on some value like `const format = useCameraFormat(device, Templates.FrameProcessingYUV);`. But behaviour is the same if you start without specifying format (black screen) -> specifying format / fast refresh (working) or the other way around.

With `photo={true}` it works.

First load (black screen):

```
02-02 11:18:27.230 17982 20099 I ReactNativeJS: -----> isVisible : true
02-02 11:18:27.250 17982 20099 I ReactNativeJS: Loading react-native-worklets-core...
02-02 11:18:27.257 17982 20099 I ReactNativeJS: Worklets loaded successfully
02-02 11:18:27.262 17982 20099 I VisionCameraProxy: Initializing VisionCameraProxy...
02-02 11:18:27.262 17982 20099 I VisionCameraProxy: Creating Worklet Context...
02-02 11:18:27.264 17982 20099 I VisionCameraProxy: Worklet Context created!
02-02 11:18:27.670  4419  5303 E adv_data_processor: adv data modified, but len check fail or data invalid, stay intact
02-02 11:18:28.458  4419  5303 E adv_data_processor: adv data modified, but len check fail or data invalid, stay intact
02-02 11:18:29.157  1290  1376 E netd    : dispachCmd:oplusGetRtt
02-02 11:18:29.157  1290  1376 E netd    : handleGetRttCmd: netId = 103, uid = 10387
02-02 11:18:29.159  1290  1376 E netd    : dispachCmd cmd:oplusGetRtt  result:6
02-02 11:18:29.785 17982 20099 I ReactNativeJS: permissionStatus=not-determined
02-02 11:18:29.797 17982 20099 I ReactNativeJS: permissionStatus=granted
02-02 11:18:29.816 17982 17982 E unknown:ReactModalHost: Creating new dialog from context: com.rgisconnect.MainActivity@c4aaf76@206221174
02-02 11:18:29.844 17982 17982 E OplusBracketLog: [OplusViewMirrorManager] updateHostViewRootIfNeeded, not support android.view.ViewRootImpl@cdf8765
02-02 11:18:29.858 17982 17982 E IPCThreadState: attemptIncStrongHandle(98): Not supported
02-02 11:18:29.858 17982 17982 E IPCThreadState: attemptIncStrongHandle(92): Not supported
02-02 11:18:29.858 17982 17982 E IPCThreadState: attemptIncStrongHandle(103): Not supported
02-02 11:18:29.861  1987  9099 E cameraserver: CameraServiceExtImpl_sm8250.cpp: 77 isFaceUnlockService, pkg_name:com.rgisconnect, isFaceUnlockService 0
02-02 11:18:29.861  1987  9099 E cameraserver: CameraServiceExtImpl_sm8250.cpp: 77 isFaceUnlockService, pkg_name:com.rgisconnect, isFaceUnlockService 0
02-02 11:18:29.861  1987  9099 E CameraService: CameraService::connect evicting conflicting client for camera ID 0
02-02 11:18:29.865 17982 17982 E SurfaceSyncer: Failed to find sync for id=0
02-02 11:18:29.869 29075 29093 E CamX    : [ERROR][HAL    ] camxhal3.cpp:2224 SetStaticUxthread() set uxThread_path:/proc/29075/task/29093/ux_state = 136
02-02 11:18:29.870 29075 29093 E CamX    : [ERROR][HAL    ] camxhal3.cpp:2224 SetStaticUxthread() set uxThread_path:/proc/29075/task/29093/ux_state = 8
02-02 11:18:29.870 29075 29093 E CamX    : [ERROR][HAL    ] camxhal3.cpp:2224 SetStaticUxthread() set uxThread_path:/proc/29075/task/29093/ux_state = 136
02-02 11:18:29.870 29075 29093 E CHIUSECASE: [ERROR  ] chxextensionmodule.cpp:1173 ExtendClose() CLEAR m_preCropWidth.
02-02 11:18:29.870 29075 29093 E CHIUSECASE: [ERROR  ] STELLA.cpp:115 setForegroundCPUSET() cannot open cpus file for writing: Permission denied
02-02 11:18:29.872  1987  9099 E CameraService: onTorchStatusChangedLocked: cannot get torch status of camera 1: No such file or directory (-2)
02-02 11:18:29.874  1987  9099 E CameraService: onTorchStatusChangedLocked: cannot get torch status of camera 4: No such file or directory (-2)
02-02 11:18:29.876 17982 17982 E unknown:ReactModalHost: Updating existing dialog with context: com.rgisconnect.MainActivity@c4aaf76@206221174
02-02 11:18:29.878 29075 29093 E CamX    : [ERROR][HAL    ] camxhal3.cpp:2224 SetStaticUxthread() set uxThread_path:/proc/29075/task/29093/ux_state = 8
02-02 11:18:29.879  3839 18117 E CameraManagerGlobal: Camera 5 is not available. Ignore physical camera status change
02-02 11:18:29.879 25120 31642 E CameraManagerGlobal: Camera 5 is not available. Ignore physical camera status change
02-02 11:18:29.879 31480 31643 E CameraManagerGlobal: Camera 5 is not available. Ignore physical camera status change
02-02 11:18:29.880 31480 31643 E CameraManagerGlobal: Camera 6 is not available. Ignore physical camera status change
02-02 11:18:29.880 17982 18577 E CameraManagerGlobal: Camera 5 is not available. Ignore physical camera status change
02-02 11:18:29.880 31480 31643 E CameraManagerGlobal: Camera 7 is not available. Ignore physical camera status change
02-02 11:18:29.881 17982 18577 E CameraManagerGlobal: Camera 6 is not available. Ignore physical camera status change
02-02 11:18:29.882 25120 31642 E CameraManagerGlobal: Camera 6 is not available. Ignore physical camera status change
02-02 11:18:29.882 25120 31642 E CameraManagerGlobal: Camera 7 is not available. Ignore physical camera status change
02-02 11:18:29.884 29075 29562 E CamX    : [ERROR][HAL    ] camxhal3.cpp:2224 SetStaticUxthread() set uxThread_path:/proc/29075/task/29562/ux_state = 136
02-02 11:18:29.884 31480 31643 E CameraManagerGlobal: Camera 5 is not available. Ignore physical camera status change
02-02 11:18:29.884 25120 30388 E CameraManagerGlobal: Camera 5 is not available. Ignore physical camera status change
[... cut ...]
02-02 11:18:38.859 29075 29097 E CamX    : [ERROR][STATS_AWB] camxcawbioutil.cpp:1766 RetrieveExtensionTriggerInfo() AI fail, aiSceneID: 0x0, aiSceneConfidence: 0x0 ,reqID: 13
02-02 11:18:38.894 29075 29103 E CamX    : [ERROR][STATS_AWB] camxcawbioutil.cpp:2450 PopulateColorSensorFrontData() NCS ColorSensor front handle is NULL
02-02 11:18:38.895 29075 29103 E CamX    : [ERROR][STATS_AWB] camxcawbioutil.cpp:1766 RetrieveExtensionTriggerInfo() AI fail, aiSceneID: 0x0, aiSceneConfidence: 0x0 ,reqID: 14
```

mrousavy, 2024-02-02:
Okay yeah I think this is one of the outputs.

closed 2024-02-06 by mrousavy

mrousavy, 2024-02-06:
Just released [VisionCamera 3.9.0-beta.1](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v3.9.0-beta.1), which includes the [big `PersistentCaptureSession` PR](https://github.com/mrousavy/react-native-vision-camera/pull/2494) that aims to fix all the blackscreen issues.... :eyes::tada: 

You can try that new beta out yourself and let me know if you still encounter blackscreen issues. I'll continue to test on more devices in the meantime, organize the code a bit better, and then I'll tackle the preview stretching, focus, and flash issues!

If that fixes the blackscreen issues for you, please consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks! ❤️