margelo/react-native-vision-camera #2378: 🐛 [Android][3.7.0]TOO_MANY_OPEN_CAMERAS

xseignard, 2024-01-11:
### What's happening?

`TOO_MANY_OPEN_CAMERAS` is still thrown with `3.7.0` on Android

### Reproduceable Code

```tsx
Check this repo https://github.com/xseignard/black-screen
```

### Relevant log output

```shell
2024-01-11T11:51:52.930Z	29542	PreviewView	Creating PreviewView...
	2024-01-11T11:51:52.931Z	29542	PreviewView	Content Size: 1920x1080 (0.5625) | Container Size: 0x0 (NaN)
	2024-01-11T11:51:52.931Z	29542	PreviewView	PreviewView is 0x0, rendering 1920x1080 content. Resizing to: 0x0 (COVER)
	2024-01-11T11:51:52.933Z	29542	CameraView	Updating CameraSession...
	2024-01-11T11:51:52.934Z	29542	CameraSession	configure { ... }: Waiting for lock...
	2024-01-11T11:51:52.936Z	29542	CameraSession	configure { ... }: Updating CameraSession Configuration... Difference(deviceChanged=true, outputsChanged=true, sidePropsChanged=true, isActiveChanged=true)
	2024-01-11T11:51:52.936Z	29542	CameraSession	isActive is false, skipping CameraDevice configuration.
	2024-01-11T11:51:52.936Z	29542	CameraSession	isActive is false, skipping CameraCaptureSession configuration.
	2024-01-11T11:51:52.936Z	29542	CameraSession	Successfully updated CameraSession Configuration! isActive: false
	2024-01-11T11:51:53.080Z	29542	CameraView	Updating CameraSession...
	2024-01-11T11:51:53.081Z	29542	CameraSession	configure { ... }: Waiting for lock...
	2024-01-11T11:51:53.082Z	29542	CameraSession	configure { ... }: Updating CameraSession Configuration... Difference(deviceChanged=true, outputsChanged=true, sidePropsChanged=true, isActiveChanged=true)
	2024-01-11T11:51:53.082Z	29542	CameraSession	Need to rebuild CameraDevice and CameraCaptureSession...
	2024-01-11T11:51:53.083Z	29542	CameraSession	Configuring Camera #1...
	2024-01-11T11:51:53.083Z	29542	CameraManager	Camera #1: Opening...
	2024-01-11T11:51:53.086Z	29542	PreviewView	Content Size: 1920x1080 (0.5625) | Container Size: 1080x1918 (0.5630865484880083)
	2024-01-11T11:51:53.087Z	29542	PreviewView	PreviewView is 1080x1918, rendering 1920x1080 content. Resizing to: 1080x1920 (COVER)
	2024-01-11T11:51:53.088Z	1969	System.err	java.lang.SecurityException: Caller with uid 1047 is not android
	2024-01-11T11:51:53.088Z	1969	System.err		at com.android.server.devicepolicy.DevicePolicyManagerService.getCallerIdentity(DevicePolicyManagerService.java:2368)
	2024-01-11T11:51:53.088Z	1969	System.err		at com.android.server.devicepolicy.DevicePolicyManagerService.getCameraDisabled(DevicePolicyManagerService.java:9311)
	2024-01-11T11:51:53.088Z	1969	System.err		at android.app.admin.DevicePolicyManager.getCameraDisabled(DevicePolicyManager.java:8423)
	2024-01-11T11:51:53.088Z	1969	System.err		at com.android.server.camera.CameraServiceProxy$2.isCameraDisabled(CameraServiceProxy.java:622)
	2024-01-11T11:51:53.088Z	1969	System.err		at android.hardware.ICameraServiceProxy$Stub.onTransact(ICameraServiceProxy.java:174)
	2024-01-11T11:51:53.088Z	1969	System.err		at android.os.Binder.execTransactInternal(Binder.java:1344)
	2024-01-11T11:51:53.088Z	1969	System.err		at android.os.Binder.execTransact(Binder.java:1275)
	2024-01-11T11:51:53.088Z	1645	CameraService	CameraService::connect call (PID 29542 "com.xseignard.blackscreen", camera ID 1) and Camera API version 2
	2024-01-11T11:51:53.089Z	1645	Camera2ClientBase	Camera 1: Opened. Client: com.xseignard.blackscreen (PID 29542, UID 10618)
	2024-01-11T11:51:53.089Z	1645	CameraDeviceClient	CameraDeviceClient 1: Opened
	2024-01-11T11:51:53.089Z	1645	CameraService	makeClient: Camera2 API, override to portrait 0
	2024-01-11T11:51:53.089Z	1304	GCH_CameraDeviceHwlImpl	CreateCameraDeviceSessionHwl CAMERA_LAUNCH 1
	2024-01-11T11:51:53.090Z	1304	GCH_HwlResultDispatcher	ResultDispatcherHwl: SetRealtimeThread OK
	2024-01-11T11:51:53.090Z	1304	GCH_CameraVendorModule	ValidateConcurrentOpen: All cameras should be opened before any session is configured.
	2024-01-11T11:51:53.090Z	1304	GCH_CameraDeviceSessionHwlImpl	DestroyPipelinesLocked Destroying pipelines
	2024-01-11T11:51:53.090Z	1304	GCH_HwlResultDispatcher	ResultDispatcherThread Result dispatcher thread is existing . . .
	2024-01-11T11:51:53.090Z	1304	GCH_CameraDevice	CreateCameraDeviceSession: Creating a CameraDeviceSessionHwl failed: Too many users(-87)
	2024-01-11T11:51:53.090Z	1304	GCH_AidlCameraDevice	open: Creating CameraDeviceSession failed: Too many users(-87)
	2024-01-11T11:51:53.090Z	1645	CameraProviderManager	openAidlSession: Transaction error opening a session for camera device 1: 
	2024-01-11T11:51:53.090Z	1645	Camera3-Device	Camera 1: initialize: Could not open camera session: Too many users (-87)
	2024-01-11T11:51:53.090Z	1645	Camera2ClientBase	initializeImpl: Camera 1: unable to initialize device: Too many users (-87)
	2024-01-11T11:51:53.090Z	1645	CameraService	connectHelper: Could not initialize client from HAL.
	2024-01-11T11:51:53.090Z	1645	Camera2ClientBase	Camera 1: start to disconnect
	2024-01-11T11:51:53.090Z	1645	Camera2ClientBase	Camera 1: serializationLock acquired
	2024-01-11T11:51:53.090Z	1645	Camera2ClientBase	Camera 1: Shutting down
	2024-01-11T11:51:53.090Z	1645	Camera2ClientBase	Camera 1: start to cacheDump
	2024-01-11T11:51:53.099Z	1645	Camera3-Device	disconnectImpl: E
	2024-01-11T11:51:53.099Z	1645	CameraService	disconnect: Disconnected client for camera 1 for PID 29542
	2024-01-11T11:51:53.099Z	1645	cameraserver	FGS Logger Transaction failed
	2024-01-11T11:51:53.099Z	1645	cameraserver	-129
	2024-01-11T11:51:53.099Z	1645	Camera2ClientBase	~Camera2ClientBase: Client object's dtor for Camera Id 1 completed. Client was: com.xseignard.blackscreen (PID 29542, UID 10618)
	2024-01-11T11:51:53.099Z	1645	Camera3-Device	disconnectImpl: E
	2024-01-11T11:51:53.099Z	3182	HbmSvManager	 Request state 0 req 1
	2024-01-11T11:51:53.100Z	29542	CameraManager	Camera #1: Error! 2
	2024-01-11T11:51:53.102Z	29542	CameraSession	
	2024-01-11T11:51:53.102Z	29542	CameraView	invokeOnError(...):
	2024-01-11T11:51:53.102Z	29542	System.err	com.mrousavy.camera.core.CameraCannotBeOpenedError: [session/camera-cannot-be-opened] The given Camera device (id: 1) could not be opened! Error: TOO_MANY_OPEN_CAMERAS
	2024-01-11T11:51:53.102Z	29542	System.err		at com.mrousavy.camera.extensions.CameraManager_openCameraKt$openCamera$2$callback$1.onError(CameraManager+openCamera.kt:47)
	2024-01-11T11:51:53.102Z	29542	System.err		at android.hardware.camera2.impl.CameraDeviceImpl$8.run(CameraDeviceImpl.java:385)
[... 40 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Pixel 5 (Android 14)

### VisionCamera Version

3.7.0

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-01-11:
Hm, I wonder what's going on here.

The ideal behaviour should be that a new Camera that has `isActive` is `true` will take precedence over older Cameras. We need to close older Cameras before opening a new one.

It is normally recommended to just stop the repeating request (`captureSession?.stopRepeating()`) so that the Camera is available again faster when you open it again - but if you don't open it again, Android OS will close the Camera for you (`onDisconnected`).

I am wondering now if we need to close the device manually ourselves when `isActive` is set to `false`? 

Camera2 is so weirdly designed...

xseignard, 2024-01-12:
The only thing that fixes this issue for me is this https://github.com/mrousavy/react-native-vision-camera/pull/2157

I don't say it is the right way to fix the issue, but at least I works 🤷

mrousavy, 2024-01-13:
Okay thanks - that helps. I have a different idea that should ideally be much more stable, but I need to figure out how to deliver that - quite complex to handle all the threading etc here

mrousavy, 2024-02-01:
Working on a fix for this issue here: https://github.com/mrousavy/react-native-vision-camera/pull/2494 👀 

Still in development, so don't test yet. Long way to go, this is a really complicated issue (as you can see in the changes/Files of that PR)...

mrousavy, 2024-02-06:
Just released [VisionCamera 3.9.0-beta.1](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v3.9.0-beta.1), which includes the [big `PersistentCaptureSession` PR](https://github.com/mrousavy/react-native-vision-camera/pull/2494) that aims to fix all the blackscreen issues.... :eyes::tada: 

You can try that new beta out yourself and let me know if you still encounter blackscreen issues. I'll continue to test on more devices in the meantime, organize the code a bit better, and then I'll tackle the preview stretching, focus, and flash issues!

If that fixes the blackscreen issues for you, please consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks! ❤️

xseignard, 2024-02-09:
Hello @mrousavy 
I just tried on my repro project: https://github.com/xseignard/black-screen and the same exact issue is still occurring 🤷 

> If that fixes the blackscreen issues for you, please consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks! ❤️

I actually already sponsor you through my company (https://github.com/alan-eu)

xseignard, 2024-02-12:
Yet another attempt to fix this in this PR https://github.com/mrousavy/react-native-vision-camera/pull/2539

mrousavy, 2024-02-13:
Let's re-open if the issue still exists!

I'll try to take another look at this, but the PR closes the session which goes against the Closeable paradigm.

Kompasua, 2024-02-16:
First of all many thanks Marc for working on this issue! I tried many recommendations from other threads and latest library version, but camera doesn't work properly. BUT! The device I have is also my main phone (Redmi Note Pro 10) and I often have a similar (same?) problem with other apps: telegram, snapchat, instagram and even original camera app. I am not encouraging to blame hardware and give up, but wanted to share this information anyways.

mrousavy, 2024-02-16:
hahaha okay well I'm currently in a bar so I don't know if my input is valuable but if not even the stock camera app (or Snapchat, which I consider a good camera app) work there's not a lot I can do 😂

I wonder what the actual issue is tho, can u share logs?

dimansyvorov, 2024-02-29:
@mrousavy 
Good afternoon. Reproduced the error in your "com.mrousavy.camera.example" project ("version": "3.9.0")

Steps to reproduce ([video](https://github.com/mrousavy/react-native-vision-camera/assets/35449356/d1601ae5-d86d-4975-a7b7-acdb329f0aaf)):
1) Grant permissions after the first launch of the application
2) Go to the Camera screen
3) Switch the camera to "front"
4) Open the Qr scanner screen
5) The camera screen turns black

If you do not switch the camera in step 2, then everything works and the error does not reproduce

Xiaomi Redmi 12 screenshots:
[black screen](https://github.com/mrousavy/react-native-vision-camera/assets/35449356/c32d1f10-9f1c-498d-96cd-c1fe1411cd56)
[error](https://github.com/mrousavy/react-native-vision-camera/assets/35449356/63e61271-eda1-46c6-bf00-23f232e5651b)

AndroidStudio screenshots:
[logcat](https://github.com/mrousavy/react-native-vision-camera/assets/35449356/1893c442-5cea-43e5-ac7c-0e7cbcd99b81)

I hope you will solve the problem soon, because now this is a blocker for us, as well as for the rest of the community 🫶

SameerAther, 2024-03-12:
Seems like issue with Xiaomi Phones. I am also facing this error in Xiaomi Redmi Note 10S.

mrousavy, 2024-03-19:
Hey! I just released a new V4 beta ([v4.0.0-beta.7](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0-beta.7)) where I fixed a bunch of issues! 💪🚀 Can you test that and see if that fixes the issue for you? 😅 

```
yarn add react-native-vision-camera@beta
```

You might need to increase the `compileSdk` to the newest Android SDK (34) if you are behind.

xseignard, 2024-03-19:
Hey :wave: 
Thanks for the v4 release, I just tested on:
- Samsung Galaxy A54 
- Pixel 5
- Huawei P30 Lite

for this specific issue only
And v4.0.0-beta.8 does fix this issue :partying_face: 

Thanks for the hard work 🙇

mrousavy, 2024-03-21:
Great to hear!

mrousavy, 2024-03-25:
Great to hear that V4 now works perfectly! If you appreciate my work, please consider [💖 sponsoring me on GitHub 💖](http://github.com/sponsors/mrousavy/) so I can keep maintaining this library, fixing bugs and building new features! :)

mrousavy, 2024-04-22:
Hey - I think this issue has been fixed in [VisionCamera 4.0.0](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0). 🥳 

Please try V4 and let me know if you still experience this issue;
- if not, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support the development of VisionCamera and thank me for my time spent on fixing bugs and building new features.
- if you still see this issue, please comment and we can re-open this. But please update your native logs with the native (Xcode/Android Studio) logs from running VisionCamera V4 so I can investigate this.

closed 2024-04-22 by mrousavy