margelo/react-native-vision-camera #2379: 🐛 Android crash: SoundPool.finalize timed out

spsaucier, 2024-01-11:
### What's happening?

In my app the user switches from one camera screen to another, between taking photos and scanning QR codes. In this case, the user goes from one take-photo screen to another and crashes.

### Reproduceable Code

```tsx
I haven't been able to reproduce this locally, but I wanted to make sure it was mentioned in case others are also experiencing it, since it is new in version `3.7.0`.
```

### Relevant log output

```shell
java.util.concurrent.TimeoutException: android.media.SoundPool.finalize() timed out after 10 seconds at android.media.SoundPool.native_release(Native Method) at android.media.SoundPool.release(SoundPool.java:177) at android.media.SoundPool.finalize(SoundPool.java:182) at java.lang.Daemons$FinalizerDaemon.doFinalize(Daemons.java:289) at java.lang.Daemons$FinalizerDaemon.runInternal(Daemons.java:276) at java.lang.Daemons$Daemon.run(Daemons.java:137) at java.lang.Thread.run(Thread.java:919)
```
```

### Camera Device

```json
Unknown, unfortunately.
```

### Device

Motorola Moto Z4 (Android 10)

### VisionCamera Version

3.7.0

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-01-15:
This is really weird. Unfortunately there is not enough detail for me. You didn't put the Camera Device here (would be maybe good to also log that in release builds), you didn't put the full logs here (the stacktrace alone is not enough, the issue template tells you to put the full logs here), and I have no repro nor code snippet from your codebase.

I only have one guess, maybe this fixes it: https://github.com/mrousavy/react-native-vision-camera/pull/2390

closed 2024-01-16 by mrousavy

spsaucier, 2024-01-16:
Thanks for the response & PR! I'll update here if it remains after the patch.

PR #2390 (2024-01-16): fix: Release `MediaActionSound` after playing

## What

Pre-loads and releases the `MediaActionSound` if `enableShutterSound` is true. 

Will make the Shutter sound better timed and release it after being done.

## Changes

## Tested on

## Related issues

- Might fix #2379