margelo/react-native-vision-camera #2380: 🐛 The camera's displayed image is flattened

BarbieriMatheus, 2024-01-11:
### What's happening?

I wanted to take a photo. But when I open the camera, the **displayed image is flattened on android**.
I created a **project from scratch and the problem persists**. I tried on two different android devices. I don't tried on IOS.

**When I open the camera in the app, the image becomes flattened. If I do a hot reload or open another app with a camera and return to my app, the image returns to normal..**

I took two screenshot from app to show the error. First image is flattened and second image is normal after make a hot reload. _These images were not photos taken, they are just a screenshot of the app._

[image]       |     [image]

In device log the **sensorOrientation** is ** landscape-left**, but **I have it set to portrait orientation**

**dependencies**
`{ "react": "18.2.0", "react-native": "0.72.9", "react-native-vision-camera": "^3.7.0" }`

### Reproduceable Code

```tsx
const device = useCameraDevice('front');

<Camera
  style={StyleSheet.absoluteFill}
  device={device}
  isActive={true}
  orientation="portrait"
/>
```

### Relevant log output

```shell
Android logcat

2024-01-11 13:03:07.756 23316-23316 PreviewView             com.testingvisioncamera              I  Creating PreviewView...
2024-01-11 13:03:07.756 23316-23316 PreviewView             com.testingvisioncamera              I  Content Size: 1920x1080 (0.5625) | Container Size: 0x0 (NaN)
2024-01-11 13:03:07.757 23316-23316 PreviewView             com.testingvisioncamera              I  PreviewView is 0x0, rendering 1920x1080 content. Resizing to: 0x0 (COVER)
2024-01-11 13:03:07.758 23316-23316 CameraView              com.testingvisioncamera              I  Updating CameraSession...
2024-01-11 13:03:07.764 23316-28160 CameraSession           com.testingvisioncamera              I  configure { ... }: Waiting for lock...
2024-01-11 13:03:07.766 23316-28160 CameraSession           com.testingvisioncamera              I  configure { ... }: Updating CameraSession Configuration... Difference(deviceChanged=true, outputsChanged=true, sidePropsChanged=true, isActiveChanged=true)
2024-01-11 13:03:07.766 23316-28160 CameraSession           com.testingvisioncamera              I  isActive is false, skipping CameraDevice configuration.
2024-01-11 13:03:07.766 23316-28160 CameraSession           com.testingvisioncamera              I  isActive is false, skipping CameraCaptureSession configuration.
2024-01-11 13:03:07.766 23316-28160 CameraSession           com.testingvisioncamera              I  Successfully updated CameraSession Configuration! isActive: false
2024-01-11 13:03:07.779 23316-23316 CameraView              com.testingvisioncamera              I  Updating CameraSession...
2024-01-11 13:03:07.779 23316-28160 CameraSession           com.testingvisioncamera              I  configure { ... }: Waiting for lock...
2024-01-11 13:03:07.779 23316-28160 CameraSession           com.testingvisioncamera              I  configure { ... }: Updating CameraSession Configuration... Difference(deviceChanged=true, outputsChanged=true, sidePropsChanged=true, isActiveChanged=true)
2024-01-11 13:03:07.780 23316-28160 CameraSession           com.testingvisioncamera              I  Need to rebuild CameraDevice and CameraCaptureSession...
2024-01-11 13:03:07.780 23316-23316 PreviewView             com.testingvisioncamera              I  Content Size: 1920x1080 (0.5625) | Container Size: 1080x2055 (0.5255474452554745)
2024-01-11 13:03:07.780 23316-23316 PreviewView             com.testingvisioncamera              I  PreviewView is 1080x2055, rendering 1920x1080 content. Resizing to: 1156x2055 (COVER)
2024-01-11 13:03:07.780 23316-28160 CameraSession           com.testingvisioncamera              I  Configuring Camera #1...
2024-01-11 13:03:07.783 23316-28160 CameraManager           com.testingvisioncamera              I  Camera #1: Opening...
2024-01-11 13:03:07.775 23316-23316 sionCamera.main         com.testingvisioncamera              W  type=1400 audit(0.0:25252): avc: denied { read } for name="u:object_r:persist_camera_prop:s0" dev="tmpfs" ino=22306 scontext=u:r:untrusted_app:s0:c133,c257,c512,c768 tcontext=u:object_r:persist_camera_prop:s0 tclass=file permissive=0
2024-01-11 13:03:07.788 23316-28160 libc                    com.testingvisioncamera              E  Access denied finding property "persist.vendor.camera.privapp.list"
2024-01-11 13:03:07.794 23316-23316 CameraSession           com.testingvisioncamera              I  PreviewView Surface created! Surface(name=null)/@0x8724d4a
2024-01-11 13:03:07.794 23316-23316 CameraSession           com.testingvisioncamera              I  Setting Preview Output...
2024-01-11 13:03:07.794 23316-23316 CameraSession           com.testingvisioncamera              I  PreviewView Surface updated! Surface(name=null)/@0x8724d4a 1156 x 2055
2024-01-11 13:03:07.797 23316-28060 CameraDevices           com.testingvisioncamera              I  Camera #1 is now unavailable.
2024-01-11 13:03:07.829 23316-28160 CameraSession           com.testingvisioncamera              I  configure { ... }: Waiting for lock...
2024-01-11 13:03:07.830 23316-28160 CameraManager           com.testingvisioncamera              I  Camera #1: Opened!
2024-01-11 13:03:07.834 23316-28160 CameraSession           com.testingvisioncamera              I  Successfully configured Camera #1!
2024-01-11 13:03:07.834 23316-28160 CameraSession           com.testingvisioncamera              I  Destroying previous outputs...
2024-01-11 13:03:07.836 23316-28160 CameraSession           com.testingvisioncamera              I  Creating outputs for Camera #1...
2024-01-11 13:03:07.837 23316-28160 CreateCaptureSession    com.testingvisioncamera              I  Camera #1: Creating Capture Session #1... (Hardware Level: 3 | Outputs: [])
2024-01-11 13:03:07.838 23316-28160 CreateCaptureSession    com.testingvisioncamera              I  Using new API (>=28)
2024-01-11 13:03:07.843 23316-28160 CreateCaptureSession    com.testingvisioncamera              I  Camera #1: Successfully created CameraCaptureSession #1!
2024-01-11 13:03:07.843 23316-28160 CameraSession           com.testingvisioncamera              I  Successfully configured Session with 0 outputs for Camera #1!
2024-01-11 13:03:07.843 23316-28160 CameraSession           com.testingvisioncamera              I  CameraSession has no repeating outputs (Preview, Video, CodeScanner), skipping CaptureRequest...
2024-01-11 13:03:07.843 23316-28160 CameraSession           com.testingvisioncamera              I  Successfully updated CameraSession Configuration! isActive: true
2024-01-11 13:03:07.843 23316-28160 CameraView              com.testingvisioncamera              I  invokeOnInitialized()
2024-01-11 13:03:07.844 23316-28160 CameraSession           com.testingvisioncamera              I  configure { ... }: Updating CameraSession Configuration... Difference(deviceChanged=false, outputsChanged=true, sidePropsChanged=true, isActiveChanged=false)
[... 35 lines cut ...]
  "hasTorch": false,
  "position": "front",
  "id": "1"
}
```

### Device

Oneplus A5010 (Android 10); Xiaomi Mi 9 Lite (Android 9)

### VisionCamera Version

3.7.0

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

BarbieriMatheus, 2024-01-11:
I found a same [issue](https://github.com/mrousavy/react-native-vision-camera/issues/2364#issue-2072266403). 👀

I tried the new version 3.7.1 with the fix but the issue still.

mrousavy, 2024-01-13:
Can you upgrade to VisionCamera 3.7.1 to see if this PR: https://github.com/mrousavy/react-native-vision-camera/pull/2377 fixed the issue?

stefanoaga, 2024-01-15:
My App has the screen orientation fixed to landscape and i have the same problem (VisionCamera 3.7.1)
[image]

mrousavy, 2024-01-15:
Orientation does not work yet. This is mentioned in the # 1 pinned issue https://github.com/mrousavy/react-native-vision-camera/issues/1891

leppid, 2024-01-15:
Still present on 3.7.1 :(
After the photo is taken, the resulting image is flipped to the left for some reason (Samsung A52, Frontal camera)

adscud, 2024-01-15:
I don't know if it can help to fix the bug but when you switch to the other camera and come back to the previous one, the camera is correct.

`front (flattened) => back => front (perfect)`

BarbieriMatheus, 2024-01-15:
> Can you upgrade to VisionCamera 3.7.1 to see if this PR: #2377 fixed the issue?

Unfortunately it didn't resolve

guilhermeadsferreira, 2024-01-15:
> > Can you upgrade to VisionCamera 3.7.1 to see if this PR: #2377 fixed the issue?
> 
> Unfortunately it didn't resolve

+1

subramani01, 2024-01-18:
`const device = useCameraDevice(cameraType);

const format = useCameraFormat(device, [
{ videoResolution: { width: 1280, height: 720 } },
{ fps: 30 }
])`

I had the same issue, during the first render of the camera, initially the camera is "front", the preview is stretched vertically, when i switch to back camera and then to front, the stretch is gone.

Using the above code, it fixed the stretch issue in the initial render itself.

zowebs, 2024-01-24:
The bug still persists in **3.8.2**, tested on both Android & iOS, happens **only on Android**.
If you change something in the code while having the camera open, the fast refresh temporarily fixes the flattening, however on the next camera activation, the flattening is back. When you take a photo, it is good - with the "correct" resolution.

theabhishek2511, 2024-01-24:
Did a bunch of testing to figure out what's going on here. The solution I ended up on is this:

```jsx
const [isCameraInitialized, setIsCameraInitialized] = useState(false)
const [isActive, setIsActive] = useState(true)
const squishFixRef = useRef(false)

const initCallback = useCallback(() => {
  if (isCameraInitialized && !squishFixRef.current) {
    squishFixRef.current = true
    setTimeout(() => {
      setIsActive(true)
    }, 150)
    setIsActive(false)
  }
  setIsCameraInitialized(true)
}, [isCameraInitialized])

return (
  <View style={$root}>
    <Camera
      // ...props
      device={device}
      isActive={isCameraInitialized && isActive}
      onInitialized={initCallback}
    />
  </View>
)
```

Essentially, once the camera is initialized, I flip `isActive` to `false` for 150 milliseconds and set it back to `true`. On my test devices, `onInitialized` was being called twice (not sure why), but since flipping the camera off and on just once fixes it, I tossed in a ref to ensure it only does so once.

AlexDef, 2024-01-24:
I'm doing the same kind of tricks for Android. Starting with a camera on "front" position then switching to "back" atfer 200ms and the image is good. I'm using the last version, 3.8.2 with React Native in 0.72.4., not using reanimated, the bug is only on android. Good luck for the debug

fobin, 2024-01-25:
Also having this issue on Android. 
[image]

I tried @theabhishek2511 mitigation method but for me even that wasn't working reliably. Usually it doesn't appear when Camera first appears but then when I retake photo it's most likely stretched. For use it then just switches from CameraPreview component back to Camera component.

```jsx 
<Camera
  photo={true}
  style={styles.camera}
  ref={camera}
  device={device}
  isActive={true}
  audio={false}
  resizeMode="cover"
/>
```

"react-native-vision-camera": "^3.8.2"
"react-native": "0.73.2",

Test device Samsung A33

mrousavy, 2024-01-29:
Duplicate of https://github.com/mrousavy/react-native-vision-camera/issues/2142 but let's still keep it open for discussion

BarbieriMatheus, 2024-01-31:
> Did a bunch of testing to figure out what's going on here. The solution I ended up on is this:
> 
> ```js
> const [isCameraInitialized, setIsCameraInitialized] = useState(false)
> const [isActive, setIsActive] = useState(true)
> const squishFixRef = useRef(false)
> 
> const initCallback = useCallback(() => {
>   if (isCameraInitialized && !squishFixRef.current) {
>     squishFixRef.current = true
>     setTimeout(() => {
>       setIsActive(true)
>     }, 150)
>     setIsActive(false)
>   }
>   setIsCameraInitialized(true)
> }, [isCameraInitialized])
> 
> return (
>   <View style={$root}>
>     <Camera
>       // ...props
>       device={device}
>       isActive={isCameraInitialized && isActive}
>       onInitialized={initCallback}
>     />
>   </View>
> )
> ```
> 
> Essentially, once the camera is initialized, I flip `isActive` to `false` for 150 milliseconds and set it back to `true`. On my test devices, `onInitialized` was being called twice (not sure why), but since flipping the camera off and on just once fixes it, I tossed in a ref to ensure it only does so once.

This implementation work in my project. I used the Oneplus A5010 (Android 10);

BarbieriMatheus, 2024-01-31:
> Still present on 3.7.1 :( After the photo is taken, the resulting image is flipped to the left for some reason (Samsung A52, Frontal camera)

I had the same issue. I searched and found others discussions with same problem:  #1401 #146

I use this [workaround](https://github.com/mrousavy/react-native-vision-camera/issues/1401#issuecomment-1383286364). When the width is greater than the height, I rotate the image by 90 degrees with the react-native-image-resizer library and it works

mrousavy, 2024-02-06:
Created a PR to fix this issue once and for all: https://github.com/mrousavy/react-native-vision-camera/pull/2519 💪🚀 
 **Thanks to everyone who sponsored me on GitHub to fix this bug!!** 💖

Try the PR and see if it works (by checking out that branch and running the example app)

closed 2024-02-07 by mrousavy

heinzbusch, 2024-08-29:
VisionCamera 3.9.2, still facing issue

mrousavy, 2024-08-30:
> VisionCamera 3.9.2, still facing issue
There have been countless new versions of VisionCamera, 4.0.0, 4.0.1, 4.1.0, 4.2.0 etc etc we're at 4.5.0 now. This issue is long fixed.

heinzbusch, 2024-08-30:
> > VisionCamera 3.9.2, still facing issue
> > There have been countless new versions of VisionCamera, 4.0.0, 4.0.1, 4.1.0, 4.2.0 etc etc we're at 4.5.0 now. This issue is long fixed.

I'm using version 3.9.2 because the project I'm working on is not compatible with the Java version required by VisionCamera starting from version 4.0.0.