margelo/react-native-vision-camera #2381: 🐛 Camera Preview becomes Black in iOS/Android after 3-4th time initializing the screen

devkawal, 2024-01-12:
### What's happening?

Hi, I am using react-native-vision-camera(3.7.0) latest version.
I have a scenario, in which I am using carousel of two slides in dashboard screen.
One slide is for scanning data through NFC(Default slide).
Second slide is for scanning data through Camera(We have to switch between these two slides using arrow icons).

While initializing the camera tab, sometimes after 3-4th time, camera becomes black. But interesting part is we are able to scan the barcode even in black screen.

Can anyone share any thoughts on this. What's wrong is happening here?

### Reproduceable Code

```tsx
Calling Camera tab
<Carousel> <View style={styles.slide}> {/* NFC Start */} <View </View> {/* NFC END */} </View> <View style={styles.slide}> {device === null ? ( <View> <Text>No Camera</Text> </View> ) : ( <View style={styles.outerScanContainer}> <View style={styles.scanContainer}> {/* Camera Start */} {cameraScreen && device !== null && ( <Camera style={styles.cameraViewStyle} codeScanner={codeScanner} device={device} isActive={isCameraInitialized} onInitialized={onCameraInitialized} /> )} </View> <Image source={IMAGES.camera_focus} style={styles.focusFrame} resizeMode="contain" /> </View> )} {/* Camera End */} </Carousel>

Initializing Function
const onCameraInitialized = useCallback(() => { if (cameraScreen) { setIsCameraInitialized(false); } else { setIsCameraInitialized(true); } }, []);

Handle slide change
const handleSlideChange = () => { if (cameraScreen) { console.log('yes'); setCameraScreen(false); } else { console.log('no'); setCameraScreen(true); } };
```

### Relevant log output

```shell
Sometimes, camera becomes black
```

### Camera Device

```json
<Camera style={styles.cameraViewStyle} codeScanner={codeScanner} device={device} isActive={isCameraInitialized} onInitialized={onCameraInitialized}
```

### Device

Oneplus 11R(Oxygen OS 13.1), iPhone8(iOS - 16.1)

### VisionCamera Version

3.4.1

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-01-13:
3.4.1 is not the latest version.

This has been fixed in 3.7.1, please upgrade. And please search for existing issues before creating a new one in the future

closed 2024-01-13 by mrousavy

devkawal, 2024-01-15:
@mrousavy Thanks for the comment. This issue is resolved in android, but I am still receiving this issue in iOS.

mrousavy, 2024-01-15:
This can't be - on iOS the lifecycle is fully stable as far as I know.

Can you share a screen recording and the full native logs from Xcode please?

devkawal, 2024-01-15:
@mrousavy Please check

`2024-01-15 16:09:29.473031+0530 Scanner[1572:398005] [native] 
2024-01-15 16:09:29.494047+0530 Scanner[1572:397492] [javascript] 'Found screens with the same name nested inside one another. Check:\n\nDashboard, Dashboard > Dashboard\n\nThis can cause confusing behavior during navigation. Consider using unique names for each screen instead.', '\n    at BaseNavigationContainer (http://10.76.0.12:8081/index.bundle//&platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.test.smartlock.testAccess-beta:128132:28)\n    at ThemeProvider (http://10.76.0.12:8081/index.bundle//&platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.test.smartlock.testAccess-beta:133037:21)\n    at NavigationContainerInner (http://10.76.0.12:8081/index.bundle//&platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.test.smartlock.testAccess-beta:132932:26)\n    at StackScreen (http://10.76.0.12:8081/index.bundle//&platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.test.smartlock.testAccess-beta:148199:24)\n    at RCTView\n    at View (http://10.76.0.12:8081/index.bundle//&platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.test.smartlock.testAccess-beta:45623:43)\n    at I18nextProvider (http://10.76.0.12:8081/index.bundle//&platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.test.smartlock.testAccess-beta:126695:20)\n    at ErrorBoundary (http://10.76.0.12:8081/index.bundle//&platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.test.smartlock.testAccess-beta:148014:36)\n    at App (http://10.76.0.12:8081/index.bundle//&platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.test.smartlock.testAccess-beta:123340:41)\n    at RCTView\n    at View (http://10.76.0.12:8081/index.bundle//&platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.test.smartlock.testAccess-beta:45623:43)\n    at RCTView\n    at View (http://10.76.0.12:8081/index.bundle//&platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.test.smartlock.testAccess-beta:45623:43)\n    at AppContainer (http://10.76.0.12:8081/index.bundle//&platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.test.smartlock.testAccess-beta:45504:36)\n    at Scanner(RootComponent) (http://10.76.0.12:8081/index.bundle//&platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.test.smartlock.testAccess-beta:83664:28)'
2024-01-15 16:09:29.542405+0530 Scanner[1572:397170] [native] VisionCamera.didSetProps(_:): Updating 14 props: [onViewReady, codeScannerOptions, cameraId, onError, onStarted, width, onInitialized, alignSelf, enableBufferCompression, height, onCodeScanned, enableFrameProcessor, onStopped, isActive]
2024-01-15 16:09:29.542454+0530 Scanner[1572:397170] [native] VisionCamera.configure(_:): configure { ... }: Waiting for lock...
2024-01-15 16:09:29.548425+0530 Scanner[1572:397659] [native] VisionCamera.configure(_:): configure { ... }: Updating CameraSession Configuration... Difference(inputChanged: true, outputsChanged: true, videoStabilizationChanged: true, orientationChanged: true, formatChanged: true, sidePropsChanged: true, torchChanged: true, zoomChanged: true, exposureChanged: true, audioSessionChanged: true)
2024-01-15 16:09:29.548525+0530 Scanner[1572:397659] [native] VisionCamera.withSessionLock(_:): Beginning CameraSession configuration...
2024-01-15 16:09:29.548560+0530 Scanner[1572:397659] [native] VisionCamera.configureDevice(configuration:): Configuring Input Device...
2024-01-15 16:09:29.548664+0530 Scanner[1572:397659] [native] VisionCamera.configureDevice(configuration:): Configuring Camera com.apple.avfoundation.avcapturedevice.built-in_video:0...
2024-01-15 16:09:29.565713+0530 Scanner[1572:397659] [native] VisionCamera.configureDevice(configuration:): Successfully configured Input Device!
2024-01-15 16:09:29.565749+0530 Scanner[1572:397659] [native] VisionCamera.configureOutputs(configuration:): Configuring Outputs...
2024-01-15 16:09:29.565849+0530 Scanner[1572:397659] [native] VisionCamera.configureOutputs(configuration:): Adding Code Scanner output...
2024-01-15 16:09:29.568934+0530 Scanner[1572:397659] [native] VisionCamera.configureOutputs(configuration:): Successfully configured all outputs!
2024-01-15 16:09:29.570237+0530 Scanner[1572:397659] [native] VisionCamera.withSessionLock(_:): Committed CameraSession configuration!
2024-01-15 16:09:29.570280+0530 Scanner[1572:397659] [native] VisionCamera.withDeviceLock(_:): Beginning CaptureDevice configuration...
2024-01-15 16:09:29.571437+0530 Scanner[1572:397659] [native] VisionCamera.withDeviceLock(_:): Committed CaptureDevice configuration!
2024-01-15 16:09:29.623342+0530 Scanner[1572:397492] [javascript] 'kas:', true
flipper: Desktop failed to provide certificates. Error from flipper desktop:
rsocket::StreamInterruptedException: connection error
2024-01-15 16:09:30.022941+0530 Scanner[1572:397659] [native] VisionCamera.onCameraStarted(): Camera started!
2024-01-15 16:09:30.023254+0530 Scanner[1572:397659] [native] VisionCamera.withDeviceLock(_:): Beginning CaptureDevice configuration...
2024-01-15 16:09:30.023355+0530 Scanner[1572:397659] [native] VisionCamera.withDeviceLock(_:): Committed CaptureDevice configuration!
[... cut ...]
2024-01-15 16:10:05.183727+0530 Scanner[1572:398005] [native] VisionCamera.configure(_:): Beginning AudioSession configuration...
2024-01-15 16:10:05.183775+0530 Scanner[1572:398005] [native] VisionCamera.configureAudioSession(configuration:): Configuring Audio Session...
2024-01-15 16:10:05.184032+0530 Scanner[1572:398005] [native] VisionCamera.configure(_:): Committed AudioSession configuration!`

https://github.com/mrousavy/react-native-vision-camera/assets/70426813/a065102a-f33f-4f2c-b276-3afbec6bb2e7

mrousavy, 2024-01-15:
Use three backticks (```) to format your logs.

mrousavy, 2024-01-15:
Hm, from the logs it looks like the Camera initializes just fine 🤔 

Instead of fully unmounting the `<Camera>`, can you try just switching `isActive` from true to false and vice versa?

devkawal, 2024-01-15:
Thanks, that patch is working, but that gives a flicker issue. Means when I slide to camera and comes back to nfc and again goto camera, then camera preview freezes for 1-2 seconds, then it will start working.

mrousavy, 2024-01-15:
That's just an issue with the native Camera - I don't think there's anything I can do to get around that problem.

devkawal, 2024-01-15:
Found one more issue, if I am trying to use isActive, then camera is being active on nfc screen also even if its value is false. And user can scan the device from nfc screen also by pointing camera towards the barcode.

devkawal, 2024-01-15:
If I am running this code, without carousel, then its working fine. But not working properly inside carousel

devkawal, 2024-01-16:
@mrousavy Can you please share any thoughts on this. 'isActive' is not working as expected in carousel. 
Also, can you open this issue, if someone else can look into this and share any thoughts

mrousavy, 2024-01-16:
When I set `isActive` to `false` on my Camera it stops streaming, so for me it works. I have no reproduction otherwise

devkawal, 2024-01-16:
Are you trying to reproduce inside carousel?

mrousavy, 2024-01-16:
No, because my app does not use a carousel.

What I'm saying is; you need to create a clear reproduction of the issue if you want me to fix the issue.

devkawal, 2024-01-17:
@mrousavy I have created a demo app which having similar kind of issue - https://github.com/devkawal/scanner-demo. In this, we have a carousel of two slides, when we switch from 1st to 2nd slide(camera screen), then sometimes camera got black screen or stuck.