margelo/react-native-vision-camera #2388: 🐛 iPhone throws an error when capturing a photo with flash

stevengoldberg, 2024-01-14:
### What's happening?

When I call `Camera.takePhoto` and pass `{ flash: 'on' }`, the camera throws an error every time. When I pass `{ flash: 'auto' }` the camera throws an error if the surroundings are dim enough that flash is required. Sometimes the flash will briefly activate before the error is thrown.

When I pass `{ flash: 'off' }`, (or `{ flash: 'auto' }` in well-lit conditions) photo capture works as expected.

The error is:

```
[capture/unknown: Error Domain=AVFoundationErrorDomain Code=-11800 "The operation could not be completed" UserInfo={NSUnderlyingError=0x285b17600 {Error Domain=NSOSStatusErrorDomain Code=-12780 "(null)"}, NSLocalizedFailureReason=An unknown error occurred (-12780), AVErrorRecordingFailureDomainKey=4, NSLocalizedDescription=The operation could not be completed}]
```

### Reproduceable Code

```tsx
const photoOptions = {
  qualityPrioritization:
    photoQualitySetingsMap[photoQualitySetting],
  enableAutoStabilization: true,
}
if (device.hasFlash) {
  photoOptions.flash = flashSettingsMap[flashSetting]
}
           
const photo = await camera.current.takePhoto(photoOptions)
```

### Relevant log output

```shell
error	13:25:58.029650-0500	audiomxd	        HALB_IOThread.cpp:44     HALB_IOThread::IsCurrentThread: this thread ought to lock the lock
error	13:25:58.132128-0500	appleh13camerad	New connection: pid <private>
error	13:25:58.132172-0500	appleh13camerad	Total number of connections: 2
error	13:25:58.132231-0500	appleh13camerad	H13ISPServicesAssistant: getProperty 103
error	13:25:58.132256-0500	appleh13camerad	H13ISPServicesAssistant: getProperty 103 complete (res=0x00000000)
error	13:25:58.132684-0500	appleh13camerad	H13ISPServicesAssistant: getProperty 104
error	13:25:58.132709-0500	appleh13camerad	H13ISPServicesAssistant: getProperty 104 complete (res=0x00000000)
error	13:25:58.132989-0500	appleh13camerad	Client disconnecting (pid <private>)
error	13:25:58.133013-0500	appleh13camerad	Active client pid = 35
error	13:25:58.133037-0500	appleh13camerad	Removing client: pid <private>
error	13:25:58.133060-0500	appleh13camerad	Total number of connections: 1
error	13:25:58.154325-0500	mediaserverd	<<<< FigCaptureSession >>>> Fig assert: "err == 0 " at bail (FigCaptureSession.m:5448) - (err=-12780)
error	13:25:58.155181-0500	mediaserverd	<<<< FigCaptureSession >>>> Fig assert: "err == 0 " at bail (FigCaptureSession.m:5448) - (err=-12780)
error	13:25:58.155258-0500	mediaserverd	<<<< FigCaptureSession >>>> Fig assert: "err == 0 " at bail (FigCaptureSession.m:5448) - (err=-12780)
error	13:25:58.155487-0500	mediaserverd	<<<< FigCaptureSession >>>> Fig assert: "err == 0 " at bail (FigCaptureSession.m:5448) - (err=-16800)
error	13:25:58.155564-0500	mediaserverd	<<<< FigCaptureSession >>>> Fig assert: "err == 0 " at bail (FigCaptureSession.m:5448) - (err=-16800)
error	13:25:58.155632-0500	mediaserverd	<<<< FigCaptureSession >>>> Fig assert: "err == 0 " at bail (FigCaptureSession.m:5448) - (err=-16800)
error	13:25:58.155699-0500	mediaserverd	<<<< FigCaptureSession >>>> Fig assert: "err == 0 " at bail (FigCaptureSession.m:5448) - (err=-16800)
error	13:25:58.155769-0500	mediaserverd	<<<< FigCaptureSession >>>> Fig assert: "err == 0 " at bail (FigCaptureSession.m:5448) - (err=-16800)
error	13:25:58.156189-0500	mediaserverd	<<<< FigCaptureSession >>>> Fig assert: "err == 0 " at bail (FigCaptureSession.m:5448) - (err=-16800)
error	13:25:58.161396-0500	appleh13camerad	H13ISPServicesAssistant: setProperty 101
error	13:25:58.161417-0500	appleh13camerad	H10ISPServicesAssistant: setProperty 101 complete (res=0x00000000)
error	13:25:58.171710-0500	Phomo	CGAffineTransformInvert: singular matrix.
error	13:25:58.226547-0500	appleh13camerad	H13ISPServicesAssistant: setProperty 15
error	13:25:58.226571-0500	appleh13camerad	H10ISPServicesAssistant: setProperty 15 complete (res=0x00000000)
error	13:25:58.226686-0500	appleh13camerad	H13ISPServicesAssistant: setProperty 16
error	13:25:58.226710-0500	appleh13camerad	H10ISPServicesAssistant: setProperty 16 complete (res=0x00000000)
error	13:25:58.236066-0500	appleh13camerad	H13ISPServicesAssistant: setProperty 2
error	13:25:58.236195-0500	appleh13camerad	H10ISPServicesAssistant: setProperty 2 complete (res=0x00000000)
error	13:25:58.236735-0500	appleh13camerad	H13ISPServicesAssistant: setProperty 15
error	13:25:58.236763-0500	appleh13camerad	H10ISPServicesAssistant: setProperty 15 complete (res=0x00000000)
error	13:25:58.236892-0500	appleh13camerad	H13ISPServicesAssistant: setProperty 16
error	13:25:58.236917-0500	appleh13camerad	H10ISPServicesAssistant: setProperty 16 complete (res=0x00000000)
error	13:25:58.238040-0500	appleh13camerad	H13ISPServicesAssistant: setProperty 2
error	13:25:58.238069-0500	appleh13camerad	H10ISPServicesAssistant: setProperty 2 complete (res=0x00000000)
error	13:25:58.240909-0500	appleh13camerad	H13ISPServicesAssistant: setProperty 15
error	13:25:58.240931-0500	appleh13camerad	H10ISPServicesAssistant: setProperty 15 complete (res=0x00000000)
error	13:25:58.241065-0500	appleh13camerad	H13ISPServicesAssistant: setProperty 16
error	13:25:58.241090-0500	appleh13camerad	H10ISPServicesAssistant: setProperty 16 complete (res=0x00000000)
error	13:25:58.244498-0500	appleh13camerad	H13ISPServicesAssistant: setProperty 2
[... 93 lines cut ...]
  "supportsRawCapture": false,
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:7",
  "maxZoom": 189
}
```

### Device

iPhone 14 Pro iOS 17.2.1

### VisionCamera Version

3.7.1

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

kolyaVV, 2024-01-29:
I found a reliable relation to failing to capture photos with flash ON. The test device is iPhone Xr 17.2.1

In format, there is `autoFocusSystem` value. It can be `phase-detection` and `contrast-detection`. If I use a format with `contrast-detection` and flash ON it will error. 
If I use a format with `phase-detection` and flash ON it takes a picture.

https://react-native-vision-camera.com/docs/api/#autofocussystem ( definitions are from  taken from [apple docs](https://developer.apple.com/documentation/avfoundation/avcaptureautofocussystem) )

My guess right now is `contrast-detection` is slower to auto-focus by definition so 
1. It might be incompatible with flash ON or 
2. Race condition.

mrousavy, 2024-01-30:
Oh nice research, thanks for getting back here! I'll try to take a look with that info

mrousavy, 2024-01-30:
Btw.; I just merged this PR where you can query for `phase-detection` https://github.com/mrousavy/react-native-vision-camera/pull/2455

ritathesaver, 2024-02-01:
still getting this issue :(

mrousavy, 2024-02-02:
@ritathesaver can you try to use this format:

```ts
const format = useCameraFormat(device, [
  { autoFocusSystem: 'phase-detection' }
])
```

pass that to the Camera, then try to take a photo and see if that works?

stevengoldberg, 2024-02-09:
> @ritathesaver can you try to use this format:
> 
> ```ts
> const format = useCameraFormat(device, [
>   { autoFocusSystem: 'phase-detection' }
> ])
> ```
> 
> pass that to the Camera, then try to take a photo and see if that works?
> 
> > (this API was introduced in VisionCamera 3.9-beta.0)

FWIW I just tried this and get the same error in 3.9.0-beta.3. I tried it with two different formats, both using phase-detection:

```
{"autoFocusSystem": "phase-detection", "fieldOfView": 101.02159118652344, "maxFps": 60, "maxISO": 5472, "maxZoom": 171.875, "minFps": 1, "minISO": 57, "photoHeight": 3024, "photoWidth": 4032, "pixelFormats": ["yuv", "yuv", "rgb", "unknown", "yuv", "unknown", "yuv", "unknown", "rgb"], "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 768, "videoStabilizationModes": ["auto", "off", "standard"], "videoWidth": 1024}
```

```
{"autoFocusSystem": "phase-detection", "fieldOfView": 103.62532043457031, "maxFps": 30, "maxISO": 12768, "maxZoom": 189, "minFps": 1, "minISO": 57, "photoHeight": 3024, "photoWidth": 4032, "pixelFormats": ["yuv", "yuv", "rgb", "unknown", "yuv", "unknown", "yuv", "unknown", "rgb"], "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 3024, "videoStabilizationModes": ["auto", "off"], "videoWidth": 4032}
```

When I try to take a flash photo I get:

```
[capture/unknown: Error Domain=AVFoundationErrorDomain Code=-11800 "The operation could not be completed" UserInfo={NSUnderlyingError=0x281d3c480 {Error Domain=NSOSStatusErrorDomain Code=-16800 "(null)"}, NSLocalizedFailureReason=An unknown error occurred (-16800), AVErrorRecordingFailureDomainKey=4, NSLocalizedDescription=The operation could not be completed}]
```

kolyaVV, 2024-02-09:
I do not know if it is related! Note what is going on in `"pixelFormats": ["yuv", "yuv", "rgb", "unknown", "yuv", "unknown", "yuv", "unknown", "rgb"],`

mrousavy, 2024-02-13:
@stevengoldberg did you try to change `enableShutterSound`?

mrousavy, 2024-02-13:
Does the flash work correctly when using a third party app like Instagram, Snapchat or WhatsApp?

stevengoldberg, 2024-03-07:
@mrousavy I just tested this on an iPhone 15 Pro running iOS 17.4 and the error is the same. autoFocusSystem is phase-detection. It occurs whether enableShutterSound is true or false.

stevengoldberg, 2024-03-16:
Another data point — I tried `expo-camera/next` in my app with no other changes and the flash works there, too.

mrousavy, 2024-03-17:
Yea, because expo-camera does not support formats at all.

On VisionCamera you can fix this by selecting a different format - there are some formats that work, some that don't.
I've seen that stuff like maxISO affects it. 

I'm trying to fix the issue where some formats don't work, but I haven't dedicated the time to do that yet.

jodrescher, 2024-04-10:
I have the same problem on an iPhone 12 mini. A fix would be very welcome.

JustinHaut, 2024-04-29:
Re: 3.9.2

TakePhotoOption notes:

-Passing flash causes Android photo capture to lag ~5 seconds minimum.
-Passing enableAutoDistortionCorrection causes iOS app to crash.

Passing no options works great! Seems fixed in v4.0.1 though.

stevengoldberg, 2024-05-20:
I updated to 4.0.4 and flash is now working as expected on an iPhone 15 Pro 🎉

razr130, 2024-06-18:
I still getting the issue
I'm using react-native-vision-camera version 4.3.2, iPhone XS 
Is there any workaround?

jkaufman, 2024-09-18:
I'm seeing this error logged in production from an iPhone 14 Pro Max capturing images in low light.
```
"format": {
  minISO": 57,
  "photoHeight": 1188,
  "fieldOfView": 106.20069885253906,
  "maxFps": 60,
  "videoHeight": 1080,
  "autoFocusSystem": "phase-detection",
  "supportsVideoHdr": false,
  "videoWidth": 1920,
  "minFps": 1,
  "supportsPhotoHdr": false,
  "photoWidth": 2112,
  "supportsDepthCapture": false,
  "maxISO": 5472,
  "videoStabilizationModes": [
    "auto",
    "cinematic",
    "off",
    "standard",
    "cinematic-extended"
  ]
}
```

jkaufman, 2024-09-18:
Logs from iPhone 15 Pro: [log.txt](https://github.com/user-attachments/files/17051053/log.txt)

khushbupatel2, 2024-12-04:
I'm using "react-native-vision-camera": "^4.5.3", it is still not working Flash on iOS.

elourenco, 2025-02-14:
I still getting the issue
I'm using react-native-vision-camera version 4.3.2, iPhone XS
Is there any workaround?