margelo/react-native-vision-camera #2393: ❓ Understanding Frames

afiller, 2024-01-15:
### Question

Hi!

I'm currently exploring RN Vision Camera since we want to use it together with a tensorflow lite model. As shown in your examples, the frame processor seems to be what we need, but I'm facing some issues with the Frame result object:

I request the following camera format:

```
const format = useCameraFormat(device, [
    { videoResolution: { width: 640, height: 480 } },
    { fps: 30 },
    { videoStabilizationMode: 'auto' }
  ])
```

...and I get:

{"autoFocusSystem": "phase-detection", "fieldOfView": 65.35665130615234, "maxFps": 60, "maxISO": 3168, "maxZoom": 189, "minFps": 1, "minISO": 33, "photoHeight": 3024, "photoWidth": 4032, "pixelFormats": ["yuv", "yuv", "rgb", "unknown", "unknown", "unknown"], "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 480, "videoStabilizationModes": ["auto", "off"], "videoWidth": 640}

The frame in the frame processor is:

**480x640 (yuv)**

But the data length is:

```
const data = frame.toArrayBuffer()
const length = data.length
```

==> **494080**

==> 
```
{"0": 0, "1": 0, "10": 0, "100": 31, "1000": 30, "10000": 77, "100000": 45, "100001": 46, "100002": 47, "100003": 47, "100004": 48, "100005": 48, "100006": 48, "100007": 49, "100008": 50, "100009": 52, "10001": 83, "100010": 54, "100011": 57, "100012": 59, "100013": 62, "100014": 64, "100015": 67, "100016": 69, "100017": 71, "100018": 73, "100019": 75, "10002": 86, "100020": 76, "100021": 78, "100022": 79, "100023": 81, "100024": 82, "100025": 83, "100026": 84, "100027": 86, "100028": 87, "100029": 88, "10003": 86, "100030": 89, "100031": 91, "100032": 92, "100033": 93, "100034": 94, "100035": 94, "100036": 95,  ...
```

**My questions:**

1) However I do then maths, I can't find out how it comes from 480*640 to a length of 494080?
2) The data array buffer is an object and not an array?
3) The first values of the array ("0", "1", "2") are always 0? Why?

### VisionCamera Version

3.7.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2024-01-15:
Hi!

1. Try logging `frame.bytesPerRow`, this is how many bytes one row has. If you do `frame.bytesPerRow / frames.width` you can figure out the bytes each pixel has. Multiply that by the `frame.height` and you should get the total amount of bytes this frame consists of - that should be the same number as `data.length`. 
2. YUV is a bit trickier than RGB, it's a 4:2:0 8-bit format, so one pixel should consist of 3 bytes. It _should_ be 480 * 640 * 3. I am not sure why it is not 921.600, but instead 494.080 though. Maybe this is a bug?
3. The data buffer is an object because react-native-worklets works that way. It is a proxy object to avoid copying :) It can be treated like an Array though. 
4. The first values might be 0 either because there is a black binned area in the Frame (some formats just have that, hardware stuff), or because I have a bug in my code. Maybe related to point 2 here. Let me know if you find out whether that's a bug or not.

afiller, 2024-01-15:
Hi @mrousavy,

thank you for your reply at the speed of light! :-)

Regarding 1:
The bytesPerRow are 772 => 772 / 480 = 1,6083333333 🤷‍♂️

Regarding 2:
Yes, 480 * 640 * 3 was what I expected. :-D

Regarding 3:
Ah, makes sense, thanks for the explanation. 👍

mrousavy, 2024-01-15:
772 is weird. To get to the first even number, we could assume that it actually is 768 instead of 772, and the first 4 bytes are just black pixels so it fills the native GPU surface bandwidth? 

I still don't get it though. Can you try with a larger format, e.g. 1080p or 4k? Is that the case as well?

afiller, 2024-01-15:
Frame: 1944x2592 (yuv)
Length: 7724160
Bytes per row: 2980

I can't find appropriate fits also with higher resolutions. I'm using and iPhone 12 with the newest iOS... how can I help you to dive deeper into this issue?

mrousavy, 2024-01-15:
Maybe related: https://stackoverflow.com/questions/46140785/cmsamplebuffer-frame-converted-to-vimage-has-wrong-colors

mrousavy, 2024-01-15:
> how can I help you to dive deeper into this issue?

What issue? We're just questioning the format/layout of the pixels, but as far as I'm seeing it there's no issue.
Just try what you're trying to do (pass it to TFLite or whatever) and see if that works.

mrousavy, 2024-01-16:
Btw: This might answer your question: https://developer.apple.com/documentation/accelerate/conversion/understanding_ypcbcr_image_formats?language=objc

As well as this: 
[image]

With YUV 4:2:0, for every 8 components in Y, we have 2 components in U+V.
Meaning It should be 1,25 bytes per pixel on average, and obviously that scales up - so for 480 pixels in width, we would need 480 * 1,25 = 600 bytes. In RGB, you would need 1.440 bytes for the same pixels.

I can only assume that the 772 pixels are some kind of additional padding - but I am honestly not sure about this either.

mrousavy, 2024-01-26:
Closing this for now.

closed 2024-01-26 by mrousavy

afiller, 2024-02-08:
Hi Marc,

thank you very much for your detailed explanation. I got quite an expert in all possible formats of YUV in my long discussions with ChatGPT 4 Data Analyst. We tried to explore the Array Buffer coming from the iPhone camera and we were able to ensure that it really contains an image but we were not at all able to find it which exact coding of YUV it is. ...and we had many and long discussions. :-D 

Since this is quite a blocker for us in one project and we really need access to a correct RGB array buffer I wanted to ask you if you would also code this for money? I have seen in your article (https://mrousavy.com/blog/VisionCamera-Pose-Detection-TFLite) that you were already working with such problems. We need a frame processor which provides us (at best converted using native code on iOS and Android) a resized RGB frame buffer as input for our TF lite model independent from any camera formats by Apple, Samsung, Google... etc.

If you are interested I would highly welcome to have a brief call with you, e.g. via Google Meet or phone (we are located in Mannheim, Germany).

Thanks in advance,

Andreas

mrousavy, 2024-02-09:
Of course, send me an email! :)