margelo/react-native-vision-camera #2403: Code scanner coordinates are relative to sensore orientation instead of Camera Preview🐛

marksadc, 2024-01-17:
### What's happening?

I'm using code scanner to detect qr codes on document. I need the coordinates of codes relative to camera preview in order to crop the image of document with qr codes. However, despite of using my device in portrait the coordinates look like i was using landscape mode: the 'y' coordinate is horizontal and 'x' is vertical. Seems like the're realative to sensor orientation of my camera.

### Reproduceable Code

```tsx
export const CameraSCreen = () => {
  const device = useCameraDevice('back');
  const camera = useRef<Camera>(null);

  const codeScanner = useCodeScanner({
    codeTypes: ['qr'],
    onCodeScanned: (codes, frame) => {
      if (codes.length) {
        console.log('frame', frame);
        console.log('codes', JSON.stringify(codes));
      }
    },
  });

  if (device == null) return <Text>no camera</Text>;

  return (
    <Camera
      style={StyleSheet.absoluteFill}
      device={device}
      isActive={true}
      codeScanner={codeScanner}
      onError={(e) => {
        console.error(e);
      }}
      ref={camera}
    />
  );
};
```

### Relevant log output

```shell
LOG  frame {"height": 720, "width": 1280}
 LOG  codes [{"frame":{"width":115,"y":533,"height":137,"x":2},"corners":[{"y":534,"x":10},{"y":533,"x":117},{"y":668,"x":110},{"y":670,"x":2}],"value":"46334-2023-11-11","type":"qr"}]
```

### Camera Device

```json
LOG  {
  "formats": [],
  "sensorOrientation": "landscape-right",
  "hardwareLevel": "full",
  "maxZoom": 7,
  "minZoom": 0.6009132266044617,
  "maxExposure": 24,
  "supportsLowLightBoost": true,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera",
    "ultra-wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": true,
  "minFocusDistance": 9.800000093460085,
  "minExposure": -24,
  "name": "BACK (0)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Pixel 6A

### VisionCamera Version

3.8.0

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-01-17:
Hey - yes this is intentional. This is the truly native coordinate. To convert this to screen coordinates, you can do this yourself by taking the `boundingBox` into consideration and also the screen position of your Preview View.

closed 2024-01-17 by mrousavy

oscar-b, 2024-01-23:
> Hey - yes this is intentional. This is the truly native coordinate. To convert this to screen coordinates, you can do this yourself by taking the `boundingBox` into consideration and also the screen position of your Preview View.

@mrousavy It's not entirely trivial juggling the different coordinate systems, I'm yet to find a working example of this. I have it working for v2 and `vision-camera-code-scanner` but v3 seems different in a couple of ways. Got any guidance?

mrousavy, 2024-01-23:
Yea, I could add functionality to convert those coordinates.

oscar-b, 2024-01-23:
> Yea, I could add functionality to convert those coordinates.

What would be awesome! I'm also lacking some understanding (and documentation) about how it works when the format is requested as `{ width: 1920, height: 1080 }` but it's displayed in a portrait `<View>`. Seems like the camera 0,0 is in the upper right, so 90 deg rotated?

I have the `<Camera>` in a `<View style={{ flex: 1 }} onLayout={handleLayoutChange}>` to get the preview size so I can calculate a scaling factor, I assume this is still needed since we're drawing on a separate SVG canvas and not on the camera frame, right?

shacharcohen1997, 2024-04-11:
> Hey - yes this is intentional. This is the truly native coordinate. To convert this to screen coordinates, you can do this yourself by taking the `boundingBox` into consideration and also the screen position of your Preview View.

how can i do that