margelo/react-native-vision-camera #2404: fatal error: 'ReactCommon/CallInvokerHolder.h' file not found🔧

Ritesh175862, 2024-01-17:
### How were you trying to build the app?

[image]

### Full build logs

```tsx
yarn add react-native-vision-camera
cd ios && npx pod-install

yarn run android
```

### Project dependencies

```json
{
  "name": "PurpleSENSE",
  "version": "1.0.13",
  "private": true,
  "scripts": {
    "clean": "react-native-clean-project",
    "android": "react-native run-android",
    "ios": "react-native run-ios",
    "lint": "eslint src",
    "start": "node node_modules/react-native/local-cli/cli.js start",
    "test": "jest --passWithNoTests",
    "postinstall": "node ./installandroid.js && ./installios.js",
    "prepare": "husky install",
    "update-icon": "node ./update-icon.js",
    "postversion": "react-native-version"
  },
  "sdkDependencies": {
    "SalesforceMobileSDK-Android": "https://github.com/ProUnitas-Mobile/SalesforceMobileSDK-Android-12.git#master",
    "SalesforceMobileSDK-iOS": "https://github.com/ProUnitas-Mobile/SalesforceMobileSDK-iOS-12.git#master"
  },
  "dependencies": {
    "@eva-design/eva": "^2.1.1",
    "@react-native-community/datetimepicker": "5.1.0",
    "@react-native-community/masked-view": "^0.1.10",
    "@react-native-firebase/analytics": "^14.11.0",
    "@react-native-firebase/app": "^14.11.0",
    "@react-native-firebase/crashlytics": "^14.11.0",
    "@react-navigation/native": "^5.8.10",
    "@react-navigation/stack": "^5.12.8",
    "@ui-kitten/components": "5.0.0",
    "android": "^0.0.8",
    "axios": "^0.24.0",
    "base-64": "^1.0.0",
    "buffer": "^6.0.3",
    "class-transformer": "^0.5.1",
    "create-react-class": "^15.7.0",
    "dayjs": "^1.10.7",
    "deprecated-react-native-prop-types": "^5.0.0",
    "emoji-strip": "^1.0.1",
    "install": "^0.13.0",
    "lodash": "^4.17.15",
    "moment": "^2.29.1",
    "native-base": "^3.4.28",
    "react": "18.1.0",
    "react-hook-form": "^7.17.5",
    "react-native": "0.70.14",
    "react-native-device-info": "^10.12.0",
    "react-native-force": "git+https://github.com/ProUnitas-Mobile/SalesforceMobileSDK-ReactNative-12.git#master",
    "react-native-gesture-handler": "^1.9.0",
    "react-native-keyboard-aware-scroll-view": "^0.9.5",
    "react-native-pager-view": "^5.4.7",
    "react-native-reanimated": "1.13.4",
    "react-native-safe-area-context": "^3.3.2",
    "react-native-screens": "^2.15.0",
    "react-native-svg": "12.2.0",
    "react-native-tab-view": "^3.1.1",
    "react-native-version-check": "^3.4.7",
    "react-native-vision-camera": "^3.8.0",
    "react-query": "^3.26.0",
    "reflect-metadata": "^0.1.13",
    "styled-components": "^5.3.1",
    "styled-system": "^5.1.5",
    "use-debounce": "^7.0.0",
    "validator": "^13.11.0"
  },
  "devDependencies": {
    "@babel/core": "^7.12.9",
    "@babel/plugin-proposal-decorators": "^7.16.0",
    "@babel/runtime": "^7.12.5",
    "@react-native-community/eslint-config": "^2.0.0",
    "@testing-library/react-native": "^7.2.0",
    "@types/jest": "^26.0.18",
    "@types/react": "^16.14.4",
    "@types/react-native": "^0.63.37",
    "@types/react-test-renderer": "^16.9.5",
    "babel-jest": "^26.6.3",
    "eslint": "7.14.0",
    "husky": "^7.0.2",
    "jest": "^26.6.3",
    "jest-circus": "^27.2.0",
    "lint-staged": "^11.1.2",
    "metro-react-native-babel-preset": "^0.64.0",
    "mockdate": "^3.0.5",
    "react-native-clean-project": "^3.6.7",
    "react-native-svg-transformer": "^0.14.3",
    "react-native-typescript-transformer": "^1.2.13",
    "react-native-version": "^4.0.0",
    "react-test-renderer": "16.13.1",
    "rimraf": "3.0.2",
    "ts-jest": "^26.4.4",
    "typescript": "^4.1.2"
  },
  "jest": {
    "preset": "react-native",
    "moduleFileExtensions": [
      "ts",
      "tsx",
      "js"
    ],
    "moduleNameMapper": {
      "\\.svg": "<rootDir>/src/__mocks__/svgMock.js"
    },
    "transform": {
      "^.+\\.(js)$": "babel-jest",
      "\\.(ts|tsx)$": "ts-jest"
    },
    "testRegex": "(/__tests__/.*|\\.(test|spec))\\.(ts|tsx|js)$",
    "testPathIgnorePatterns": [
      "\\.snap$",
      "node_modules/"
    ],
    "cacheDirectory": ".jest/cache",
    "collectCoverage": true,
    "modulePaths": [
      "<rootDir>"
    ],
    "transformIgnorePatterns": [
      "node_modules/@ui-kitten"
    ]
  },
  "lint-staged": {
    "./*.{js,tsx}": [
      "eslint --fix"
    ]
  },
  "engines": {
    "node": ">=14.0.0",
    "npm": "please-use-yarn",
    "yarn": ">= 1.22.5"
  },
  "resolutions": {
    "typescript": "4.4.3"
  },
  "husky": {
    "hooks": {
      "pre-push": "yarn test"
    }
  }
}
```

### VisionCamera Version

3.8.0

### Target platforms

Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-01-18:
You need RN 0.71+ for VisionCamera 3.

closed 2024-01-18 by mrousavy

hanzhenk, 2024-01-19:
"react-native": "0.66.5",
Can I use this warehouse? In which version can it be run?

hanzhenk, 2024-01-19:
@mrousavy  I also encountered this problem, "react-native": "0.66.5", because upgrading 'react-native' may cause other problems, which version of this repository can I use?

mrousavy, 2024-01-19:
Warehouse? What warehouse?

And idk which version works on react-native 0.66, that is a pretty old version. Probably some VisionCamera 2 version, not sure.

hanzhenk, 2024-01-19:
Can I use a version of 'react-native-vision-camera'? It runs on the iOS is very good, but he reported the same mistake on android (' ReactCommon/CallInvokerHolder. H 'file not found)
[image]

hanzhenk, 2024-01-19:
createDevice = (pos = 'back') => {
    this.cameraPos = pos;
    const devices = Camera.getAvailableCameraDevices()
    console.log('devices--------------------', devices);
    const device = devices.find((d) => d.position === pos)
    this.setState({ device })
  }
  // devices-------------------- {"_A": null, "_x": 0, "_y": 0, "_z": null}
  
  Console error:
  TypeError: devices.find is not a function. (In 'devices.find(function (d) {
  return d.position === pos;
  })', 'devices.find' is undefined)

@mrousavy  In my "react-native": "0.66.5", I tried "React-native-vision-camera ": "^2.16.1", it initializes the whole project smoothly, but there is no way to initialize the camera function, In the "react-native": "0.66.5" version, is there still a way to use "React-native-vision-camera"? Or does it have some other way of initializing the camera? Looking forward to your reply.

  
[image]

mrousavy, 2024-01-19:
In V2, getAvailableCameraDevices() returns a Promise. You need to also read the docs from V2, which are in the v2 branch of the repo - not on the public website.

hanzhenk, 2024-01-19:
> In V2, getAvailableCameraDevices() returns a Promise. You need to also read the docs from V2, which are in the v2 branch of the repo - not on the public website.

Ok, I'll look for it. Thank you very much.

Ritesh175862, 2024-01-23:
In V2 can we use QR/Barcode scanner