margelo/react-native-vision-camera #2405: 🐛 Black screen on Redmi K30 Pro

ooxoxx, 2024-01-17:
### What's happening?

I'm working on some QR code scanning functionality. It works well on android emulator and most devices. But on Redmi K30 pro, the camera component failed to destroy itself leaving a "black mask" (maybe a surface view?) out there, when I switch back with navigation.

### Reproduceable Code

```tsx
import {useIsFocused} from '@react-navigation/core';

export const useIsForeground = (): boolean => {
  const [isForeground, setIsForeground] = useState(true);

  useEffect(() => {
    const onChange = (state: AppStateStatus): void => {
      setIsForeground(state === 'active');
    };
    const listener = AppState.addEventListener('change', onChange);
    return () => listener.remove();
  }, [setIsForeground]);

  return isForeground;
};

export default function ({navigation}: RootStackScreenProps<'QrScanner'>) {
  const camera = useCameraDevice('back');
  const isFocused = useIsFocused();
  const isForeground = useIsForeground();
  const isActive = isFocused && isForeground;

  // ...

  const codeScanner = useCodeScanner({
    codeTypes: ['qr'],
    onCodeScanned,
  });

  console.log(`isFocused: ${isFocused}`);

  if (Platform.OS === 'android' && !isFocused) {
    return <View />;
  }

  return (
    <View style={styles.container}>
      {camera != null && (
        <Camera
          style={StyleSheet.absoluteFill}
          device={camera}
          isActive={isActive}
          codeScanner={codeScanner}
          enableZoomGesture={true}
          torch={torch ? 'on' : 'off'}
        />
      )}
    </View>
  );
}

const styles = StyleSheet.create({
  container: {flex: 1, backgroundColor: 'black'},
});
```

### Relevant log output

```shell
2024-01-15 17:23:22.318   903-1010  AppScoutStateMachine    com.lisgard                          D  903-ScoutStateMachinecreated
2024-01-15 17:23:22.321  5442-7904  SuggestManager          com.miui.securitycenter.remote       E  openApp name = com.lisgard
2024-01-15 17:23:22.413  3144-3505  ActivityManagerWrapper  com.miui.home                        E  getRecentTasks: taskId=12772   userId=0   baseIntent=Intent
                                                                                                    { act=android.intent.action.MAIN flag=268435456
[... 275 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Redmi K30 pro

### VisionCamera Version

3.7.1

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

hengkx, 2024-01-18:
I only have a small portion of the bottom screen black, I'm not sure what the reason is

mrousavy, 2024-01-18:
Hm, this is weird. Maybe this is an actual bug in the Redmi phones?

I am not sure - from the logs it looks like it destroys everything properly 🤔

ooxoxx, 2024-01-18:
> Hm, this is weird. Maybe this is an actual bug in the Redmi phones?
> 
> I am not sure - from the logs it looks like it destroys everything properly 🤔

Thanks for your attention. Other camera and qr scanning apps works properly on this phone. This bug is also reproducible on HONOR Magic4 Pro. What do you suggest to do next?

ooxoxx, 2024-01-18:
Is `Device error received, code 4` there related to this? surface view is created twice due to the error? That is the only difference in logs between the problem phone and other devices that work.

mrousavy, 2024-01-18:
I don't know. I'd need to thoroughly investigate this, preferrably with the device in my hands.

francesco-clementi-92, 2024-01-30:
The bug is reproducible also on a honor x6a

chj-damon, 2024-01-31:
> I only have a small portion of the bottom screen black, I'm not sure what the reason is

here facing the same problem when taking photos using `Redmi K30` or `Huawei Honor`

also, it will leave a black screen after navigating to other screen when qr code is scanned.

mrousavy, 2024-02-01:
Working on a fix for this issue here: https://github.com/mrousavy/react-native-vision-camera/pull/2494 👀 

Still in development, so don't test yet. Long way to go, this is a really complicated issue (as you can see in the changes/Files of that PR)...

closed 2024-02-06 by mrousavy

mrousavy, 2024-02-06:
Just released [VisionCamera 3.9.0-beta.1](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v3.9.0-beta.1), which includes the [big `PersistentCaptureSession` PR](https://github.com/mrousavy/react-native-vision-camera/pull/2494) that aims to fix all the blackscreen issues.... :eyes::tada: 

You can try that new beta out yourself and let me know if you still encounter blackscreen issues. I'll continue to test on more devices in the meantime, organize the code a bit better, and then I'll tackle the preview stretching, focus, and flash issues!

If that fixes the blackscreen issues for you, please consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks! ❤️

cauyyl, 2025-03-05:
@mrousavy  I mett a new  issue, the black screes keep 7seconds before the camera show.
 redmi 11, android 11, react native vision camrea 3.9.2, react native 0.72.5.

cauyyl, 2025-03-06:
> [@mrousavy](https://github.com/mrousavy) I meet a new issue, the black screes keep 7seconds before the camera show. [socket]:check permission begin! spend too much time. redmi 11, android 11, react native vision camrea 3.9.2, react native 0.72.5. Is there any related issues?

solved by add format to <Camera />