margelo/react-native-vision-camera #2413: 🐛 Permissions don't update for external changes

mattwoolford, 2024-01-18:
### What's happening?

The permission hooks (e.g. `useCameraPermission()`) don't listen for external changes, for example in device settings like suggested in the documentation via the `Linking` API. Therefore, using `hasPermission` will not reliably inform the app without the hook being reloaded.

### Reproduceable Code

```tsx
const [hasPermission, requestPermission] = useCameraPermission();

const [isExplicitlyDenied, setIsExplicitlyDenied] = useState<boolean>(false);

return (
    <>
        {hasPermission ? (
            <Text>Permission Granted</Text>
        ) : (
            <Pressable onPress={async () => {
                if(isExplicitlyDenied) Linking.openSettings();
                else {
                    const permissionGranted = await requestPermission();
                    if(!permissionGranted) setIsExplicitlyDenied(true);
                }
            }}>Ask permission to use the camera</Pressable>
        )}
    </>
)

/**
Actions: In runtime, press the `<Pressable>` component and deny access to the camera when prompted. Press the `<Pressable>` component again and change the `Camera` permission in the opened settings to a variant of "Allow". Return to the app, and the `<Pressable>` component will still be visible, despite the permission being granted.
**/
```

### Relevant log output

```shell
// N/A
```

### Camera Device

```json
// N/A
```

### Device

Google Pixel 8 Pro

### VisionCamera Version

3.8.0

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-01-18:
Doesn't the app restart itself when permissions change externally? At least that's how it works on iOS

mattwoolford, 2024-01-18:
Seems only to restart if the permissions change from allowed -> denied. The app doesn't restart when it goes from denied -> allowed on Android

mrousavy, 2024-01-18:
Well that's weird again. I don't see any Java/Kotlin API to listen to such changes though?

mattwoolford, 2024-01-18:
Modifying the hooks with this workaround would appear to work:

```
export function useCameraPermission(): PermissionState {

    const [hasPermission, setHasPermission] = useState(() => Camera.getCameraPermissionStatus() === "granted");

    const requestPermission = useCallback(async () => {
        const result = await Camera.requestCameraPermission();
        return result === "granted";
    }, []);

    // Listen for changes in the app state and use this to recheck permissions
    useEffect(() => {

        const subscription = AppState.addEventListener("change", (nextAppState) => {
            setHasPermission(Camera.getCameraPermissionStatus() === "granted");
        });

        return () => {
            subscription.remove();
        };

    }, []);

    return {
        hasPermission,
        requestPermission
    };
}
```

As an additional observation, the app state changes when the request prompt appears, so it would seem you can also rely on the event listener to update the `hasPermission` state, without also having to do this in `request`?

mrousavy, 2024-01-18:
Yea that is indeed a good workaround. Maybe we can even add this to the repo? Can you create a PR?

mrousavy, 2024-01-19:
created a PR: https://github.com/mrousavy/react-native-vision-camera/pull/2420

closed 2024-01-19 by mrousavy

mattwoolford, 2024-01-20:
Great. Thanks Marc. Apologies, I was trying to test the solution on iOS too before creating a PR but having build issues. Glad this has been fixed!