margelo/react-native-vision-camera #2416: 🐛 enableFpsGraph doesn't show anything on Pixel 7

jslok, 2024-01-19:
### What's happening?

I'm assuming enableFpsGraph shows the fps over the camera preview. On Pixel 7 with enableFpsGraph enabled I don't see it shown. I don't have another android device to test with.

### Reproduceable Code

```tsx
<Camera
        frameProcessor={frameProcessor}
        format={format}
        ref={camera}
        style={StyleSheet.absoluteFill}
        device={device}
        isActive={true}
        pixelFormat={pixelFormat}
        enableFpsGraph={true}
        onError={console.log}
        videoStabilizationMode={videoStabilizationMode}
        videoHdr={false}
        photo={false}
        resizeMode={'contain'}
      />
```

### Relevant log output

```shell
na
```

### Camera Device

```json
{"hardwareLevel": "full", "hasFlash": true, "hasTorch": true, "id": "0", "isMultiCam": true, "maxExposure": 24, "maxZoom": 8, "minExposure": -24, "minFocusDistance": 0, "minZoom": 0.6704425811767578, "name": "BACK (0)", "neutralZoom": 1, "physicalDevices": ["wide-angle-camera", "ultra-wide-angle-camera", "telephoto-camera"], "position": "back", "sensorOrientation": "landscape-right", "supportsFocus": true, "supportsLowLightBoost": true, "supportsRawCapture": true}
```

### Device

Pixel 7

### VisionCamera Version

3.8.1

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-01-19:
Ah yeah the FPS graph isn't supported on Android yet. I think there are native tools to investigate SurfaceView's performance though, some debug views in Settings

mrousavy, 2024-01-30:
Working on adding this here: https://github.com/mrousavy/react-native-vision-camera/pull/2460

closed 2024-01-30 by mrousavy