margelo/react-native-vision-camera #2418: 🔧 Execution failed for task ':react-native-vision-camera:buildCMakeDebug[arm64-v8a ]

hilarionOrganit, 2024-01-19:
### How were you trying to build the app?

yarn start --reset-cache

### Full build logs

```tsx
FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:buildCMakeDebug[arm64-v8a
]'.
> com.android.ide.common.process.ProcessException: ninja: Entering directory `C:
\laragon\www\haccpv2-mobile\node_modules\react-native-vision-camera\android\.cxx
\Debug\676r4v4a\arm64-v8a'
  [1/18] Building CXX object CMakeFiles/VisionCamera.dir/C_/laragon/www/haccpv2-
mobile/node_modules/react-native-vision-camera/cpp/MutableRawBuffer.cpp.o
  [2/18] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/PassThroug
hShader.cpp.o
  [3/18] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/OpenGLCont
ext.cpp.o
  [4/18] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCame
ra.cpp.o
  [5/18] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/OpenGLRend
erer.cpp.o
  [6/18] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/MutableJBy
teBuffer.cpp.o
  C:/laragon/www/haccpv2-mobile/node_modules/react-native-vision-camera/android/
src/main/cpp/MutableJByteBuffer.cpp:17:21: warning: function previously declared
 with an implicit exception specification redeclared with an explicit exception
specification [-Wimplicit-exception-spec-mismatch]
  MutableJByteBuffer::~MutableJByteBuffer() noexcept {
                      ^
  C:/laragon/www/haccpv2-mobile/node_modules/react-native-vision-camera/android/
src/main/cpp/MutableJByteBuffer.h:22:3: note: previous declaration is here
    ~MutableJByteBuffer();
    ^
  1 warning generated.
  [7/18] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/VideoPipel
ine.cpp.o
  [8/18] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameproce
ssor/FrameProcessorPluginHostObject.cpp.o
  [9/18] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameproce
ssor/FrameHostObject.cpp.o
  [10/18] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameproc
essor/VisionCameraProxy.cpp.o
  [11/18] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameproc
essor/JSIJNIConversion.cpp.o
  [12/18] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameproc
essor/java-bindings/JFrame.cpp.o
  [13/18] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameproc
essor/java-bindings/JFrameProcessor.cpp.o
  [14/18] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameproc
essor/java-bindings/JSharedArray.cpp.o
  C:/laragon/www/haccpv2-mobile/node_modules/react-native-vision-camera/android/
src/main/cpp/frameprocessor/java-bindings/JSharedArray.cpp:36:88: warning: forma
t specifies type 'int' but the argument has type 'size_t' (aka 'unsigned long')
[-Wformat]
    __android_log_print(ANDROID_LOG_INFO, TAG, "Allocating ArrayBuffer with size
 %i...", byteBuffer->getDirectSize());

 ~~      ^~~~~~~~~~~~~~~~~~~~~~~~~~~

 %zu
  1 warning generated.
  [15/18] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameproc
essor/java-bindings/JFrameProcessorPlugin.cpp.o
  [16/18] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameproc
essor/java-bindings/JVisionCameraScheduler.cpp.o
  [17/18] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameproc
[... 261 lines cut ...]
```

### VisionCamera Version

3.8.0

### Target platforms

Android

### Operating system

Windows

### Can you build the VisionCamera Example app?

No, I cannot build the Example app either

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-01-19:
Hey - this has been posted before on this repo and it is always a cache issue. Try fully clearing your cache by deleting node_modules, as well as any cache folders in `android/` (cxx, gradle, build, ...) and in `android/app`.

closed 2024-01-19 by mrousavy

hilarionOrganit, 2024-01-19:
Do I have to clear caches every time the error occurs?
Are there other alternatives?

mrousavy, 2024-01-19:
> Do I have to clear caches every time the error occurs?

Yes. Because it is a cache error. Normally, this error shouldn't occur often though. In the past few months I've only had it once myself.

Maybe there is something we can do in the `build.gradle` setup of react-native-worklets-core to fix this 🤷

anuragkaushik10, 2024-03-06:
@mrousavy  How to fully clear cache ?