margelo/react-native-vision-camera #2419: 🐛 Crash when scanning barcodes + repeatedly mounting the camera.

daankennes, 2024-01-19:
### What's happening?

The screen shared below is being mounted and unmounted multiple times. After being mounted a random number of times (around 5 leads to the crash), the app crashes without providing any additional information (it simply closes). I've only run into this on Android.

### Reproduceable Code

```tsx
import React, {useCallback, useContext, useEffect, useRef, useState} from 'react';
import {View, StyleSheet, Dimensions, ScrollView} from 'react-native';
import {
    Camera,
    CameraPermissionStatus,
    useCameraDevice,
    useCodeScanner,
    Code,
    CodeScannerFrame,
} from 'react-native-vision-camera';
import {ActivityIndicator, Button, Text} from 'react-native-paper';
import {useTranslation} from 'react-i18next';
import {getStatusBarHeight} from 'react-native-status-bar-height';
import BarcodeMask from 'react-native-barcode-mask';
import {green, red} from 'library/utilities/DevelopmentUtilities';
import {ThemeContext} from 'library/context/ThemeContext';
import {BarcodeScannerProps} from './BarcodeScanner';
import {useIsFocused, useNavigation} from '@react-navigation/native';

type CornerPointsSet = {x: number; y: number}[];
type CornerPointsSets = CornerPointsSet[];

const finderWidth = 280;
const finderHeight = 180;
const bottomComponentExtraMargin = 20;

export function VisionBarcodeScanner(props: BarcodeScannerProps) {
    const {styleSheet} = useContext(ThemeContext);
    const {t} = useTranslation();
    const device = useCameraDevice('back');

    const [cameraPermissionStatus, setCameraPermissionStatus] =
        useState<CameraPermissionStatus>('not-determined');
    const codeScanner = useCodeScanner({
        codeTypes: ['code-128', 'code-39', 'code-93', 'codabar', 'ean-13', 'ean-8', 'upc-e', 'qr'],
        onCodeScanned,
    });
    const isFocused = useIsFocused();
    const navigation = useNavigation();

    const [dimensions, setDimensions] = useState({
        width: Dimensions.get('window').width,
        height: Dimensions.get('window').height,
    });
    const [topContentHeight, setTopContentHeight] = useState(0);
    const [cornerPointsSets, setCornerPointsSets] = useState<CornerPointsSets>();
    const [cornerPointsVisualizationColor, setCornerPointsVisualizationColor] = useState<
        'green' | 'red'
    >('red');
    // display after scan if async code is executed
    const [displaySpinner, setDisplaySpinner] = useState(false);

    const validBarcodeValue = useRef<string | undefined>();
    const screenActive = useRef(true);
    const processingFrame = useRef(false);
    const waitingForResult = useRef(false);

    const finderMinX = Math.floor((dimensions.width - finderWidth) / 2);
    const finderMinY = Math.floor((dimensions.height - finderHeight) / 2);

    React.useEffect(() => {
        const unsubscribe = navigation.addListener('blur', () => {
            screenActive.current = false;
[... 404 lines cut ...]
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
```

### Device

Pocophone F1

### VisionCamera Version

3.8.0

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

marcshilling, 2024-01-22:
Stacktrace looks pretty similar to https://github.com/mrousavy/react-native-vision-camera/issues/2116?

DevUser0491, 2024-01-22:
Hello @daankennes. 
Your code is a good example with cornerPointSets. Is it possible to do a one square frame instead of 4 corners? Like this?
<details>
[image]
</details>

daankennes, 2024-01-22:
@marcshilling, thank you - yes that's the same thing. Seems like I did not go through open issues very well. 

@DevUser0491: that is off-topic, but yes, just use CodeScannerFrame's width and height.

daankennes, 2024-01-23:
#2213 fixed it for me.

closed 2024-01-23 by daankennes

ArturoTorresMartinez, 2024-03-05:
Any update on this? I tried that PR but the build throws an error

```
* What went wrong:
Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.
> A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction
   > Compilation error. See log for more details

```

sphex1987, 2024-09-13:
Tried the #2213 fix but I'm getting 
`The boolean literal does not conform to the expected type CameraConfiguration?`

Is there an updated fix for this? Thanks!