margelo/react-native-vision-camera #2431: [V3][iOS][CodeScanner] CodeScanner only captures one barcode at a time🐛

induRaju, 2024-01-25:
### What's happening?

I have built a barcode scanner using useCodeScanner but in IOS it scans only one code at a time. It works perfectly fine in Android.

### Reproduceable Code

```tsx
const device = useCameraDevice('back');
const codeScanner = useCodeScanner({
    codeTypes: ['qr', 'ean-13', 'ean-8', 'upc-e', 'code-128', 'code-39', 'pdf-417', 'data-matrix'],
    onCodeScanned: (codes, frame) => {
      if (codes && codes.length === 0) {
        return;
      }
      const barcodes = codes.filter(barcode => isValidBarcode(barcode));
      if(barcodes && barcodes.length !== 0) {
          if (barcodes.length >= 1) {
console.log(barcodes.length)
           }
}
    }
  });
 <Camera
        style={styles.preview}
        codeScanner={codeScanner}
        device={device}
        isActive={isCameraActive}
        ref={rnCameraRef}
        torch={flashMode ? 'on' : 'off'}
      />
```

### Relevant log output

```shell
I cannot attach log as it for my company but I don't see any issue in the log as well.
```

### Camera Device

```json
{
“hardwareLevel":"full",
   "hasFlash":false,
   "hasTorch":false,
   "id":"com.apple.avfoundation.avcapturedevice.built-in_video:0",
   "isMultiCam":false,
   "maxExposure":8,
   "maxZoom":95.625,
   "minExposure":-8,
   "minZoom":1,
   "name":"Back Camera",
   "neutralZoom":1,
   "physicalDevices":[
      "wide-angle-camera"
   ],
   "position":"back",
   "sensorOrientation":"landscape-right",
   "supportsFocus":true,
   "supportsLowLightBoost":false,
   "supportsRawCapture":false,
   "formats": {"autoFocusSystem":"contrast-detection",
         "fieldOfView":54.26699447631836,
         "maxFps":30,
         "maxISO":768,
         "maxZoom":153,
         "minFps":2,
         "minISO":24,
         "photoHeight":2448,
         "photoWidth":3264,
         "pixelFormats":[
            "Array"
         ],
         "supportsDepthCapture":false,
         "supportsPhotoHdr":false,
         "supportsVideoHdr":false,
         "videoHeight":144,
         "videoStabilizationModes":[
            "Array"
         ],
         "videoWidth":192
      }
}
```

### Device

iPad (7th Generation) (iOS 16.2)

### VisionCamera Version

3.7.0

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

hilkbahar, 2024-01-25:
Using this package in a production environment is a significant risk, especially considering the ongoing issues with barcode scanning. Why is it still challenging to release a stable version with reliable barcode reading capabilities?

mrousavy, 2024-01-25:
@hilkbahar I'm not sure if you're serious or not, but if you are serious: This package is running fully stable in multiple production apps. You can use Frame Processors to do a ton of magic, including fully custom QR code scanning with MLKit, ZXing or whatever library you prefer.

I built CodeScanner/`useCodeScanner` afterwards because some people found Frame Processors to be too difficult, so it is an easy to use alternative that does not require Frame Processors or Worklets.
The CodeScanner API exposes the recommended underlying native platform features, `Vision` on iOS and `MLKit` on Android.

For me, both of those APIs worked fine for scanning a simple QR code. That's what it was tested on. It also works on other types of codes.
If you need to scan multiple codes, increase performance, or whatever - you can always use a custom Frame Processor to achieve what you need.

> Why is it still challenging to release a stable version with reliable barcode reading capabilities?

This is a free library you are using to gain profit for yourself/your company. If you want something fixed, pay for it and don't complain to authors and maintainers like me who do it in their free time.

induRaju, 2024-01-25:
@mrousavy - Thank you for the response. From custom Frame processors do you mean to use vision-camera-code-scanner which uses scanBarcodes and does v3 visioncamera supports vision-camera-code-scanner?

mrousavy, 2024-01-25:
Yep, either [vision-camera-dynamsoft-barcode-reader](https://github.com/tony-xlh/vision-camera-dynamsoft-barcode-reader) by @xulihang or [vision-camera-code-scanner](https://github.com/rodgomesc/vision-camera-code-scanner) by @rodgomesc - not sure if there are more?

mrousavy, 2024-01-25:
Ah sorry there's also [vision-camera-barcode-scanner](https://github.com/mgcrea/vision-camera-barcode-scanner) which uses MLKit on Android and Vision on iOS, by @mgcrea. This looks really stable and maintained to me.

mrousavy, 2024-01-26:
I just tried it and you are right, it captures only one `ean-13` code at a time, but can capture 4 or more `qr` codes at a time.

I am not sure why this is happening, but I am using the underlying platform native APIs and there's nothing I can do. This is an Apple bug, so file a bug using their bug reporter. The related class is called `AVCaptureMetadataOutput`, this can only capture 1 ean-13 code at a time.

Maybe it will get fixed in the next iOS update :)

closed 2024-01-26 by mrousavy