margelo/react-native-vision-camera #2432: 🐛 Camera Preview becomes black screen if use it in carousel(iOS/Android)

devkawal, 2024-01-25:
### What's happening?

Hi, I am using react-native-vision-camera(3.7.1) latest version.
I have a scenario, in which I am using carousel of two slides in dashboard screen.
One slide is for scanning data through NFC(Default slide).
Second slide is for scanning data through Camera(We have to switch between these two slides using arrow icons).

While initializing the camera tab, sometimes after 3-4th time, camera becomes black. But interesting part is we are able to scan the barcode even in black screen. And in android it crashes after 4th-5th time.

Can anyone share any thoughts on this. What's wrong is happening here?

Also created a repo which is contains this issue - https://github.com/devkawal/scanner-demo

### Reproduceable Code

```tsx
Calling Camera tab
<Carousel> <View style={styles.slide}> {/* NFC Start */} <View </View> {/* NFC END */} </View> <View style={styles.slide}> {device === null ? ( <View> <Text>No Camera</Text> </View> ) : ( <View style={styles.outerScanContainer}> <View style={styles.scanContainer}> {/* Camera Start */} {cameraScreen && device !== null && ( <Camera style={styles.cameraViewStyle} codeScanner={codeScanner} device={device} isActive={isCameraInitialized} onInitialized={onCameraInitialized} /> )} </View> <Image source={IMAGES.camera_focus} style={styles.focusFrame} resizeMode="contain" /> </View> )} {/* Camera End */} </Carousel>

Initializing Function
const onCameraInitialized = useCallback(() => { if (cameraScreen) { setIsCameraInitialized(false); } else { setIsCameraInitialized(true); } }, []);

Handle slide change
const handleSlideChange = () => { if (cameraScreen) { console.log('yes'); setCameraScreen(false); } else { console.log('no'); setCameraScreen(true); } };
```

### Relevant log output

```shell
Sometimes, camera becomes black. And camera crashes in android.
```

### Camera Device

```json
2024-01-15 16:09:29.494047+0530 Scanner[1572:397492] [javascript] 'Found screens with the same name nested inside one another. Check:\n\nDashboard, Dashboard > Dashboard\n\nThis can cause confusing behavior during navigation. Consider using unique names for each screen instead.', '\n    at BaseNavigationContainer (http://10.76.0.12:8081/index.bundle//&platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.test.smartlock.testAccess-beta:128132:28)\n    at ThemeProvider (http://10.76.0.12:8081/index.bundle//&platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.test.smartlock.testAccess-beta:133037:21)\n    at NavigationContainerInner (http://10.76.0.12:8081/index.bundle//&platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.test.smartlock.testAccess-beta:132932:26)\n    at StackScreen (http://10.76.0.12:8081/index.bundle//&platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.test.smartlock.testAccess-beta:148199:24)\n    at RCTView\n    at View (http://10.76.0.12:8081/index.bundle//&platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.test.smartlock.testAccess-beta:45623:43)\n    at I18nextProvider (http://10.76.0.12:8081/index.bundle//&platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.test.smartlock.testAccess-beta:126695:20)\n    at ErrorBoundary (http://10.76.0.12:8081/index.bundle//&platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.test.smartlock.testAccess-beta:148014:36)\n    at App (http://10.76.0.12:8081/index.bundle//&platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.test.smartlock.testAccess-beta:123340:41)\n    at RCTView\n    at View (http://10.76.0.12:8081/index.bundle//&platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.test.smartlock.testAccess-beta:45623:43)\n    at RCTView\n    at View (http://10.76.0.12:8081/index.bundle//&platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.test.smartlock.testAccess-beta:45623:43)\n    at AppContainer (http://10.76.0.12:8081/index.bundle//&platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.test.smartlock.testAccess-beta:45504:36)\n    at Scanner(RootComponent) (http://10.76.0.12:8081/index.bundle//&platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.test.smartlock.testAccess-beta:83664:28)'
2024-01-15 16:09:29.542405+0530 Scanner[1572:397170] [native] VisionCamera.didSetProps(_:): Updating 14 props: [onViewReady, codeScannerOptions, cameraId, onError, onStarted, width, onInitialized, alignSelf, enableBufferCompression, height, onCodeScanned, enableFrameProcessor, onStopped, isActive]
2024-01-15 16:09:29.542454+0530 Scanner[1572:397170] [native] VisionCamera.configure(_:): configure { ... }: Waiting for lock...
2024-01-15 16:09:29.548425+0530 Scanner[1572:397659] [native] VisionCamera.configure(_:): configure { ... }: Updating CameraSession Configuration... Difference(inputChanged: true, outputsChanged: true, videoStabilizationChanged: true, orientationChanged: true, formatChanged: true, sidePropsChanged: true, torchChanged: true, zoomChanged: true, exposureChanged: true, audioSessionChanged: true)
2024-01-15 16:09:29.548525+0530 Scanner[1572:397659] [native] VisionCamera.withSessionLock(_:): Beginning CameraSession configuration...
2024-01-15 16:09:29.548560+0530 Scanner[1572:397659] [native] VisionCamera.configureDevice(configuration:): Configuring Input Device...
2024-01-15 16:09:29.548664+0530 Scanner[1572:397659] [native] VisionCamera.configureDevice(configuration:): Configuring Camera com.apple.avfoundation.avcapturedevice.built-in_video:0...
2024-01-15 16:09:29.565713+0530 Scanner[1572:397659] [native] VisionCamera.configureDevice(configuration:): Successfully configured Input Device!
2024-01-15 16:09:29.565749+0530 Scanner[1572:397659] [native] VisionCamera.configureOutputs(configuration:): Configuring Outputs...
2024-01-15 16:09:29.565849+0530 Scanner[1572:397659] [native] VisionCamera.configureOutputs(configuration:): Adding Code Scanner output...
2024-01-15 16:09:29.568934+0530 Scanner[1572:397659] [native] VisionCamera.configureOutputs(configuration:): Successfully configured all outputs!
2024-01-15 16:09:29.570237+0530 Scanner[1572:397659] [native] VisionCamera.withSessionLock(_:): Committed CameraSession configuration!
2024-01-15 16:09:29.570280+0530 Scanner[1572:397659] [native] VisionCamera.withDeviceLock(_:): Beginning CaptureDevice configuration...
2024-01-15 16:09:29.571437+0530 Scanner[1572:397659] [native] VisionCamera.withDeviceLock(_:): Committed CaptureDevice configuration!
2024-01-15 16:09:29.623342+0530 Scanner[1572:397492] [javascript] 'kas:', true
flipper: Desktop failed to provide certificates. Error from flipper desktop:
rsocket::StreamInterruptedException: connection error
2024-01-15 16:09:30.022941+0530 Scanner[1572:397659] [native] VisionCamera.onCameraStarted(): Camera started!
2024-01-15 16:09:30.023254+0530 Scanner[1572:397659] [native] VisionCamera.withDeviceLock(_:): Beginning CaptureDevice configuration...
2024-01-15 16:09:30.023355+0530 Scanner[1572:397659] [native] VisionCamera.withDeviceLock(_:): Committed CaptureDevice configuration!
2024-01-15 16:09:30.023382+0530 Scanner[1572:397659] [native] VisionCamera.onSessionInitialized(): Camera initialized!
2024-01-15 16:09:30.023768+0530 Scanner[1572:397659] [native] VisionCamera.configure(_:): Beginning AudioSession configuration...
2024-01-15 16:09:30.023816+0530 Scanner[1572:397659] [native] VisionCamera.configureAudioSession(configuration:): Configuring Audio Session...
2024-01-15 16:09:30.026987+0530 Scanner[1572:397659] [native] VisionCamera.configure(_:): Committed AudioSession configuration!
2024-01-15 16:09:32.278409+0530 Scanner[1572:397492] [javascript] -----lock found
flipper: Desktop failed to provide certificates. Error from flipper desktop:
rsocket::StreamInterruptedException: connection error
2024-01-15 16:09:32.901696+0530 Scanner[1572:397475] [native] 
flipper: Desktop failed to provide certificates. Error from flipper desktop:
rsocket::StreamInterruptedException: connection error
2024-01-15 16:09:36.662633+0530 Scanner[1572:397170] [native] VisionCamera.didSetProps(_:): Updating 14 props: [onViewReady, codeScannerOptions, cameraId, onError, onStarted, width, onInitialized, alignSelf, enableBufferCompression, height, onCodeScanned, enableFrameProcessor, onStopped, isActive]
[... cut ...]
### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-01-30:
There's multiple issues opened here where people report black screen issues. If someone pays me to work on fixing the issues I will do so

ErAmanDhiman, 2024-02-01:
@mrousavy I think this is the biggest issue Almost everyone is facing , It should be done on Priority Basis , I will buy you a coffee 

https://github.com/mrousavy/react-native-vision-camera/assets/85603231/26fbf84d-5e10-45bd-9418-bdb256842f06
[image]

mrousavy, 2024-02-01:
Thanks for the coffee @ErAmanDhiman!

mrousavy, 2024-02-01:
Working on a PR here: https://github.com/mrousavy/react-native-vision-camera/pull/2494 👀 

Still in development, so don't test yet. Long way to go, this is a really complicated issue (as you can see in the changes/Files of that PR)...

closed 2024-02-06 by mrousavy

mrousavy, 2024-02-06:
Just released [VisionCamera 3.9.0-beta.1](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v3.9.0-beta.1), which includes the [big `PersistentCaptureSession` PR](https://github.com/mrousavy/react-native-vision-camera/pull/2494) that aims to fix all the blackscreen issues.... :eyes::tada: 

You can try that new beta out yourself and let me know if you still encounter blackscreen issues. I'll continue to test on more devices in the meantime, organize the code a bit better, and then I'll tackle the preview stretching, focus, and flash issues!

If that fixes the blackscreen issues for you, please consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks! ❤️

devkawal, 2024-02-06:
Hi @mrousavy . Thanks for the update.
I installed the latest 3.9.0-beta version, I am not getting the black screen issue. But now app is crashing, when we try to mount-unmount the camera continuously for 3-4 times or  try to scan any barcode. Attaching the logs for the same.

```
392 12020-12981 CreateCaptureSession    com.scannerApp.access.beta            I  Camera #2: Successfully created CameraCaptureSession #11!
2024-02-06 20:45:21.392 12020-12878 CameraDevice-JV-2       com.scannerApp.access.beta            V  Stream 1 has been removed. Skipping buffer lost callback
2024-02-06 20:45:21.392 12020-12878 CameraDevice-JV-2       com.scannerApp.access.beta            V  Stream 2 has been removed. Skipping buffer lost callback
2024-02-06 20:45:21.392 12020-12020 Persistent...ureSession com.scannerApp.access.beta            D  Stopping repeating request...
2024-02-06 20:45:21.392 12020-12020 Persistent...ureSession com.scannerApp.access.beta            D  Configure() done! isActive: false, ID: 2, device: android.hardware.camera2.impl.CameraDeviceImpl@b6b76e1, session: android.hardware.camera2.impl.CameraCaptureSessionImpl@cc24d08
2024-02-06 20:45:21.392 12020-12020 CameraSession           com.scannerApp.access.beta            I  configure { ... }: Completed CameraSession Configuration! (isActive: true, isRunning: false)
2024-02-06 20:45:21.392 12020-12878 CameraDevice-JV-2       com.scannerApp.access.beta            V  Stream 1 has been removed. Skipping buffer lost callback
2024-02-06 20:45:21.392 12020-12020 CameraSession           com.scannerApp.access.beta            I  Preview Output destroyed!
2024-02-06 20:45:21.392 12020-12020 BufferQueueProducer     com.scannerApp.access.beta            D  [SurfaceView[com.scannerApp.access.beta/com.scannerApp.smartlock.MainActivity]#5(BLAST Consumer)5](id:2ef40000000f,api:0,p:-1,c:12020) disconnect: api -1
2024-02-06 20:45:21.392 12020-12020 BLASTBufferQueue        com.scannerApp.access.beta            D  [SurfaceView[com.scannerApp.access.beta/com.scannerApp.smartlock.MainActivity]#5](f:0,a:1) destructor()
2024-02-06 20:45:21.392 12020-12878 CameraDevice-JV-2       com.scannerApp.access.beta            V  Stream 2 has been removed. Skipping buffer lost callback
2024-02-06 20:45:21.393 12020-12020 BufferQueueConsumer     com.scannerApp.access.beta            D  [SurfaceView[com.scannerApp.access.beta/com.scannerApp.smartlock.MainActivity]#5(BLAST Consumer)5](id:2ef40000000f,api:0,p:-1,c:12020) disconnect
2024-02-06 20:45:21.393 12020-12878 CameraDevice-JV-2       com.scannerApp.access.beta            V  Stream 1 has been removed. Skipping buffer lost callback
2024-02-06 20:45:21.393 12020-12878 CameraDevice-JV-2       com.scannerApp.access.beta            V  Stream 2 has been removed. Skipping buffer lost callback
2024-02-06 20:45:21.393 12020-12878 CameraDevice-JV-2       com.scannerApp.access.beta            V  Stream 1 has been removed. Skipping buffer lost callback
2024-02-06 20:45:21.393 12020-12878 CameraDevice-JV-2       com.scannerApp.access.beta            V  Stream 2 has been removed. Skipping buffer lost callback
2024-02-06 20:45:21.393 12020-12020 CameraView              com.scannerApp.access.beta            I  Updating CameraSession...
2024-02-06 20:45:21.394 12020-12980 CameraSession           com.scannerApp.access.beta            I  configure { ... }: Waiting for lock...
2024-02-06 20:45:21.394 12020-12980 CameraSession           com.scannerApp.access.beta            I  configure { ... }: Updating CameraSession Configuration... Difference(deviceChanged=false, outputsChanged=false, sidePropsChanged=false, isActiveChanged=true)
2024-02-06 20:45:21.394 12020-12980 Persistent...ureSession com.scannerApp.access.beta            D  --> setIsActive(false)
2024-02-06 20:45:21.394 12020-12980 Persistent...ureSession com.scannerApp.access.beta            D  Configure() with isActive: false, ID: 2, device: android.hardware.camera2.impl.CameraDeviceImpl@b6b76e1, session: android.hardware.camera2.impl.CameraCaptureSessionImpl@cc24d08
2024-02-06 20:45:21.394 12020-12981 CreateCaptureSession    com.scannerApp.access.beta            I  Camera #2: CameraCaptureSession #10 has been closed.
2024-02-06 20:45:21.394 12020-12981 Persistent...ureSession com.scannerApp.access.beta            I  Session android.hardware.camera2.impl.CameraCaptureSessionImpl@65bb5d5 closed!
2024-02-06 20:45:21.394 12020-12020 CameraSession           com.scannerApp.access.beta            I  Closing CameraSession...
2024-02-06 20:45:21.395 12020-12980 Persistent...ureSession com.scannerApp.access.beta            D  Stopping repeating request...
2024-02-06 20:45:21.395 12020-12980 Persistent...ureSession com.scannerApp.access.beta            D  Configure() done! isActive: false, ID: 2, device: android.hardware.camera2.impl.CameraDeviceImpl@b6b76e1, session: android.hardware.camera2.impl.CameraCaptureSessionImpl@cc24d08
2024-02-06 20:45:21.395 12020-12980 CameraSession           com.scannerApp.access.beta            I  configure { ... }: Completed CameraSession Configuration! (isActive: false, isRunning: false)
2024-02-06 20:45:21.395 12020-12020 CameraSession           com.scannerApp.access.beta            I  Destroying session..
2024-02-06 20:45:21.450 12020-14555 BpBinder                com.scannerApp.access.beta            I  onLastStrongRef automatically unlinking death recipients: 
[... cut ...]
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:257)
                                                                                                    	at android.os.Looper.loop(Looper.java:368)
                                                                                                    	at android.os.HandlerThread.run(HandlerThread.java:67)
                                                                                                    	at com.android.server.ServiceThread.run(ServiceThread.java:46)
```

mrousavy, 2024-02-06:
@devkawal does this also happen when you are not using a code-scanner? Try only enabling `photo={true}`, no video, no frame processor and no code-scanner and see if that happens.

devkawal, 2024-02-06:
No, if commented the codeScanner, then on mounting-unmounting. It is not crashing.

```
<Camera
                style={styles.cameraViewStyle}
                // codeScanner={codeScanner}
                photo={true}
                device={device}
                isActive={isActive}
                onInitialized={onCameraInitialized}
                onError={onError}
                enableZoomGesture={true}
                exposure={0}
              />
```

mrousavy, 2024-02-06:
Okay, so it's definitely related to the CodeScanner then. Maybe I'll swap that out to use ZXing 🤔

mrousavy, 2024-02-06:
In other words; Looks like it is a bug in Google's MLKit CodeScanner, not in VisionCamera.

devkawal, 2024-02-06:
So, are we expecting changes of scanner in upcoming release?

mrousavy, 2024-02-06:
lol, no. Not working on that for now unless someone sponsors it.
First I'm focusing on preview issues and stability, as well as flash and focus

ErAmanDhiman, 2024-02-07:
> So, are we expecting changes of scanner in upcoming release?

I think this has to be solve in next upcoming release because no library is supporting ml kit, and many devs are waiting , Thanks

podger, 2024-02-14:
> In other words; Looks like it is a bug in Google's MLKit CodeScanner, not in VisionCamera.

I just opened an issue on 3.9.0-Beta.3 on this exact same issue: #2554 

I think it's more related to how the camera or our codes handle the unmount after a code scanned (usually done in the codeScanner callback). The code scanner works without crashes if we do not try to dismiss the camera, which I would say It's not  real life use case.