margelo/react-native-vision-camera #2438: Failed to start Recording🐛

Josiphiah, 2024-01-25:
### What's happening?

i want to start recording a video but it fails with an error

### Reproduceable Code

```tsx
import React, { useState, useEffect, useRef } from 'react';
import { View, Text, SafeAreaView, ActivityIndicator, StyleSheet, ScrollView, Image, TouchableOpacity } from 'react-native';
import { Camera, useCameraDevice, useCameraPermission, useMicrophonePermission } from 'react-native-vision-camera';
import { height, styles as sx, width } from '../helpers/styles';
import { logo } from '../assets/exports';

export default function FaceScanComponent() {
  const [cameraPermission, setCameraPermission] = useState<'pending' | 'authorized' | 'denied'>('pending');
  const { hasPermission: cameraHasPermission, requestPermission: requestCameraPermission } = useCameraPermission();
  const { hasPermission: microphoneHasPermission, requestPermission: requestMicrophonePermission } = useMicrophonePermission();
  const [isRecording, setIsRecording] = useState(false);

  useEffect(() => {
    const requestPermissions = async () => {
      const cameraStatus = await requestCameraPermission();
      const microphoneStatus = await requestMicrophonePermission();

      if (cameraStatus === true && microphoneStatus === true) {
        setCameraPermission('authorized');
      } else {
        setCameraPermission('denied');
      }
    };

    requestPermissions();
  }, [requestCameraPermission, requestMicrophonePermission]);

  const device = useCameraDevice('front');
  const camera = useRef<Camera>(null)

  

  if (device == null) return <NoCameraDeviceError />;

  const startRecording = async () => {
    if (camera.current) {
      camera.current.startRecording({
        onRecordingFinished: async (video) => {
          const path = video.path;
        },
        videoBitRate: 'low',
        videoCodec: 'h265',
        onRecordingError: (error) => console.error(error),
      });
      setIsRecording(true);
    }
  };
  
  

  const stopRecording = () => {
    if (camera.current) {
      camera.current.stopRecording();
      setIsRecording(false);
    }
  };

  function renderDetectorContent() {
    if (device && cameraPermission === 'authorized') {
      return (
        <>
          <Camera ref={camera} style={styles.camera} device={device} isActive={true} video={true} audio={true} />
          <View style={styles.buttonContainer}>
            <TouchableOpacity onPress={isRecording ? stopRecording : startRecording} style={styles.recordButton}>
              <Text style={styles.recordButtonText}>{isRecording ? 'Stop Recording' : 'Start Recording'}</Text>
            </TouchableOpacity>
          </View>
        </>
      );
    } else if (cameraPermission === 'denied') {
      return <Text style={styles.errorText}>No camera permission.</Text>;
    }
    return <ActivityIndicator style={styles.loading} size="large" color="#1C6758" />;
  }

  return (
    <SafeAreaView >
      <ScrollView contentContainerStyle={[styles.body]}>
      <Image
        source={logo}
        style={styles.logo}
      />
      
      <View style={styles.container}>
      <SafeAreaView style={styles.header}>
        <Text style={styles.headerText}>React Native Image Detector</Text>
      </SafeAreaView>

      <View style={styles.caption}>
        <Text style={styles.captionText}>Welcome To React-Native-Vision-Camera Tutorial</Text>
      </View>

      {renderDetectorContent()}

      <View style={styles.caption}>
        <Text style={styles.captionText}>Welcome To React-Native-Vision-Camera Tutorial</Text>
      </View>

    </View>

     
      
</ScrollView>
    </SafeAreaView>
  );
}
```

### Relevant log output

```shell
ERROR  {"cause": {"message": "prepare failed.", "stacktrace": "java.io.IOException: prepare failed.
	at android.media.MediaRecorder._prepare(Native Method)
	at android.media.MediaRecorder.prepare(MediaRecorder.java:1377)
	at com.mrousavy.camera.core.RecordingSession.start(RecordingSession.kt:92)
	at com.mrousavy.camera.core.CameraSession.startRecording(CameraSession.kt:647)
	at com.mrousavy.camera.CameraView_RecordVideoKt.startRecording(CameraView+RecordVideo.kt:36)
	at com.mrousavy.camera.CameraViewModule$startRecording$1.invokeSuspend(CameraViewModule.kt:91)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:584)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:793)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:697)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:684)
"}, "code": "capture/unknown", "message": "An unknown error occurred while trying to start a video recording! prepare failed.", "userInfo": null}
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "full",
  "maxZoom": 10,
  "minZoom": 1,
  "maxExposure": 24,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": false,
  "minFocusDistance": 0,
  "minExposure": -24,
  "name": "FRONT (1)",
  "hasFlash": false,
  "hasTorch": false,
  "position": "front",
  "id": "1"
}
```

### Device

Xiaomi Redmi Note 12

### VisionCamera Version

3.8.2

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-01-26:
Hey - does it work if you remove the `videoBitRate` and `videoCodec` options?

Josiphiah, 2024-01-26:
no same error i actually added the videoBitRate and videoCodec options to see if it would work

mrousavy, 2024-01-26:
Hm. Can you please upload the full adb logcat logs, not just the error? I wanna see what happened before that.

Josiphiah, 2024-01-26:
i'm not exactly sure how to get what you asked for

mrousavy, 2024-01-26:
Run the app in Android Studio, there's a logcat window. Also the issue template tells you how to do that

mrousavy, 2024-01-30:
Hey - did you try to run the app in Android studio yet so I can see the logs?

MuhammadFaisal215, 2024-02-01:
@mrousavy I have the same issue. When i hit the startRecording function my app suddenly crashes. I have check and this issue came on android 11. When i use android 13 it works fine. 

Is there any solution that i can run this package on android 10,11,12,13 and 14.

mrousavy, 2024-03-19:
Hey! I just released a new V4 beta ([v4.0.0-beta.7](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0-beta.7)) where I fixed a bunch of issues! 💪🚀 Can you test that and see if that fixes the issue for you? 😅 

```
yarn add react-native-vision-camera@beta
```

You might need to increase the `compileSdk` to the newest Android SDK (34) if you are behind.

aharwood9, 2024-03-19:
@mrousavy I was also getting a "java.io.IOException: prepare failed" error but on OnePlus Nord CE 2 when I tried to record. I have just upgraded to `beta.8` and the issue has now been resolved 🎉 thank you very much for everything.

mrousavy, 2024-03-21:
Awesome, great to hear!

mrousavy, 2024-03-25:
Great to hear that V4 now works perfectly! If you appreciate my work, please consider [💖 sponsoring me on GitHub 💖](http://github.com/sponsors/mrousavy/) so I can keep maintaining this library, fixing bugs and building new features! :)

mrousavy, 2024-04-22:
Hey - I think this issue has been fixed in [VisionCamera 4.0.0](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0). 🥳 

Please try V4 and let me know if you still experience this issue;
- if not, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support the development of VisionCamera and thank me for my time spent on fixing bugs and building new features.
- if you still see this issue, please comment and we can re-open this. But please update your native logs with the native (Xcode/Android Studio) logs from running VisionCamera V4 so I can investigate this.

closed 2024-04-22 by mrousavy