margelo/react-native-vision-camera #2440: Record & display with delay❓

vanjaoljaca, 2024-01-26:
### Question

I'm trying to record video and show the delayed video on the screen (lets say about 5 seconds). I'm not sure if this is something that already indirectly exists in this library as a feature, or if I should go down a custom path.

For a custom solution I'm thinking of just recording ~2second clips in 1by1 and then playing back with a delay. Does this seem reasonable? I worry there will be too much delay between loading/playing clips (about to create a test app)

### What I tried

looked for similar questions to do with delay or buffer

### VisionCamera Version

latest

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2024-01-26:
Hey - sorry I don't understand the question. What do you mean by delay?

vanjaoljaca, 2024-01-26:
as in what the camera sees is delayed by 5 seconds before shown on screen. currently the ui view shows live view of what the camera sees, or i can record 5 seconds of video, stop recording then play from the start... but then its not recording new content. does that make sense?

mrousavy, 2024-01-26:
Ah, yea I get that - but that's nothing a Camera library should support imo.

You could write to file and display later, or eg do it in a Frame Processor somehow.
It's def possible but requires a bit od native code

closed 2024-01-26 by mrousavy