margelo/react-native-vision-camera #2442: 🐛 Crash on Android using code scanner

oscar-b, 2024-01-26:
### What's happening?

After reading barcodes using `useCodeScanner` the app crashes on Android.

Tried with both default `pixelFormat` and `yuv` as you suggested.

### Reproduceable Code

```tsx
<Camera
	ref={camera}
	style={StyleSheet.absoluteFill}
	device={device}
	isActive={show}
	codeScanner={codeScanner}
	onError={onError}
	pixelFormat="yuv"
/>
```

### Relevant log output

```shell
01-26 10:46:26.355 31206 31206 I PreviewView: Creating PreviewView...
01-26 10:46:26.356 31206 31206 I PreviewView: PreviewView is 0x0, rendering 720x1473 content. Resizing to: 0x0 (COVER)
01-26 10:46:26.358 31206 31206 I CameraView: Updating CameraSession...
01-26 10:46:26.360 31206  1041 I CameraSession: configure { ... }: Waiting for lock...
01-26 10:46:26.362 31206  1041 I CameraSession: configure { ... }: Updating CameraSession Configuration... Difference(deviceChanged=true, outputsChanged=true, sidePropsChanged=true, isActiveChanged=true)
01-26 10:46:26.362 31206  1041 I CameraSession: Need to rebuild CameraDevice and CameraCaptureSession...
01-26 10:46:26.362 31206  1041 I CameraSession: CameraDevice and CameraCaptureSession is torn down but Camera is not active, skipping update...
01-26 10:46:26.362 31206  1041 I CameraSession: Successfully updated CameraSession Configuration! isActive: false
01-26 10:46:26.363 31206  1041 I CameraView: invokeOnInitialized()
01-26 10:46:26.486 31206 31206 E unknown:ReactModalHost: Creating new dialog from context: eu.company.my_app.dev.MainActivity@86dfd26@141425958
01-26 10:46:26.532 31206 31206 E unknown:ReactModalHost: Updating existing dialog with context: eu.company.my_app.dev.MainActivity@86dfd26@141425958
01-26 10:46:26.593 31206 31206 I CameraView: Updating CameraSession...
01-26 10:46:26.594 31206  1041 I CameraSession: configure { ... }: Waiting for lock...
01-26 10:46:26.594 31206  1041 I CameraSession: configure { ... }: Updating CameraSession Configuration... Difference(deviceChanged=true, outputsChanged=true, sidePropsChanged=true, isActiveChanged=true)
01-26 10:46:26.594 31206  1041 I CameraSession: Need to rebuild CameraDevice and CameraCaptureSession...
01-26 10:46:26.594 31206  1041 I CameraSession: Need to rebuild CameraDevice and CameraCaptureSession...
01-26 10:46:26.594 31206  1041 I CameraSession: Configuring Camera #0...
01-26 10:46:26.595 31206  1041 I CameraManager: Camera #0: Opening...
01-26 10:46:26.599 31206 31206 I PreviewView: PreviewView is 720x1078, rendering 720x1473 content. Resizing to: 720x1473 (COVER)
01-26 10:46:26.618 31206  1041 I CameraManager: Camera #0: Opened!
01-26 10:46:26.623 31206  1041 I CameraSession: Successfully configured Camera #0!
01-26 10:46:26.624 31206  1041 I CameraSession: Destroying previous outputs...
01-26 10:46:26.625 31206   992 I CameraDevices: Camera #0 is now unavailable.
01-26 10:46:26.630 31206  1041 I CameraSession: Creating outputs for Camera #0...
01-26 10:46:26.634 31206 31206 I CameraSession: PreviewView Surface created! Surface(name=null)/@0x6d006ae
01-26 10:46:26.634 31206  1041 I CameraSession: Adding 1280x720 CodeScanner Output in YUV_420_888...
01-26 10:46:26.634 31206 31206 I CameraSession: Setting Preview Output...
01-26 10:46:26.635 31206 31206 I CameraSession: PreviewView Surface updated! Surface(name=null)/@0x6d006ae 720 x 1473
01-26 10:46:26.679 31206 31206 I PreviewView: PreviewView is 720x1132, rendering 720x1473 content. Resizing to: 720x1473 (COVER)
01-26 10:46:26.680 31206  1245 I DynamiteModule: Considering local module com.google.mlkit.dynamite.barcode:10000 and remote module com.google.mlkit.dynamite.barcode:0
01-26 10:46:26.681 31206  1245 I DynamiteModule: Selected local version of com.google.mlkit.dynamite.barcode
01-26 10:46:26.691 31206  1041 I CreateCaptureSession: Camera #0: Creating Capture Session #1... (Hardware Level: 3 | Outputs: [VIDEO (1280 x 720 CodeScanner for [QR, EAN_13, CODE_128, DATA_MATRIX] (35))])
01-26 10:46:26.692 31206  1041 I CreateCaptureSession: Using new API (>=28)
01-26 10:46:26.764 31206 31230 D TransportRuntime.SQLiteEventStore: Storing event with priority=DEFAULT, name=FIREBASE_ML_SDK for destination cct
01-26 10:46:26.773 31206 31230 D TransportRuntime.JobInfoScheduler: Upload for context TransportContext(cct, DEFAULT, MSRodHRwczovL2ZpcmViYXNlbG9nZ2luZy5nb29nbGVhcGlzLmNvbS92MGNjL2xvZy9iYXRjaD9mb3JtYXQ9anNvbl9wcm90bzNc) is already scheduled. Returning...
01-26 10:46:26.781 31206 31230 D TransportRuntime.SQLiteEventStore: Storing event with priority=VERY_LOW, name=FIREBASE_ML_SDK for destination cct
01-26 10:46:26.787 31206 31230 D TransportRuntime.JobInfoScheduler: Upload for context TransportContext(cct, VERY_LOW, MSRodHRwczovL2ZpcmViYXNlbG9nZ2luZy5nb29nbGVhcGlzLmNvbS92MGNjL2xvZy9iYXRjaD9mb3JtYXQ9anNvbl9wcm90bzNc) is already scheduled. Returning...
01-26 10:46:26.849 31206  1041 I CameraSession: configure { ... }: Waiting for lock...
01-26 10:46:26.851 31206  1041 I CreateCaptureSession: Camera #0: Successfully created CameraCaptureSession #1!
01-26 10:46:26.851 31206  1041 I CameraSession: Successfully configured Session with 1 outputs for Camera #0!
01-26 10:46:26.856 31206  1041 I CameraView: invokeOnStarted()
01-26 10:46:26.857 31206  1041 I CameraSession: Successfully updated CameraSession Configuration! isActive: true
01-26 10:46:26.857 31206  1041 I CameraView: invokeOnInitialized()
01-26 10:46:26.859 31206  1041 I CameraSession: configure { ... }: Updating CameraSession Configuration... Difference(deviceChanged=false, outputsChanged=true, sidePropsChanged=true, isActiveChanged=false)
01-26 10:46:26.859 31206  1041 I CameraSession: Destroying previous outputs...
01-26 10:46:26.859 31206  1041 I SurfaceOutput: Closing BarcodeScanner..
01-26 10:46:26.860 31206  1041 I CameraView: invokeOnStopped()
[... 99 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Android Motorola Defy (Android 11)

### VisionCamera Version

3.8.2

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-01-30:
Hey - seems like the logs just cut off, there is no tombstone there? Can you post the full logs

oscar-b, 2024-01-30:
> Hey - seems like the logs just cut off, there is no tombstone there? Can you post the full logs

I'm on vacation without access to the test device right now, but I'm quite sure that's all there was in the logs. The logs are from `adb logcat` so maybe doesn't have it all?

Devyadav3001, 2024-01-31:
i have same problem in my app one time scan 2 time then crash the android app plz solve my problem

xiongxiongjiang, 2024-01-31:
> i have same problem in my app one time scan 2 time then crash the android app plz solve my problem

+1 same here

mrousavy, 2024-01-31:
> plz solve my problem

@Devyadav3001 lol, plz pay me to solve ur problem

mrousavy, 2024-01-31:
> > Hey - seems like the logs just cut off, there is no tombstone there? Can you post the full logs
> 
> I'm on vacation without access to the test device right now, but I'm quite sure that's all there was in the logs. The logs are from `adb logcat` so maybe doesn't have it all?

Okay yea no I think there should've been something afterwards, because the tombstone isn't there it's just the message _that something_ has crashed.

You see:

```
--------- beginning of crash
```

there should also be a `end of crash`.

Anyways, have a nice rest of your vacation - let me know if you find the logs when you're back :)

tatiana-lonestone, 2024-02-02:
Hello,

I have the same kind of error : after reading barcodes using useCodeScanner, the app crashes on Android randomly.

But my logcat is different, so maybe it can help ?

````

2024-02-02 17:24:41.393 14562-16177 y-package          com.my-package                W  Accessing hidden method Lsun/misc/Unsafe;->getObject(Ljava/lang/Object;J)Ljava/lang/Object; (greylist, linking, allowed)
2024-02-02 17:24:41.394 14562-16177 y-package          com.my-package                W  Accessing hidden method Lsun/misc/Unsafe;->putObject(Ljava/lang/Object;JLjava/lang/Object;)V (greylist, linking, allowed)
2024-02-02 17:24:41.509 14562-16178 y-package          com.my-package                W  Accessing hidden method Lsun/misc/Unsafe;->getObject(Ljava/lang/Object;J)Ljava/lang/Object; (greylist, linking, allowed)
2024-02-02 17:24:41.509 14562-16178 y-package          com.my-package                W  Accessing hidden method Lsun/misc/Unsafe;->putObject(Ljava/lang/Object;JLjava/lang/Object;)V (greylist, linking, allowed)
2024-02-02 17:24:41.531 14562-14562 CameraSession           com.my-package                I  PreviewView Surface destroyed! Surface(name=null)/@0x77e9c8c
2024-02-02 17:24:41.531 14562-14562 CameraSession           com.my-package                I  Destroying Preview Output...
2024-02-02 17:24:41.534 14562-14562 CameraSession           com.my-package                I  configure { ... }: Waiting for lock...
2024-02-02 17:24:41.534 14562-14562 CameraSession           com.my-package                I  configure { ... }: Updating CameraSession Configuration... Difference(deviceChanged=false, outputsChanged=true, sidePropsChanged=true, isActiveChanged=false)
2024-02-02 17:24:41.534 14562-14562 CameraSession           com.my-package                I  Destroying previous outputs...
2024-02-02 17:24:41.534 14562-14562 SurfaceOutput           com.my-package                I  Closing BarcodeScanner..
2024-02-02 17:24:41.536 14562-14562 CameraView              com.my-package                I  invokeOnStopped()
2024-02-02 17:24:41.543 14562-14562 CameraSession           com.my-package                I  Creating outputs for Camera #0...
2024-02-02 17:24:41.548 14562-14562 CameraSession           com.my-package                I  Adding 1280x720 CodeScanner Output in YUV_420_888...
2024-02-02 17:24:41.554 14562-14562 CreateCaptureSession    com.my-package                I  Camera #0: Creating Capture Session #3... (Hardware Level: 3 | Outputs: [VIDEO (1280 x 720 CodeScanner for [CODE_128] (35))])
2024-02-02 17:24:41.555 14562-14562 CreateCaptureSession    com.my-package                I  Using new API (>=28)
2024-02-02 17:24:41.566 14562-16008 BufferQueueProducer     com.my-package                E  [ImageReader-1280x720f23m2-14562-1] queueBuffer: BufferQueue has been abandoned
2024-02-02 17:24:41.608 14562-16008 BufferQueueProducer     com.my-package                E  [ImageReader-1280x720f23m2-14562-1] queueBuffer: BufferQueue has been abandoned
2024-02-02 17:24:41.645 14562-16008 BufferQueueProducer     com.my-package                E  [ImageReader-1280x720f23m2-14562-1] queueBuffer: BufferQueue has been abandoned
2024-02-02 17:24:41.742 14562-16186 BufferQueueProducer     com.my-package                E  [ImageReader-1280x720f23m2-14562-1] queueBuffer: BufferQueue has been abandoned
2024-02-02 17:24:41.757 14562-16179 libc                    com.my-package                A  Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x7b673b59c0 in tid 16179 (pool-11-thread-), pid 14562 (y-packages)
2024-02-02 17:24:41.814 14562-16099 CreateCaptureSession    com.my-package                I  Camera #0: Successfully created CameraCaptureSession #3!
2024-02-02 17:24:41.815 14562-16099 CreateCaptureSession    com.my-package                I  Camera #0: CameraCaptureSession #2 has been closed.
2024-02-02 17:24:41.815 14562-14562 CameraSession           com.my-package                I  Successfully configured Session with 1 outputs for Camera #0!
2024-02-02 17:24:41.815 14562-16099 CameraSession           com.my-package                I  Camera Session android.hardware.camera2.impl.CameraCaptureSessionImpl@8a05784 has been closed.
2024-02-02 17:24:41.821 14562-14562 CameraView              com.my-package                I  invokeOnStarted()
2024-02-02 17:24:41.822 14562-14562 CameraSession           com.my-package                I  Successfully updated CameraSession Configuration! isActive: true
2024-02-02 17:24:41.822 14562-14562 CameraSession           com.my-package                I  Preview Output destroyed!
2024-02-02 17:24:41.875 14562-14562 CameraView              com.my-package                I  Updating CameraSession...
2024-02-02 17:24:41.876 14562-16099 CameraSession           com.my-package                I  configure { ... }: Waiting for lock...
2024-02-02 17:24:41.876 14562-16099 CameraSession           com.my-package                I  configure { ... }: Updating CameraSession Configuration... Difference(deviceChanged=true, outputsChanged=true, sidePropsChanged=true, isActiveChanged=true)
2024-02-02 17:24:41.876 14562-16099 CameraSession           com.my-package                I  isActive is false, skipping CameraDevice configuration.
2024-02-02 17:24:41.876 14562-16099 CameraSession           com.my-package                I  isActive is false, skipping CameraCaptureSession configuration.
2024-02-02 17:24:41.876 14562-16099 CameraView              com.my-package                I  invokeOnStopped()
[... cut ...]
2024-02-02 17:24:42.365  1615-1778  ActivityManager         system_server                        I  Showing crash dialog for package com.my-package u0
2024-02-02 17:24:42.487  1615-2524  InputDispatcher         system_server                        W  channel '4f3d0f com.my-package/com.my-package.MainActivity (server)' ~ Consumer closed input channel or an error occurred.  events=0x9
---------------------------- PROCESS ENDED (14562) for package com.my-package ----------------------------
````

Vision Camera : 3.8.2
React native : 0.73.3

thib-info, 2024-02-04:
Hello there ! I'm also using the package for my RN app, and I'm facing the same issue on my Android when trying to scan a QR code. 
On IOS, everything works as expected, but on Android the app crashes sadly.

It looks like the issue is more complicated than expected... @oscar-b the commit you've mentioned (https://github.com/mrousavy/react-native-vision-camera/pull/2494), is it almost finished or not at all ? 

In the documentation, the usage of MLKit is also mentioned but not really how to install it. I've read the official documentation of Google, but just to make sure I'm not missing a dependencies or something else, I'm asking you if it could come from there ? 

Thank you for your time all !

closed 2024-02-06 by mrousavy

mrousavy, 2024-02-06:
Just released [VisionCamera 3.9.0-beta.1](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v3.9.0-beta.1), which includes the [big `PersistentCaptureSession` PR](https://github.com/mrousavy/react-native-vision-camera/pull/2494) that aims to fix all the blackscreen issues.... :eyes::tada: 

You can try that new beta out yourself and let me know if you still encounter blackscreen issues. I'll continue to test on more devices in the meantime, organize the code a bit better, and then I'll tackle the preview stretching, focus, and flash issues!

If that fixes the blackscreen issues for you, please consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks! ❤️

tatiana-lonestone, 2024-02-07:
Hello !

Thanks for all your works on it @mrousavy !

I tried the new beta version, but it still sometimes crashes when I scan a barcode.

This error is the same as before for me in the logcat: 

````
Closing BarcodeScanner..
CameraSession closed!
Camera #0 is now available.
 Chor uploadMDM JANK_TYPE_ONCE mViewTitle = com.my-app/com.my-app.MainActivity--- jank level = 2
        A        #35 pc 0000000000149b0e  [anon:dalvik-classes16.dex extracted in memory from /data/app/com.my-app-nG2pu9xR6uaJUmO6BsxkAg==/base.apk!classes16.dex] (com.google.android.gms.internal.mlkit_vision_barcode.zza.zzb+14)
        A        #38 pc 000000000016ba5a  [anon:dalvik-classes16.dex extracted in memory from /data/app/com.my-app-nG2pu9xR6uaJUmO6BsxkAg==/base.apk!classes16.dex] (com.google.android.gms.internal.mlkit_vision_barcode.zzvt.zzd+22)
        A        #41 pc 00000000002419fe  [anon:dalvik-classes16.dex extracted in memory from /data/app/com.my-app-nG2pu9xR6uaJUmO6BsxkAg==/base.apk!classes16.dex] (com.google.mlkit.vision.barcode.internal.zzn.zza+206)
        A        #44 pc 0000000000240d8c  [anon:dalvik-classes16.dex extracted in memory from /data/app/com.my-app-nG2pu9xR6uaJUmO6BsxkAg==/base.apk!classes16.dex] (com.google.mlkit.vision.barcode.internal.zzk.zze+24)
        A        #47 pc 0000000000240d58  [anon:dalvik-classes16.dex extracted in memory from /data/app/com.my-app-nG2pu9xR6uaJUmO6BsxkAg==/base.apk!classes16.dex] (com.google.mlkit.vision.barcode.internal.zzk.run+4)
        A        #50 pc 00000000002443d2  [anon:dalvik-classes16.dex extracted in memory from /data/app/com.my-app-nG2pu9xR6uaJUmO6BsxkAg==/base.apk!classes16.dex] (com.google.mlkit.vision.common.internal.MobileVisionBase.zza+22)
        A        #53 pc 00000000002448c8  [anon:dalvik-classes16.dex extracted in memory from /data/app/com.my-app-nG2pu9xR6uaJUmO6BsxkAg==/base.apk!classes16.dex] (com.google.mlkit.vision.common.internal.zza.call+8)
        A        #56 pc 000000000023a26e  [anon:dalvik-classes16.dex extracted in memory from /data/app/com.my-app-nG2pu9xR6uaJUmO6BsxkAg==/base.apk!classes16.dex] (com.google.mlkit.common.sdkinternal.ModelResource.zza+74)
        A        #59 pc 000000000023edc8  [anon:dalvik-classes16.dex extracted in memory from /data/app/com.my-app-nG2pu9xR6uaJUmO6BsxkAg==/base.apk!classes16.dex] (com.google.mlkit.common.sdkinternal.zzn.run+20)
        A        #62 pc 000000000023ef98  [anon:dalvik-classes16.dex extracted in memory from /data/app/com.my-app-nG2pu9xR6uaJUmO6BsxkAg==/base.apk!classes16.dex] (com.google.mlkit.common.sdkinternal.zzt.run+20)
        A        #65 pc 0000000000239fa6  [anon:dalvik-classes16.dex extracted in memory from /data/app/com.my-app-nG2pu9xR6uaJUmO6BsxkAg==/base.apk!classes16.dex] (com.google.mlkit.common.sdkinternal.MlKitThreadPool.zze+26)
        A        #68 pc 0000000000239f48  [anon:dalvik-classes16.dex extracted in memory from /data/app/com.my-app-nG2pu9xR6uaJUmO6BsxkAg==/base.apk!classes16.dex] (com.google.mlkit.common.sdkinternal.MlKitThreadPool.zzc+16)
        A        #71 pc 000000000023ecb0  [anon:dalvik-classes16.dex extracted in memory from /data/app/com.my-app-nG2pu9xR6uaJUmO6BsxkAg==/base.apk!classes16.dex] (com.google.mlkit.common.sdkinternal.zzk.run+4)
        A        #80 pc 0000000000239f74  [anon:dalvik-classes16.dex extracted in memory from /data/app/com.my-app-nG2pu9xR6uaJUmO6BsxkAg==/base.apk!classes16.dex] (com.google.mlkit.common.sdkinternal.MlKitThreadPool.zzd+20)
        A        #83 pc 000000000023ec34  [anon:dalvik-classes16.dex extracted in memory from /data/app/com.my-app-nG2pu9xR6uaJUmO6BsxkAg==/base.apk!classes16.dex] (com.google.mlkit.common.sdkinternal.zzi.run+4)
unregisterListenerImpl callingApp: com.my-app,callingPid:20158,callingUid:10832

````

I have some screenshot for logcat with and without the error: 

[image]

[image]