margelo/react-native-vision-camera #2443: 🐛 [V3][Android]When starting recording,App clash on the Huawei device

jamesbayy, 2024-01-26:
### What's happening?

When starting recording, App Clash on the Huawei device。But it runs normally on other Android devices

### Reproduceable Code

```tsx
export default function Page() {
    const device = useCameraDevice('back')
    const camera = useRef<Camera>(null)
    const [results, setResults] = useState<string>()
    const startRecording = () => {
        camera.current!.startRecording({
            onRecordingFinished: (video) => {
                setResults(video.path)
            },
            onRecordingError: (error) => console.error(error),
        })
    }
    const stopRecording = async () => {
        await camera.current!.stopRecording()
    }
    return (
        <View style={styles.container}>
            <Camera
                style={StyleSheet.absoluteFill}
                device={device!}
                ref={camera}
                video={true}
                audio={false}
                isActive={true}
            ></Camera>

            <View
                style={{
                    position: 'absolute',
                    width: '100%',
                    bottom: 10,
                    display: 'flex',
                    flexDirection: 'row',
                    justifyContent: 'space-around',
                }}
            >
                <Pressable
                    style={{
                        width: 50,
                        height: 50,
                        margin: 3,
                        backgroundColor: 'green',
                    }}
                    onPress={() => {
                        startRecording()
                    }}
                >
                    <Text>start</Text>
                </Pressable>
                <Pressable
                    style={{
                        width: 50,
                        height: 50,
                        margin: 3,
                        backgroundColor: '#333',
                    }}
                    onPress={() => {
                        stopRecording()
                    }}
                >
                    <Text>stop</Text>
                </Pressable>
                <Pressable
[... 128 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

HUAWEI nova 9

### VisionCamera Version

3.8.2

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

chege4179, 2024-01-28:
Hello @jamesbayy 
Did you find what was causing this error and how to fix it ?
I have the same issue

jamesbayy, 2024-01-29:
> Hello @jamesbayy Did you find what was causing this error and how to fix it ? I have the same issue

 haven't found a solution yet

mrousavy, 2024-01-30:
Just to provide more insights on those logs:

```
18:38:35.691 RecordingSession        I  Starting RecordingSession..
18:38:35.731 MediaRecorder           I  start
18:38:35.731 HwDeviceMangerFactory   I  Create actual factory for mdm part.
18:38:35.732 TransactionSponsor      I  init device policy service.
18:38:35.740 HwMediaRecorderImpl     I  HwMediaRecorderImpl
18:38:35.740 HwMediaRecorderImpl     I  sendStateChangedIntent, state=3
18:38:35.742 CameraSession           I  Updating Video Outputs...
18:38:35.742 VideoPipeline           I  Removing FrameProcessor Output...
18:38:35.742 VideoPipeline           I  Setting 3072 x 3072 RecordingSession Output...
18:38:35.775 VideoPipeline           I  Rendering to RecordingSession..
18:38:35.775 OpenGLRenderer          I  Creating Window Surface...
18:38:35.805 VideoPipeline           I  Rendering to RecordingSession..
18:38:35.845 VideoPipeline           I  Rendering to RecordingSession..
18:38:35.855 RecordingSession        E  MediaRecorder Error: 268435556 (-1007)
18:38:35.855 RecordingSession        I  Stopping RecordingSession..
18:38:35.855 MediaRecorder           I  stop
18:38:35.876 MediaRecorder           E  stop failed: -1007
18:38:35.877 AndroidRuntime          D  Shutting down VM
18:38:35.881 AndroidRuntime          E  FATAL EXCEPTION: main
```

You start the reocrding, then immediately we get this error in the logs:

```
18:38:35.855 RecordingSession        E  MediaRecorder Error: 268435556 (-1007)
```

And it fails. The `stop failed` error is a bit misleading, the actual error is the one above.

MuhammadFaisal215, 2024-02-01:
@mrousavy I have the same issue. When i hit the startRecording function my app suddenly crashes. I have check and this issue came on android 11. When i use android 13 it works fine.

Is there any solution that i can run this package on android 10,11,12,13 and 14.

mrousavy, 2024-03-19:
Hey! I just released a new V4 beta ([v4.0.0-beta.7](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0-beta.7)) where I fixed a bunch of issues! 💪🚀 Can you test that and see if that fixes the issue for you? 😅 

```
yarn add react-native-vision-camera@beta
```

You might need to increase the `compileSdk` to the newest Android SDK (34) if you are behind.

mrousavy, 2024-03-25:
If latest V4 fixes this bug, please consider [💖 sponsoring me on GitHub 💖](http://github.com/sponsors/mrousavy/) so I can keep maintaining this library, fixing bugs and building new features! :)

closed 2024-03-25 by mrousavy

dipankar4765, 2024-08-02:
Hi any update on this I am facing this issue with version 3.9.1 and even 4.3.1, also it happens frequently on any device. 
**compileSDK** 34 
**react-native** 0.71.14