margelo/react-native-vision-camera #2447: 🐛 image distorded at startup and a bit zoomed in

gdebre, 2024-01-28:
### What's happening?

### First i'm sorry i didn't try to reproduce the issue in the VisionCamera Example app but I've never managed to make the VisionCamera Example app run, i've always had build issues !

I have this weird bug where when I open the camera page, the image feedback of my back camera is completly distorted, here is an image :

[image]

Now this problem is completly temporary because if i then flip the camera twice (back=>front=>back) it disappears :

[image]

Note that it also disappears if i leave the app and open it back imedialtly.

==============================================================

Also, another problem I have which my be linked to this one (i'm not 100% sure) is that the camera is by default way more zoomed in  that my default OS camera. 

Here is the zoom with react-native-vision-camera : 

[image]

and here is the default OS camera zoom ;
[image]

Note that this also happens if I use  "<Camera resizeMode={"contain"} ..../>

### Reproduceable Code

```tsx
import React, { useRef, useState } from "react";
import { TouchableOpacity, View } from "react-native";
import { useCameraDevice, Camera, useCameraPermission } from "react-native-vision-camera";

import { useIsFocused } from "@react-navigation/native";
import { useAppState } from "@react-native-community/hooks";

export default function CameraScreen_Test({}): JSX.Element {
  const isFocused = useIsFocused();
  const appState = useAppState();
  const isActive = isFocused && appState === "active"; // for more informations https://react-native-vision-camera.com/docs/guides/lifecycle
  const [selectedCamera, setSelectedCamera] = useState("back"); // Start with the back camera

  const camera = useRef<Camera>(null);

  const device = useCameraDevice(selectedCamera);

  console.log(JSON.stringify(device, (k, v) => k === "formats" ? [] : v, 2))

  return (
    <View style={{ flex: 1, backgroundColor: "pink" }}>
      <TouchableOpacity
        onPress={() => {
          setSelectedCamera((currentCamera) => (currentCamera === "back" ? "front" : "back"));
        }}
        style={{ width: 50, height: 50, borderRadius: 25, position: "absolute", bottom: 100, alignSelf: "center", backgroundColor: "red" }}
      ></TouchableOpacity>
      <Camera
        ref={camera}
        zoom={device.neutralZoom}
        enableZoomGesture={true}
        resizeMode={"cover"}
        style={{
          width: "100%",
          alignSelf: "center",
          height: "70%",
          backgroundColor: "blue",
        }}
        device={device}
        isActive={isActive}
        photo={true}
        torch={"off"}
      />
    </View>
  );
}
```

### Relevant log output

```shell
nsor mode: -1 Camera position: 1 Camera lens_type: 0 Camera spectrum: -1 load default tuning  Usecase: 0 Sensor mode: -1 Camera position: 1 Camera lens_type: 0 Camera spectrum: -1
01-29 00:43:10.997   909 12359 W libgcam : [ae_helper.cc:575]: FinalizeAeImpl: Desired short TET 33966.00 outside of the achievable shot range, clamped to 6120.31
01-29 00:43:10.997   909 12359 W libgcam : [ae_helper.cc:581]: FinalizeAeImpl: Desired long TET 33966.00 outside of the achievable shot range, clamped to 6120.31
01-29 00:43:11.001   909 12329 W G3ACoordinator: tuning_manager.h:68: Node type 2, Tuning key not found:  Usecase: 13 Sensor mode: -1 Camera position: 1 Camera lens_type: 0 Camera spectrum: -1 load default tuning  Usecase: 0 Sensor mode: -1 Camera position: 1 Camera lens_type: 0 Camera spectrum: -1
01-29 00:43:11.048   909 12381 W Lyric   : external_auto_focus_controller_proxy.cc:1192: In EarlyInterrupt: Set focus happens 79169956 ns after start of exposure time
01-29 00:43:11.057   909 12359 W libgcam : [ae_helper.cc:575]: FinalizeAeImpl: Desired short TET 33966.00 outside of the achievable shot range, clamped to 6120.31
01-29 00:43:11.057   909 12359 W libgcam : [ae_helper.cc:581]: FinalizeAeImpl: Desired long TET 33966.00 outside of the achievable shot range, clamped to 6120.31
01-29 00:43:11.061   909 12329 W G3ACoordinator: tuning_manager.h:68: Node type 2, Tuning key not found:  Usecase: 13 Sensor mode: -1 Camera position: 1 Camera lens_type: 0 Camera spectrum: -1 load default tuning  Usecase: 0 Sensor mode: -1 Camera position: 1 Camera lens_type: 0 Camera spectrum: -1
01-29 00:43:11.063   909 12392 W Lyric   : external_auto_focus_controller_proxy.cc:435: No SW PD stats are valid; Status: INTERNAL: xx_pdaf_results_payload has null payload
01-29 00:43:11.071  3918 12238 D ModemODPMPoller: Current Modem ODPM (mw): 47, threshold: 800
^C
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-right",
  "hardwareLevel": "full",
  "maxZoom": 8,
  "minZoom": 0.5304937958717346,
  "maxExposure": 24,
  "supportsLowLightBoost": true,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera",
    "ultra-wide-angle-camera",
    "telephoto-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": true,
  "minFocusDistance": 9.800000093460085,
  "minExposure": -24,
  "name": "BACK (0)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Google Pixel 7a - Android 14

### VisionCamera Version

3.8.2

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-01-29:
Duplicate of https://github.com/mrousavy/react-native-vision-camera/issues/2380.

closed 2024-01-29 by mrousavy

gdebre, 2024-01-31:
> Duplicate of #2380.

I'm sorry I didn't know this issue was already addressed, but I still have the issue with the camera being zoomed in. Any ideas why ?