margelo/react-native-vision-camera #2452: 🔧 Android Build FAILED Task :app:compileDevelopmentDebugJavaWithJavac

Maheshbhusanoor, 2024-01-29:
### How were you trying to build the app?

When I tried to run my app using yarn android. Its fails.

### Full build logs

```tsx
> Task :app:compileDevelopmentDebugJavaWithJavac
/Users/maheshbhusanoor/MCOM_HM/MPOS/android/app/src/main/java/com/instore/visioncameracodescanner/VisionCameraCodeScannerPluginPackage.java:19: error: cannot find symbol
    FrameProcessorPlugin.register(new VisionCameraCodeScannerPlugin());
                        ^
  symbol:   method register(VisionCameraCodeScannerPlugin)
  location: class FrameProcessorPlugin
/Users/maheshbhusanoor/MCOM_HM/MPOS/android/app/src/main/java/com/instore/visioncameracodescanner/VisionCameraCodeScannerPlugin.java:40: error: VisionCameraCodeScannerPlugin is not abstract and does not override abstract method callback(Frame,Map<String,Object>) in FrameProcessorPlugin
public class VisionCameraCodeScannerPlugin extends FrameProcessorPlugin {
       ^
/Users/maheshbhusanoor/MCOM_HM/MPOS/android/app/src/main/java/com/instore/visioncameracodescanner/VisionCameraCodeScannerPlugin.java:62: error: method does not override or implement a method from a supertype
  @Override
  ^
/Users/maheshbhusanoor/MCOM_HM/MPOS/android/app/src/main/java/com/instore/visioncameracodescanner/VisionCameraCodeScannerPlugin.java:251: error: incompatible types: String cannot be converted to Map<String,Object>
    super("scanCodes");
          ^
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some messages have been simplified; recompile with -Xdiags:verbose to get full output
4 errors
```

### Project dependencies

```json
"react-native": "0.72.5",
"react-native-reanimated": "3.3.0",
 "react-native-vision-camera": "3.7.1",
 "react-native-worklets-core": "^0.2.0",
```

### VisionCamera Version

3.7.1

### Target platforms

Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

No, I cannot build the Example app either

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-01-29:
Hey.

According to these error messages:

```
    FrameProcessorPlugin.register(new VisionCameraCodeScannerPlugin());
                        ^
  symbol:   method register(VisionCameraCodeScannerPlugin)
  location: class FrameProcessorPlugin
/Users/maheshbhusanoor/android/app/src/main/java/com/instore/visioncameracodescanner/VisionCameraCodeScannerPlugin.java:40: error: VisionCameraCodeScannerPlugin is not abstract and does not override abstract method callback(Frame,Map<String,Object>) in FrameProcessorPlugin
public class VisionCameraCodeScannerPlugin extends FrameProcessorPlugin {
       ^
/Users/maheshbhusanoor/android/app/src/main/java/com/instore/visioncameracodescanner/VisionCameraCodeScannerPlugin.java:62: error: method does not override or implement a method from a supertype
  @Override
  ^
/Users/maheshbhusanoor/android/app/src/main/java/com/instore/visioncameracodescanner/VisionCameraCodeScannerPlugin.java:251: error: incompatible types: String cannot be converted to Map<String,Object>
    super("scanCodes");
          ^
```

..I can assume that you are using a QR Code scanner plugin (which you did not mention here) and this fails to build. If you would remove the Code Scanner plugin, it would build.

So you should probably create the issue in the Code Scanner plugin, not in the main VisionCamera repo.

---

From this line specifically:

```
/Users/maheshbhusanoor/android/app/src/main/java/com/instore/visioncameracodescanner/VisionCameraCodeScannerPlugin.java:251: error: incompatible types: String cannot be converted to Map<String,Object>
    super("scanCodes");
          ^
```

I can see that the Code Scanner plugin is using an old API, this constructor doesn't exist like that anymore, that was V2. The plugin needs to be upgraded, see the documentation. But you also didn't mention anything about the plugin, so I cannot help you

---

> **Can you build the VisionCamera Example app?**
> 
> No, I cannot build the Example app either

Did you really try to build the example app, or did you just pick that value so I don't ignore your issue?

The errors are code scanner plugin specific, so I doubt that the example app also fails to build.

closed 2024-01-29 by mrousavy

whwz163, 2025-01-14:
我换了3.9.0跑起来了