margelo/react-native-vision-camera #2463: Recorded video path is corrupted that return from react native vision camera working fine on picture not in video   🐛

HaseebAbbas6586, 2024-01-31:
### What's happening?

i want to record video and click picture using this library but in capturing picture it work fine and return valid path but in video case it record video fine but return corrupted path , i am using react native video to show captured video but the path that return from react native vision camera this corrupted

### Reproduceable Code

```tsx
import {
  Camera,
  useCameraDevice,
  useCameraPermission,
  useMicrophonePermission,
} from 'react-native-vision-camera';

 const [flash, setFlash] = useState(false);
 const [isActive, setIsActive] = useState(true);
 const [mediaType, setMediaType] = useState('image');
 const [storyMedia, setStoryMedia] = useState(null);

 const [deviceCamera, setDeviceCamera] = useState('front');
 const cameraRef = useRef();
 const device = useCameraDevice(deviceCamera);
  if (device == null)
    return (
      <View style={styles.not_found_container}>
        <Text style={styles.camera_not_found_text}>Camera not found</Text>
      </View>
    );
  const {hasPermission, requestPermission} = useCameraPermission();
  const {
    hasPermission: microPhonePermission,
    requestPermission: requestMicroPhonePermission,
  } = useMicrophonePermission();
  useMicrophonePermission();

  useEffect(() => {
    if (!hasPermission) {
      requestPermission();
    }
    if (!microPhonePermission) {
      requestMicroPhonePermission();
    }
  }, []);

  const takePhoto = async () => {
    try {
      const photo = await cameraRef.current.takePhoto({
        flash: flash ? 'on' : 'off',
      });
      setStoryMedia({
        uri: `file://${photo?.path}`,
        type: 'image',
        name: photo?.path.split('/').pop(),
      });
    } catch (error) {}
  };

 const takeVideo = async () => {
    try {
      cameraRef.current.startRecording({
        onRecordingFinished: async video => {
          setStoryMedia({
            uri: `${video?.path}`,
            type: 'video',
            name: video?.path.split('/').pop(),
          });

        },
        onRecordingError: error => {},
        videoBitRate: 'high',
        fileType: 'mp4',
      });
    } catch (error) {
      console.log(error);
    }
  };
  const stopVideo = async () => {
    try {
      await cameraRef.current.stopRecording();
    } catch (error) {}
  };

  <Camera
  ref={cameraRef}
  style={StyleSheet.absoluteFill}
  device={device}
  isActive={isActive}
  photo={mediaType === 'image' ? true : false}
  video={mediaType === 'video' ? true : false}
  audio={false}
  torch={flash ? 'on' : 'off'}
  />
```

### Relevant log output

```shell
on console show this for picture :
photo : {"height": 3456, "isMirrored": false, "isRawPhoto": false, "orientation": "landscape-right", "path": "/data/user/0/com.yekbunsocialapp/cache/mrousavy1904075463334453756.jpg", "width": 4608}

but for video show this :
video : {"duration": 4.75, "height": 1920, "path": "/data/user/0/com.yekbunsocialapp/cache/mrousavy6058063492249876576.mp4", "width": 2560}
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-right",
  "hardwareLevel": "full",
  "maxZoom": 10,
  "minZoom": 1,
  "maxExposure": 20,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 10,
  "minExposure": -20,
  "name": "BACK (0)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

tecno camon 20 , nokia 6.1 , infinix zera 30 

### VisionCamera Version

^3.8.2"

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

HaseebAbbas6586, 2024-01-31:
some time show this error :
 error {"error": {"extra": -38, "what": 1}}
 LOG  error {"error": {"extra": -2147483648, "what": 1}}
 LOG  error {"error": {"extra": -38, "what": 1}}

mrousavy, 2024-01-31:
1. There is no relevant log output in this issue. I added details on how to get those logs, but you simply ignored that
2. I have no idea what "output is corrupted" means.
3. You didn't try it in the example app, so I'm gonna close this issue.

Even tho I have no idea what this issue is about, a friendly tip: try adding `file://` before the Video path. Then it might load fine.

closed 2024-01-31 by mrousavy