margelo/react-native-vision-camera #2465: 🐛 App Crashing when using Frame Processor

Abdulkader-Saoud, 2024-01-31:
### What's happening?

When i use useFrameProcessor the app is crashing while it is working fine when i remove it
 
    console.log(`Frame: ${frame.width}x${frame.height} (${frame.pixelFormat})`)
    this line is not printing anything to console
    
    
I have been trying with different gradle and java versions all the day and nothing worked 
now im using gradle 8.0 and jdk 17.0.10
any help will be appreciated
Thank you
   

### Reproduceable Code

```tsx
import React, {useEffect, useState } from 'react';
import {View, StyleSheet} from 'react-native';
import {
  Camera,
  useCameraFormat,
  useFrameProcessor,
} from 'react-native-vision-camera';

const CCamera = (props) => {
  const {isActive, cam, toggleCam} = props;

  const format = useCameraFormat(cam, [
    {videoResolution: 'max',fps: 'max'},
  ]);

  const frameProcessor = useFrameProcessor((frame) => {
    'worklet'
    console.log(`Frame: ${frame.width}x${frame.height} (${frame.pixelFormat})`)
  }, [])
 

  const handleCameraError = (e) => { console.warn('error', e); };
  return (
    <View style={styles.cameraStyle}>
      {isActive && cam && (
        <Camera
          device={cam}
          video={true}
          audio={true}
          isActive={isActive}
          style={StyleSheet.absoluteFill}
          format={format}
          onError={handleCameraError}
          frameProcessor={frameProcessor }
        />
      )}
    </View>
  );
};

const styles = StyleSheet.create({
  cameraStyle: {
    width: '49%',
    aspectRatio: 3 / 4,
    borderRadius: 10,
  },
});

export default CCamera;
```

### Relevant log output

[... 284 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

sdk_gphone64_x86_64

### VisionCamera Version

3.8.2

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-01-31:
> 01-31 14:29:59.809 16085 16162 I HermesVM: Compiling JS failed: 1:1:invalid empty parentheses '( )',  Buffer size 3 starts with: 280a29 and has protection mode(s): rw-p

Are you using reanimated? See this issue: https://github.com/software-mansion/react-native-reanimated/issues/5576 - enabling `processNestedWorklets` in the babel plugin will probably fix your issue.

closed 2024-01-31 by mrousavy

Abdulkader-Saoud, 2024-02-01:
Sorry again But my babel.config.js is 
```
module.exports = {
  presets: ['module:@react-native/babel-preset'],
  plugins: [
    [
      'react-native-reanimated/plugin',
      {
        processNestedWorklets: true,
      },
    ],
    ['react-native-worklets-core/plugin']
  ]
};
```

and im using those versions 

```
"dependencies": {
    "react": "18.2.0",
    "react-native": "0.73.3",
    "react-native-reanimated": "^3.6.2",
    "react-native-vision-camera": "^3.8.2",
    "react-native-worklets-core": "^0.3.0"
  }
```  

still having the same issue am i missing something ?

Can You provide me with tested versions ?
I mean every thing if possible
jdk gradle AGP rn vision camera .....
Thank you for the response

Abdulkader-Saoud, 2024-02-02:
Finally !!!
This combination worked thank you .
```
"react-native-reanimated": "^3.6.0",
    "react-native-vision-camera": "^3.9.0-beta.0",
    "react-native-worklets-core": "^0.3.0"
```

nurikjohn, 2024-02-03:
Still happening. I am not using reanimated. Created new project with expo.

```
"react-native-vision-camera": "^3.9.0-beta.0",
"react-native-worklets-core": "^0.3.0",
```

Charles-Johnson, 2024-03-04:
I'm getting the same error message, I'm using the following versions:
```json
"react-native": "0.73.4",
"react-native-reanimated": "3.6.2",
"react-native-vision-camera": "3.7.1",
"react-native-worklets-core": "0.2.0"
```
It was working with react-native v0.72.6 and react-native-reanimated v3.7.1 but I had to downgrade react-native-reanimated to v3.6.2 to get other stuff in the app working with react-native v0.73.4

MobileDevPronisi, 2024-06-25:
Did the downgrade work?

ArtRomanov, 2025-04-21:
Gives the same error for me:

"react-native-vision-camera": "^4.0.5",
"react-native-worklets-core": "^1.3.3",
"react-native-reanimated": "~3.10.1",