margelo/react-native-vision-camera #2496: 💭Zoom behaving weirdly

Filocava99, 2024-02-01:
### Question

Hi, I was wondering how the zoom works between different camera devices.
I am using an iPhone 15 Pro and I initialise the camera device like that:
```javascript
const cameraDevice = useCameraDevice('back', {
        physicalDevices: [
            // 'ultra-wide-angle-camera',
            'wide-angle-camera',
            'telephoto-camera'
        ]
    })
```

The zoom as always been neutral but after version 3.6 the neutral zoom is more zoomed than it used to be. Furthermore, using cameraDevice.minZoom when defining the Camera doesn't seem to make a difference:
```javascript
<Camera
                                ref={camera}
                                photo={true}
                                style={{height: '100%', width: '100%'}}
                                device={cameraDevice}
                                format={format}
                                isActive={true}
                                zoom={cameraDevice.minZoom}
                                torch={flash ? "on" : "off"}
                                enableDepthData={true}
                                hdr={true}
                                enableHighQualityPhotos={highQualityPhoto}
                            />
```

My question is then, does zoom change between different camera devices? How does it work when using dual cameras? What was changed that modified the zoom behaviour in recent versions?

### VisionCamera Version

3.8

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2024-02-02:
Nothing has changed between zoom, but I wouldn't be so sure it is zoom issues that you're experiencing. Of course I don't really know a lot about your problem because you created a **Question** instead of an **Issue** with logs, so I can't tell for sure.

Maybe try using `resizeMode="contain"` on the Camera to see if it actually is the zoom or just the format.

Filocava99, 2024-02-02:
I did try that but it make any difference. What's itneresting is that if I change the zoom on the <Camera> it doesn't work. I tried with minZoom and neutralZoom, same result; I tried with numeric values, 0.5, 1, 2, same result.
Might there be a bug?

> Of course I don't really know a lot about your problem because you created a **Question** instead of an **Issue** with logs, so I can't tell for sure.

I wasn't sure if it was actually an issue, anyway there aren't error logs to display

mrousavy, 2024-02-02:
resizeMode="contain" didn't make _any_ difference? It looked exactly the same as "cover"? Then that's a bug.

As for the logs; I'm not looking for error logs, but for any logs that give me information about what sizes are being selected and how the camera is initialized.

Filocava99, 2024-02-02:
Here's the format that I chose from the available formats:
```
 {"autoFocusSystem": "phase-detection", "fieldOfView": 71.28567504882812, "maxFps": 30, "maxISO": 12320, "maxZoom": 189, "minFps": 1, "minISO": 55, "photoHeight": 3024, "photoWidth": 4032, "pixelFormats": ["yuv", "yuv", "rgb", "unknown", "yuv", "unknown", "yuv", "unknown", "rgb"], "supportsDepthCapture": true, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 3024, "videoStabilizationModes": ["auto", "off"], "videoWidth": 4032}
```
Those are all the available formats:
```
 [{"autoFocusSystem": "phase-detection", "fieldOfView": 71.28567504882812, "maxFps": 30, "maxISO": 12320, "maxZoom": 189, "minFps": 1, "minISO": 55, "photoHeight": 3024, "photoWidth": 4032, "pixelFormats": ["yuv", "yuv", "rgb", "unknown", "yuv", "unknown", "yuv", "unknown", "rgb"], "supportsDepthCapture": true, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 3024, "videoStabilizationModes": ["auto", "off"], "videoWidth": 4032}, {"autoFocusSystem": "phase-detection", "fieldOfView": 71.28567504882812, "maxFps": 30, "maxISO": 12320, "maxZoom": 189, "minFps": 1, "minISO": 55, "photoHeight": 3024, "photoWidth": 4032, "pixelFormats": ["yuv", "yuv", "rgb", "unknown", "yuv", "unknown", "yuv", "unknown", "rgb"], "supportsDepthCapture": true, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 3024, "videoStabilizationModes": ["auto", "off"], "videoWidth": 4032}, {"autoFocusSystem": "phase-detection", "fieldOfView": 73.82826232910156, "maxFps": 30, "maxISO": 5280, "maxZoom": 129.125, "minFps": 1, "minISO": 55, "photoHeight": 2376, "photoWidth": 4224, "pixelFormats": ["yuv", "yuv", "rgb", "unknown", "yuv", "unknown", "yuv", "unknown", "rgb"], "supportsDepthCapture": true, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 2160, "videoStabilizationModes": ["auto", "cinematic", "off", "standard", "cinematic-extended"], "videoWidth": 3840}, {"autoFocusSystem": "phase-detection", "fieldOfView": 73.82826232910156, "maxFps": 30, "maxISO": 5280, "maxZoom": 129.125, "minFps": 1, "minISO": 55, "photoHeight": 2376, "photoWidth": 4224, "pixelFormats": ["yuv", "yuv", "rgb", "unknown", "yuv", "unknown", "yuv", "unknown", "rgb"], "supportsDepthCapture": true, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 2160, "videoStabilizationModes": ["auto", "cinematic", "off", "standard", "cinematic-extended"], "videoWidth": 3840}, {"autoFocusSystem": "phase-detection", "fieldOfView": 73.82826232910156, "maxFps": 30, "maxISO": 5280, "maxZoom": 129.125, "minFps": 1, "minISO": 55, "photoHeight": 2376, "photoWidth": 4224, "pixelFormats": ["yuv", "yuv", "rgb", "unknown", "yuv", "unknown", "yuv", "unknown", "rgb"], "supportsDepthCapture": true, "supportsPhotoHdr": false, "supportsVideoHdr": true, "videoHeight": 2160, "videoStabilizationModes": ["auto", "cinematic", "off", "standard", "cinematic-extended"], "videoWidth": 3840}, {"autoFocusSystem": "phase-detection", "fieldOfView": 73.82826232910156, "maxFps": 30, "maxISO": 5280, "maxZoom": 129.125, "minFps": 1, "minISO": 55, "photoHeight": 2376, "photoWidth": 4224, "pixelFormats": ["yuv", "yuv", "rgb", "unknown", "yuv", "unknown", "yuv", "unknown", "rgb"], "supportsDepthCapture": true, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 2160, "videoStabilizationModes": ["auto", "cinematic", "off", "standard", "cinematic-extended"], "videoWidth": 3840}, {"autoFocusSystem": "phase-detection", "fieldOfView": 71.28567504882812, "maxFps": 30, "maxISO": 5280, "maxZoom": 189, "minFps": 1, "minISO": 55, "photoHeight": 3024, "photoWidth": 4032, "pixelFormats": ["yuv", "yuv", "rgb", "unknown", "yuv", "unknown", "yuv", "unknown", "rgb"], "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 2448, "videoStabilizationModes": ["auto", "off"], "videoWidth": 3264}, {"autoFocusSystem": "phase-detection", "fieldOfView": 71.28567504882812, "maxFps": 30, "maxISO": 5280, "maxZoom": 189, "minFps": 1, "minISO": 55, "photoHeight": 3024, "photoWidth": 4032, "pixelFormats": ["yuv", "yuv", "rgb", "unknown", "yuv", "unknown", "yuv", "unknown", "rgb"], "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 2448, "videoStabilizationModes": ["auto", "off"], "videoWidth": 3264}, {"autoFocusSystem": "phase-detection", "fieldOfView": 71.28567504882812, "maxFps": 30, "maxISO": 5280, "maxZoom": 189, "minFps": 1, "minISO": 55, "photoHeight": 3024, "photoWidth": 4032, "pixelFormats": ["yuv", "yuv", "rgb", "unknown", "yuv", "unknown", "yuv", "unknown", "rgb"], "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 1944, "videoStabilizationModes": ["auto", "off"], "videoWidth": 2592}, {"autoFocusSystem": "phase-detection", "fieldOfView": 71.28567504882812, "maxFps": 30, "maxISO": 5280, "maxZoom": 189, "minFps": 1, "minISO": 55, "photoHeight": 3024, "photoWidth": 4032, "pixelFormats": ["yuv", "yuv", "rgb", "unknown", "yuv", "unknown", "yuv", "unknown", "rgb"], "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 1944, "videoStabilizationModes": ["auto", "off"], "videoWidth": 2592}, {"autoFocusSystem": "phase-detection", "fieldOfView": 73.82826232910156, "maxFps": 30, "maxISO": 5280, "maxZoom": 123.75, "minFps": 1, "minISO": 55, "photoHeight": 2376, "photoWidth": 4224, "pixelFormats": ["yuv", "yuv", "rgb", "u [... cut ...]
                                device={cameraDevice}
                                format={format}
                                isActive={true}
                                zoom={cameraDevice.minZoom}
                                resizeMode="contain"
                                torch={flash ? "on" : "off"}
                                enableDepthData={true}
                                hdr={true}
                                enableHighQualityPhotos={highQualityPhoto}
                            />
```

mrousavy, 2024-02-02:
videoWidth/videoHeight is 4032 x 3024 - that's a 4:3 format. Not 16:9. Is that intentional?

Filocava99, 2024-02-02:
It is. The thing is, if I use the camera app with that resolution and 4:3 format, I see the scene with the zoom level that I expect. When I use the camera from inside my app, the zoom is messed up and it's unresponsive to any change made to <Camera>

mrousavy, 2024-02-05:
No, it was weekend.

Filocava99, 2024-02-09:
Hi @mrousavy , maybe the issue is the same of there: https://github.com/react-native-camera/react-native-camera/issues/2472

mrousavy, 2024-02-13:
Can you post one screenshot with `resizeMode="cover"` and one with `resizeMode="contain"`?

Filocava99, 2024-02-17:
> Can you post one screenshot with `resizeMode="cover"` and one with `resizeMode="contain"`?

Yes I will. In the meantime have you taken a look here? https://github.com/react-native-camera/react-native-camera/issues/2472

Filocava99, 2024-02-17:
@mrousavy here's the screenshots: https://imgur.com/a/v2ha01h
One is with iPhone camera and the other two using contain and cover

Filocava99, 2024-02-18:
Fixed it. Using telephoto override the min zoom of the wide camera. I had to use wide and ultra-wide to be able to use the dual camera with depth map

closed 2024-02-18 by Filocava99