margelo/react-native-vision-camera #2497: 🐛 Recording video duration incorrect

JaxonLee0710, 2024-02-02:
### What's happening?

I wanted video recording app with pause/resume.
Of course video recording is perfect.
But there is some issue on video recording duration.
For example,
start record
pause record after 1 second.
resume record after 5 seconds.
stop record after 2 seconds.
Expected video duration is 3 seonds.
But onRecordingFinished callback return video object with 8 seconds duration.

### Reproduceable Code

```tsx
const recordTest = async () => {
   console.log(cameraDevice);
   // Start
   let timeRecord = moment().unix();
   await _camera.current.startRecording({
      onRecordingError: () => {},
      onRecordingFinished: video => {
         console.log(video);
      },
   });
   console.log(`start record ${moment().unix() - timeRecord}s`);
   await timeout(1000);
   // Pause
   await _camera.current.pauseRecording();
   console.log(`pause record ${moment().unix() - timeRecord}s`);
   await timeout(5000);
   // Resume
   await _camera.current.resumeRecording();
   console.log(`resume record ${moment().unix() - timeRecord}s`);
   await timeout(2000);
   // Stop
   const video = await _camera.current.stopRecording();
   console.log(`finish record ${moment().unix() - timeRecord}s`);
};
```

### Relevant log output

```shell
LOG  start record 0s
 LOG  pause record 1s
 LOG  resum record 7s
 LOG  {"duration": 8.74, "height": 960, "path": "/data/user/0/com.testvisioncamera.dev/cache/mrousavy3304552134792529569.mov", "width": 1280}
 LOG  finish record 10s
```

### Camera Device

```json
{ 
    "formats": [] //36,hardwareLevel:
    'hardwareLevel':limited',
    hasFlash: false
    hasTorch: false
    id:'0'
    isMultiCam: false
    maxExposure:9
    maxZoom: 10
    minExposure:-9
    minFocusDistance: NaN
    minZoom: 1
    name:'BACK (0)'
    neutralZoom: 1
    physicalDevices: (1) ['telephoto-camera']
    position: 'back'
    sensorOrientation: 'landscape-right'
    supportsFocus: true
    supportsLowLightBoost: false
}
```

### Device

AVD emulator(api 33)

### VisionCamera Version

3.8.2

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-02-02:
Hey! Thanks for the issue report.

You didn't put the native adb / Xcode logs in here so it's impossible for me to figure out when exactly the start/stop happened.
I need those logs. Maybe I need to do some timestamp correction, but not sure how that'd work on Android.
Can you also test on iOS please?

Also, unrelated but why is `minFocusDistance` `NaN` 🤨

mrousavy, 2024-02-05:
1. Format your logs please. Three backticks (```)
2. > I can't test on IOS, but I think that's same.

    I doubt that. iOS is a completely different implementation in a completely different language and environment, it is not the same. If it's the same that'd honestly surprise me.

JaxonLee0710, 2024-02-05:
Video is correctly recorded, but only duration time(result of onFinishRecording callback) is problem.
So I think that's not considered.

stevenkang12, 2024-04-17:
We're also running into the same issue. During the pause, the full video includes the pause time with a still frame. I also found that the audio that gets captured after resuming gets overlayed onto the pause time still frames.

lee-byeoksan, 2024-05-31:
@stevenkang12 maybe related to #2790

mrousavy, 2024-06-08:
Hey all! 

I just spent a few days on thinking about a battleproof timestamp synchronization solution, and I came up with a great idea.
I built a `TrackTimeline` helper class which represents a video or audio track - it can be started & stopped, paused & resumed, and even supports nesting pauses without issues.
- The total duration of the video is summed up from the difference between the first and the last actually written timestamps, minus the total duration of all pauses between a video. No more incorrect `video.duration`! 🥳 
- Whereas before I just had a 4 second timeout if no frames arrive, I now just wait twice the frame latency (a few milliseconds) to ensure no frames are left out at maximum! 🎉 
- A video can be stopped while it is paused without any issues, as a pause call is taken into consideration before stopping 💪 
- A video file's session now exactly starts at the start() timestamp, and ends at the exact timestamp of the last video frame - this ensures there can never be any blank frames in the video, even if the audio track is longer 🤩 

This was really complex to built as I had to synchronize timestamps between capture sessions, and the entire thing is a producer model - a video buffer can come like a second or so later than the audio buffer, but I need to make sure the video track starts before the audio track starts, and ends after the audio track ends - that's a huge brainf*ck! 🤯😅 

There's also no helper APIs for this on iOS, and it looks like **no other Camera framework** (not even native Swift/ObjC iOS Camera libraries) **support this** - they all break when timestamps have a delay (e.g. video stabilization enabled) (or dont even support delays at all) ; so I had to build the thing myself.

Check out this PR and try if it fixes the issue for you; https://github.com/mrousavy/react-native-vision-camera/pull/2948

Thanks! ❤️