margelo/react-native-vision-camera #2498: ✨ Cancel Recording without capture

JaxonLee0710, 2024-02-02:
### What feature or enhancement are you suggesting?

I think it would be great to support cancel recording function.

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

I am currently camera recording app using this package.
But I want to cancel recording without store video.
Currently, this package is not support cancel recording function.
Of course, that's support onStopRecording, but that's need store file to temp so there is some minutes to store video file.
But video file doesn't need because user cancel recording.

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2024-02-02:
Good feature request 👍

mrousavy, 2024-04-22:
Added `cancelRecording()` in VisionCamera 4.0.0! 🥳 

If you appreciate VisionCamera, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support the development of VisionCamera and thank me for my time spent on fixing bugs and building new features.

closed 2024-04-22 by mrousavy