margelo/react-native-vision-camera #2500: 💭 Where exactly the flag for QR code scanning need to be set?

balazsgerlei, 2024-02-02:
### Question

The [guide ](https://react-native-vision-camera.com/docs/guides/code-scanning) says now you need to set the flag `VisionCamera_enableCodeScanner=true` in `build.gradle` on Android, but which `build.gradle`? The _project root_ or inside the _app module_? Furthermore, where exactly this flag should be set inside the gradle script?

Is it inside this in the `build.gradle` in the root of the project e.g.:
```
buildscript {
    ext {
        VisionCamera_enableCodeScanner=true
        ...
    }
}
```

If yes, please add a clear example, like above.

While we are at it, is the previous method (before 3.7.0 - declaring a dependency in the Manifest) no longer needed? If it is needed still for older versions, could you state this in the docs (e.g. from 3.7.0 the new method with the flag, before the Manifest addition).

Also if you would consider updating the example app with how to do this, it would be awesome!

Thank you for your kind consideration!

### VisionCamera Version

3.8.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

balazsgerlei, 2024-02-02:
Ah okay, I misread it somehow, thank you for your fast repsonse!

balazsgerlei, 2024-02-02:
And the Manifest addition is no longer needed, right?

mrousavy, 2024-02-02:
No, no longer needed.