margelo/react-native-vision-camera #2504: How do you draw on frame?

tomerh2001, 2024-02-03:
### Question

I am using frame processors and I can't seem to find a way to draw on frame, shown do I do it? it's such a basic feature for an advanced library like this, there's no way it doesn't exist.

### VisionCamera Version

^3.8.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2024-02-05:
See [this page in the docs](https://react-native-vision-camera.com/docs/guides/skia-frame-processors).

mrousavy, 2024-02-05:
[image]

closed 2024-02-05 by mrousavy

mrousavy, 2024-02-06:
> So basically there isn't currently any way to draw on the frame? At all? What? For a library that claims to do a lot, this is a very very basic feature that's missing...

 1. VisionCamera does not claim that it can draw on Frames. You must've read wrong.
 2. If it's a "very very basic feature" go ahead and implement it yourself 😉 

> Just give us an optional JS callback function that we can pass to `Camera` that receives `frame` and returns the updated frame we want to display... even if it's not that efficient, at least we'll have the ability to do it...

Sure if you add this feature, you can open a PR in this repo so we can have the code merged upstream, thank you for your contribution! 👍