margelo/react-native-vision-camera #2505: ✨ maxDuration and maxFileSize

joebochill, 2024-02-03:
### What feature or enhancement are you suggesting?

Extend `RecordVideoOptions` to include properties for `maxDuration` and `maxFileSize`. 

Related questions around duration [here](https://github.com/mrousavy/react-native-vision-camera/issues/1578) and [here](https://github.com/mrousavy/react-native-vision-camera/issues/571).

I am migrating an app from using react-native-camera (now deprecated) to this library. One of the nice things about react-native-camera was the ability to set a file size/duration limit, after which the video recording is stopped automatically and the file returned as normal. This is especially useful in my app as users are recording themselves while the camera is on a tripod / otherwise not in their hand to stop the recording.

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

`maxDuration` is easy enough to work around using a timer (would still be nice to have the option built-in though).

`maxFileSize` requires post-processing after the video is recorded — as far as I can tell there's not a straightforward way to do this except to guess and check with trimming / compression settings until the video size is below the maximum.

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2024-02-05:
Good feature request, thanks! 👍

a0m0rajab, 2024-06-21:
I was opening a new issue with similar request. the expo camera have this features though: 

https://docs.expo.dev/versions/latest/sdk/camera/#camerarecordingoptions 

As for the max duration: 

We can achieve this programmatically by using the stop recording function: https://react-native-vision-camera.com/docs/api/classes/Camera#stoprecording

but I am not sure about the maxFileSize yet.

Rossella-Mascia-Neosyn, 2024-09-12:
Is there any update?

mrousavy, 2024-09-12:
Nope, I don't have this on my roadmap for now.