margelo/react-native-vision-camera #2506: 🐛 Slightly glitchy change between wide and ultra-wide camera

griffinbaker12, 2024-02-03:
### What's happening?

https://github.com/mrousavy/react-native-vision-camera/assets/96966609/946e6732-b702-44f8-b96c-138b6feb63c1

When switching for the first time between the 0.5x and 1x camera, you can see that the view is somewhat glitchy as the view re-adjusts itself. I was wondering if this is normal, and, if not, how it can be fixed? 

I did download the example app and play around with it, but the camera switching behavior in my app is fundamentally different as it does not re-direct the user to a menu screen that is different from where the camera is rendered.

### Reproduceable Code

```tsx
// This is the code that dictates which device types I use: 

 const physicalDevices = useMemo(() => {
        const arr: PhysicalCameraDeviceType[] = ['wide-angle-camera'];
        if (includeUltraWide) {
            arr.push('ultra-wide-angle-camera');
        }
        return arr;
 }, [includeUltraWide]);

// I have a button in the video that updates this array accordingly.
```

### Relevant log output

```shell
...
```

### Camera Device

```json
{
  "name": "Back Camera",
  "hasFlash": true,
  "position": "back",
  "hardwareLevel": "full",
  "supportsLowLightBoost": false,
  "sensorOrientation": "landscape-right",
  "isMultiCam": false,
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:0",
  "formats": [],
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "hasTorch": true,
  "maxZoom": 123.75,
  "minZoom": 1
}
 LOG  {
  "sensorOrientation": "landscape-right",
  "hasTorch": false,
  "isMultiCam": false,
  "minZoom": 1,
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:1",
  "supportsFocus": true,
  "supportsRawCapture": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "hardwareLevel": "full",
  "position": "front",
  "supportsLowLightBoost": false,
  "name": "Front Camera",
  "hasFlash": true,
  "maxZoom": 128.875,
  "formats": []
}
```

### Device

iPhone 15 Pro

### VisionCamera Version

3.6.8

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

stevengoldberg, 2024-02-04:
I think that's normal; if you want to transition smoothly, you need to include both physical devices in the array and change the camera zoom to switch between them.

mrousavy, 2024-02-05:
Exactly what @stevengoldberg said, thanks Steven!

@griffinbaker12 read the Camera Devices docs, you need to **always** include both ultra-wide and wide camera in the useCameraDevice hook, and the 0.5x and 1x buttons should just switch the zoom between `device.minZoom` and `device.neutralZoom` 😉

closed 2024-02-05 by mrousavy

griffinbaker12, 2024-02-05:
Thank you both for the response!. I gave it a shot using the techniques you described, and this is the behavior that I observe:

https://github.com/mrousavy/react-native-vision-camera/assets/96966609/c33c0b43-b106-40b0-b6ca-1fe011b48981

It seems that the glitchy behavior from before happens the **_first time in a given session_** that I open the camera and attempt to switch from back to front. If I close the camera and re-open within the same session, the switch looks as it does in the video when I switch from front to back the **_second time._** Is this normal behavior, or can it be improved in some way? 

Thank you!