margelo/react-native-vision-camera #2507: 🐛 `useFrameProcessor` causes app to crash [Expo]

lucksp, 2024-02-04:
### What's happening?

[There is a similar issue, but #2465 is closed, so starting new]
[I also originally reported issue on [Discord channel](https://discord.com/channels/1147256486762913884/1147256815474725044/1202791450639868005), but now confirmed is related to just `useProcessor`]
[*I DID NOT* try to replicate in your example repo, because I am using Expo and it's not a 1:1 comparison]

## Steps to replicate:
1. Within repo link below, run dev build & open app
2. You will see camera running in view finder
3. toggle the `frameProcessor` prop passed to camera ([here](https://github.com/lucksp/rnvc-test/blob/main/src/components/CameraView/index.tsx#L18))
4. App will now crash

## Notes:

I am unable to locate any meaningful crash logs to my untrained eye, but it's definitely from within the `useFrameProcessor` hook, since if I remove it the app does not crash.  

```
    "react-native": "0.72.6",
    "expo": "^49.0.1",
    "react-native-vision-camera": "^3.8.2",
    "react-native-worklets-core": "^0.3.0",
```

I use just the hook without any logic, [copied from docs](https://react-native-vision-camera.com/docs/guides/frame-processors#what-are-frame-processors), and it crashes with no output:

```
  const frameProcessor = useFrameProcessor(frame => {
    'worklet';
    console.log(`Frame: ${frame.width}x${frame.height} (${frame.pixelFormat})`);
  }, []);
```

I was able to go into the package code & debug the JS:
something about [`createFrameProcessor`](https://github.com/mrousavy/react-native-vision-camera/blob/97168c647c4fc8cd48c9989e6d6ef70ac65d88b4/package/src/hooks/useFrameProcessor.ts#L14) code is not being triggered??

```js
export function useFrameProcessor(frameProcessor: (frame: Frame) => void, dependencies: DependencyList): FrameProcessor {
  console.log('inside useFrameProcessor package'); // this gets run in console
  
  return useMemo(() => {
    console.log('createProcessor'); // this gets run in console
    const process = createFrameProcessor(frameProcessor, 'frame-processor'); // no logs from inside get triggered.
  
  debugger; // this does not get triggered
  return process;
  }, dependencies)
```

### Reproduceable Code

```tsx
[https://github.com/lucksp/rnvc-test](https://github.com/lucksp/rnvc-test)

Install latest EAS Dev Build to run the repo in development mode: 
https://expo.dev/accounts/lucksp/projects/rnvc-test/builds/ed8e64d9-7de2-4f14-8aa1-322de7bffa95
- (Scan QR code inside link)

Code Snippet:

export function useCamera() {
  // UNCOMMENT HERE TO ENABLE & SEE APP CRASH
  let frameProcessor;
  // const frameProcessor = useFrameProcessor(frame => {
  //   'worklet';
  //   console.log(`Frame: ${frame.width}x${frame.height} (${frame.pixelFormat})`);
  // }, []);

  const cameraRef = useRef<Component<CameraProps>>(null);
  const { hasPermission, requestPermission } = useCameraPermission();
  const device = useCameraDevice('back', {
    physicalDevices: ['ultra-wide-angle-camera', 'wide-angle-camera'],
  });

  console.log(JSON.stringify(device, (k, v) => k === "formats" ? [] : v, 2))

  const neutralZoom = device?.neutralZoom ?? MINZOOM;
  const minZoom = device?.minZoom ?? MINZOOM;
  const maxZoom = Math.min(device?.maxZoom ?? MINZOOM, MAXZOOM);

  const [isCapturing, setIsCapturing] = useState(false);
  const isFocused = useIsFocused();
  const isForeground = useIsForeground();
  const isActive = isFocused && isForeground;

  const format = useCameraFormat(device, Templates.FrameProcessingRGB);

  useEffect(
    function checkPermissions() {
      // TODO - open nice permissions UI & if denied link to settings
      if (!hasPermission) {
        requestPermission();
      }
    },
    [hasPermission, requestPermission]
  );

  return {
    device,
    maxZoom,
    minZoom,
    Camera,
    isActive,
    cameraRef,
    isCapturing,
    setIsCapturing,
    format,
    frameProcessor,
  };
}

Wrap my UI in a provider:

```js
const CameraProvider = ({ children }: {children: ReactNode;}) => {
  const camera = useCamera();

  return (
    // eslint-disable-next-line @typescript-eslint/ban-ts-comment
    // @ts-ignore
    <CameraContext.Provider value={{ ...camera }}>
      {children}
    </CameraContext.Provider>
  );
};
```

build UI:
```js
const CameraView = () => {
  const { device, Camera, cameraRef, isActive, format, frameProcessor } =
    useCameraContext();
  return (
    <>
      {device ? (
        <>
          <View>
            {/* <Reanimated.View> */}
            <Camera
              // animatedProps={animatedProps}
              device={device}
              format={format}
              frameProcessor={frameProcessor}
              isActive={isActive}
              photo
              ref={cameraRef}
            />
            {/* </Reanimated.View> */}
          </View>
        </>
      ): <Text>Loading Camera or not found</Text>}
    </>
  );
};
```
```

### Relevant log output

```shell
error	12:00:01.350017-0700	SpringBoard	Advisor: No handle found for currently focused PID: 6045; sceneIdentity: com.apple.frontboard.systemappservices::FBSceneManager:sceneID%3Acom.lucksp.rnvctest-default
error	12:00:01.350045-0700	SpringBoard	Advisor: No handle found for currently focused PID: 6045; sceneIdentity: com.apple.frontboard.systemappservices::FBSceneManager:sceneID%3Acom.lucksp.rnvctest-default
error	12:00:01.583403-0700	SpringBoard	Ignoring update for invalidated scene: (FBSceneManager):sceneID:com.lucksp.rnvctest-default
error	12:00:01.583424-0700	SpringBoard	Ignoring update for invalidated scene: (FBSceneManager):sceneID:com.lucksp.rnvctest-default
error	12:00:01.584615-0700	SpringBoard	Scene FBSceneManager/sceneID:com.lucksp.rnvctest-default update failed: <NSError: 0x28007eb50; domain: FBSceneErrorDomain; code: 1 ("operation-failed"); "Scene update failed."> {
    NSUnderlyingError = <NSError: 0x280012070; domain: FBWorkspace; code: 1; "Scene client is invalid.">;
}
error	12:00:01.584638-0700	SpringBoard	Scene FBSceneManager/sceneID:com.lucksp.rnvctest-default update failed: <NSError: 0x28007eb50; domain: FBSceneErrorDomain; code: 1 ("operation-failed"); "Scene update failed."> {
    NSUnderlyingError = <NSError: 0x280012070; domain: FBWorkspace; code: 1; "Scene client is invalid.">;
}
error	12:00:01.617273-0700	runningboardd	RBSStateCapture remove item called for untracked item <RBConnectionClient| 6045 name:app<com.lucksp.rnvctest(DD75145D-5214-4263-AF89-DCD614B687C6)> entitlements:<RBEntitlements| [
			
			]> inheritanceManager:<RBClientInheritanceManager| >>
error	12:00:01.617295-0700	runningboardd	RBSStateCapture remove item called for untracked item <RBConnectionClient| 6045 name:app<com.lucksp.rnvctest(DD75145D-5214-4263-AF89-DCD614B687C6)> entitlements:<RBEntitlements| [
			
			]> inheritanceManager:<RBClientInheritanceManager| >>
error	12:00:01.618284-0700	symptomsd	COSMCtrl _foregroundAppActivity incoming bundle com.lucksp.rnvctest has nil supplied UUID, finds existing EA936BA4-CD05-3451-9D53-AFDC9FEB2C0C
error	12:00:01.618345-0700	symptomsd	COSMCtrl _foregroundAppActivity incoming bundle com.lucksp.rnvctest has nil supplied UUID, finds existing EA936BA4-CD05-3451-9D53-AFDC9FEB2C0C
```

### Camera Device

```json
{
  "hardwareLevel": "full",
  "isMultiCam": true,
  "minZoom": 1,
  "maxZoom": 123.75,
  "hasTorch": true,
  "sensorOrientation": "landscape-right",
  "maxExposure": 8,
  "supportsLowLightBoost": false,
  "hasFlash": true,
  "minExposure": -8,
  "name": "Back Dual Wide Camera",
  "supportsRawCapture": false,
  "neutralZoom": 2,
  "supportsFocus": true,
  "physicalDevices": [
    "ultra-wide-angle-camera",
    "wide-angle-camera"
  ],
  "position": "back",
  "formats": [],
  "minFocusDistance": 2,
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:6"
}
```

### Device

iPhone 13Pro

### VisionCamera Version

^3.8.2

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-02-05:
Hey! Thanks for the report, this seems to be very tightly integrated in the Expo environment. Is there any chance you could try to reproduce this bug in a bare react-native app?

Or at least get the full adb logcat logs of the crash? In the logs you provided there doesn't seem to be any camera related stuff going on, so I'm not sure if those are related to the crash.

lucksp, 2024-02-05:
Hi. Babel plugin is setup here:
https://github.com/lucksp/rnvc-test

I’ll truly to get more logs and info

Mnelemos, 2024-02-06:
Same versions, standard react-native env, and I am getting the same Frame Processor crashes with the following logcat error:

```2024-02-06 00:20:27.590 10321-10446 VideoPipeline           com.cnn                              E  Failed to call Frame Processor!
                                                                                                    com.facebook.jni.CppException: Compiling JS failed: 1:1:invalid empty parentheses '( )' Buffer size 3 starts with: 280a29
                                                                                                    	at com.mrousavy.camera.frameprocessor.FrameProcessor.call(Native Method)
                                                                                                    	at com.mrousavy.camera.core.VideoPipeline._init_$lambda$0(VideoPipeline.kt:117)
                                                                                                    	at com.mrousavy.camera.core.VideoPipeline.$r8$lambda$EMriwS_FKhtjTFtppM84Z5V8tiI(Unknown Source:0)
                                                                                                    	at com.mrousavy.camera.core.VideoPipeline$$ExternalSyntheticLambda0.onImageAvailable(Unknown Source:2)
                                                                                                    	at android.media.ImageReader$1.run(ImageReader.java:916)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:942)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:99)
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:226)
                                                                                                    	at android.os.Looper.loop(Looper.java:313)
                                                                                                    	at android.os.HandlerThread.run(HandlerThread.java:67)
2024-02-06 00:20:27.628  1047-10557 ExynosCame...estManager ven...amera.provider@4.0-service_64  E  [CAM(-1)][]-(hasStream[665]):Invalid Stream id (20)
2024-02-06 00:20:27.628  1047-10557 ExynosCame...estManager ven...amera.provider@4.0-service_64  E  [CAM(-1)][]-(hasStream[665]):Invalid Stream id (20)
2024-02-06 00:20:27.656 10321-10446 VideoPipeline           com.cnn                              E  Failed to call Frame Processor!
                                                                                                    com.facebook.jni.CppException: Compiling JS failed: 1:1:invalid empty parentheses '( )' Buffer size 3 starts with: 280a29
                                                                                                    	at com.mrousavy.camera.frameprocessor.FrameProcessor.call(Native Method)
                                                                                                    	at com.mrousavy.camera.core.VideoPipeline._init_$lambda$0(VideoPipeline.kt:117)
                                                                                                    	at com.mrousavy.camera.core.VideoPipeline.$r8$lambda$EMriwS_FKhtjTFtppM84Z5V8tiI(Unknown Source:0)
                                                                                                    	at com.mrousavy.camera.core.VideoPipeline$$ExternalSyntheticLambda0.onImageAvailable(Unknown Source:2)
                                                                                                    	at android.media.ImageReader$1.run(ImageReader.java:916)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:942)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:99)
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:226)
                                                                                                    	at android.os.Looper.loop(Looper.java:313)
                                                                                                    	at android.os.HandlerThread.run(HandlerThread.java:67)
2024-02-06 00:20:27.661  1047-10557 ExynosCame...estManager ven...amera.provider@4.0-service_64  E  [CAM(-1)][]-(hasStream[665]):Invalid Stream id (20)
2024-02-06 00:20:27.661  1047-10557 ExynosCame...estManager ven...amera.provider@4.0-service_64  E  [CAM(-1)][]-(hasStream[665]):Invalid Stream id (20)
2024-02-06 00:20:27.690 10321-10446 VideoPipeline           com.cnn                              E  Failed to call Frame Processor!
                                                                                                    com.facebook.jni.CppException: Compiling JS failed: 1:1:invalid empty parentheses '( )' Buffer size 3 starts with: 280a29
                                                                                                    	at com.mrousavy.camera.frameprocessor.FrameProcessor.call(Native Method)
[... cut ...]
                                                                                                    	at android.os.Looper.loop(Looper.java:313)
                                                                                                    	at android.os.HandlerThread.run(HandlerThread.java:67)```
Similar to the problem in this issue: https://github.com/mrousavy/react-native-vision-camera/issues/2199 , where they managed to solve it by reversing their versions, but I haven't tried reversing mine yet due to my Gradle Version not supporting earlier versions of react-native-vision-camera

Mnelemos, 2024-02-06:
Same versions, standard react-native env, and I am getting the same Frame Processor crashes.

Similar to the problem in this issue: https://github.com/mrousavy/react-native-vision-camera/issues/2199 , where they managed to solve it by reversing their versions, but I haven't tried reversing mine yet due to my Gradle Version not supporting earlier versions of react-native-vision-camera.

Logcat results:

```2024-02-06 00:20:27.590 10321-10446 VideoPipeline           com.cnn                              E  Failed to call Frame Processor!
                                                                                                    com.facebook.jni.CppException: Compiling JS failed: 1:1:invalid empty parentheses '( )' Buffer size 3 starts with: 280a29
                                                                                                    	at com.mrousavy.camera.frameprocessor.FrameProcessor.call(Native Method)
                                                                                                    	at com.mrousavy.camera.core.VideoPipeline._init_$lambda$0(VideoPipeline.kt:117)
                                                                                                    	at com.mrousavy.camera.core.VideoPipeline.$r8$lambda$EMriwS_FKhtjTFtppM84Z5V8tiI(Unknown Source:0)
                                                                                                    	at com.mrousavy.camera.core.VideoPipeline$$ExternalSyntheticLambda0.onImageAvailable(Unknown Source:2)
                                                                                                    	at android.media.ImageReader$1.run(ImageReader.java:916)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:942)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:99)
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:226)
                                                                                                    	at android.os.Looper.loop(Looper.java:313)
                                                                                                    	at android.os.HandlerThread.run(HandlerThread.java:67)
2024-02-06 00:20:27.628  1047-10557 ExynosCame...estManager ven...amera.provider@4.0-service_64  E  [CAM(-1)][]-(hasStream[665]):Invalid Stream id (20)
2024-02-06 00:20:27.628  1047-10557 ExynosCame...estManager ven...amera.provider@4.0-service_64  E  [CAM(-1)][]-(hasStream[665]):Invalid Stream id (20)
2024-02-06 00:20:27.656 10321-10446 VideoPipeline           com.cnn                              E  Failed to call Frame Processor!
                                                                                                    com.facebook.jni.CppException: Compiling JS failed: 1:1:invalid empty parentheses '( )' Buffer size 3 starts with: 280a29
                                                                                                    	at com.mrousavy.camera.frameprocessor.FrameProcessor.call(Native Method)
                                                                                                    	at com.mrousavy.camera.core.VideoPipeline._init_$lambda$0(VideoPipeline.kt:117)
                                                                                                    	at com.mrousavy.camera.core.VideoPipeline.$r8$lambda$EMriwS_FKhtjTFtppM84Z5V8tiI(Unknown Source:0)
                                                                                                    	at com.mrousavy.camera.core.VideoPipeline$$ExternalSyntheticLambda0.onImageAvailable(Unknown Source:2)
                                                                                                    	at android.media.ImageReader$1.run(ImageReader.java:916)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:942)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:99)
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:226)
                                                                                                    	at android.os.Looper.loop(Looper.java:313)
                                                                                                    	at android.os.HandlerThread.run(HandlerThread.java:67)
2024-02-06 00:20:27.661  1047-10557 ExynosCame...estManager ven...amera.provider@4.0-service_64  E  [CAM(-1)][]-(hasStream[665]):Invalid Stream id (20)
2024-02-06 00:20:27.661  1047-10557 ExynosCame...estManager ven...amera.provider@4.0-service_64  E  [CAM(-1)][]-(hasStream[665]):Invalid Stream id (20)
2024-02-06 00:20:27.690 10321-10446 VideoPipeline           com.cnn                              E  Failed to call Frame Processor!
                                                                                                    com.facebook.jni.CppException: Compiling JS failed: 1:1:invalid empty parentheses '( )' Buffer size 3 starts with: 280a29
[... cut ...]
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:99)
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:226)
                                                                                                    	at android.os.Looper.loop(Looper.java:313)
                                                                                                    	at android.os.HandlerThread.run(HandlerThread.java:67)```

lucksp, 2024-02-06:
I am not getting a crash on Android...
Here is my `adb` logs @mrousavy , as requested:

```bash
2024-02-05 18:04:54.695   956-996   UsageStatsService       system_server                        W  Unexpected activity event reported! (com.android.camera2/com.dream.camera.modules.qr.QrScanResultActivity event : 23 instanceId : 188415557)
2024-02-05 18:04:54.708   956-1556  InputManager-JNI        system_server                        W  Input channel object 'Letterbox_left_ActivityRecord{ff5799a u0 com.android.camera2/com.dream.camera.modules.qr.QrScanResultActivity} t502 d0} (client)' was disposed without first being removed with the input manager!
2024-02-05 18:04:54.710   956-1556  InputManager-JNI        system_server                        W  Input channel object 'Letterbox_top_ActivityRecord{ff5799a u0 com.android.camera2/com.dream.camera.modules.qr.QrScanResultActivity} t502 d0} (client)' was disposed without first being removed with the input manager!
2024-02-05 18:04:54.710   956-1556  InputManager-JNI        system_server                        W  Input channel object 'Letterbox_right_ActivityRecord{ff5799a u0 com.android.camera2/com.dream.camera.modules.qr.QrScanResultActivity} t502 d0} (client)' was disposed without first being removed with the input manager!
2024-02-05 18:04:54.711   956-1556  InputManager-JNI        system_server                        W  Input channel object 'Letterbox_bottom_ActivityRecord{ff5799a u0 com.android.camera2/com.dream.camera.modules.qr.QrScanResultActivity} t502 d0} (client)' was disposed without first being removed with the input manager!
2024-02-05 18:04:54.713   956-996   UsageStatsService       system_server                        W  Unexpected activity event reported! (com.android.camera2/com.android.camera.CameraActivity event : 23 instanceId : 89220309)
2024-02-05 18:05:07.683 29783-29924 CameraManagerGlobal     com.lucksp.rnvctest                  I  Connecting to camera service
2024-02-05 18:05:07.875 29783-29930 unknown:Vi...rtyUpdater com.lucksp.rnvctest                  W  Could not find generated setter for class com.mrousavy.camera.CameraViewManager
2024-02-05 18:05:07.896 29783-29930 CameraDevices           com.lucksp.rnvctest                  I  Camera #0 is now available.
2024-02-05 18:05:07.897 29783-29930 CameraDevices           com.lucksp.rnvctest                  I  Camera #1 is now available.
2024-02-05 18:05:10.157 29783-29929 VisionCameraProxy       com.lucksp.rnvctest                  I  Initializing VisionCameraProxy...
2024-02-05 18:05:10.158 29783-29929 VisionCameraProxy       com.lucksp.rnvctest                  I  Creating Worklet Context...
2024-02-05 18:05:10.162 29783-29929 VisionCameraProxy       com.lucksp.rnvctest                  I  Worklet Context created!
2024-02-05 18:05:10.197   741-4495  ServiceManager          cameraserver                         W  Permission failure: android.permission.CAMERA from uid=10073 pid=15171
2024-02-05 18:05:10.200   741-4495  ServiceManager          cameraserver                         W  Permission failure: android.permission.CAMERA from uid=10073 pid=15171
2024-02-05 18:05:17.532 29783-29783 CameraView              com.lucksp.rnvctest                  I  Updating CameraSession...
2024-02-05 18:05:17.537 29783-29961 CameraSession           com.lucksp.rnvctest                  I  configure { ... }: Waiting for lock...
2024-02-05 18:05:17.545 29783-29961 CameraSession           com.lucksp.rnvctest                  I  configure { ... }: Updating CameraSession Configuration... Difference(deviceChanged=true, outputsChanged=true, sidePropsChanged=true, isActiveChanged=true)
2024-02-05 18:05:17.546 29783-29961 CameraSession           com.lucksp.rnvctest                  I  Need to rebuild CameraDevice and CameraCaptureSession...
2024-02-05 18:05:17.546 29783-29961 CameraSession           com.lucksp.rnvctest                  I  CameraDevice and CameraCaptureSession is torn down but Camera is not active, skipping update...
2024-02-05 18:05:17.547 29783-29961 CameraSession           com.lucksp.rnvctest                  I  Successfully updated CameraSession Configuration! isActive: false
2024-02-05 18:05:17.550 29783-29961 CameraView              com.lucksp.rnvctest                  I  invokeOnInitialized()
2024-02-05 18:05:17.565 29783-29783 CameraView              com.lucksp.rnvctest                  I  Updating CameraSession...
2024-02-05 18:05:17.565 29783-29961 CameraSession           com.lucksp.rnvctest                  I  configure { ... }: Waiting for lock...
2024-02-05 18:05:17.566 29783-29961 CameraSession           com.lucksp.rnvctest                  I  configure { ... }: Updating CameraSession Configuration... Difference(deviceChanged=true, outputsChanged=true, sidePropsChanged=true, isActiveChanged=true)
2024-02-05 18:05:17.568 29783-29961 CameraSession           com.lucksp.rnvctest                  I  Need to rebuild CameraDevice and CameraCaptureSession...
2024-02-05 18:05:17.569 29783-29961 CameraSession           com.lucksp.rnvctest                  I  Need to rebuild CameraDevice and CameraCaptureSession...
2024-02-05 18:05:17.569 29783-29961 CameraSession           com.lucksp.rnvctest                  I  Configuring Camera #0...
2024-02-05 18:05:17.573 29783-29961 CameraManager           com.lucksp.rnvctest                  I  Camera #0: Opening...
[... cut ...] agram.android-xHyJ98z8rGGzkmk6FuXOTQ==/split_pytorch.apk!/lib/arm64-v8a:/vendor/lib64, permitted_path=/data:/mnt/expand:/data/user/0/com.instagram.android:/vendor/lib64
```

what's interesting is that the error row:

```
2024-02-05 18:05:22.491   575-30037 camera_metadata         and...e.camera.provider@2.4-service  E  validate_camera_metadata_structure: metadata is null!
``` 
goes on infinite loop, but maybe that's related to frameprocessor going at 30FPS?

So this issue seems limited to iOS...

MSchmidt, 2024-02-06:
This is quite a convoluted setup to demonstrate the crash. Does it still happen without all the hooks, context and routing? Also did you try to build your project locally with `expo prebuild` already?

mrousavy, 2024-02-06:
> This is quite a convoluted setup to demonstrate the crash. Does it still happen without all the hooks, context and routing? Also did you try to build your project locally with `expo prebuild` already?

I agree, it's really hard for me to find the true cause of this and I currently don't have time to clone the repo and run everything.

@lucksp please try to run the app through Xcode, as that will show you the logs and stacktrace then. Then post that here :)

MSchmidt, 2024-02-06:
I've build your project locally. Prepared with `npx expo prebuild --platform ios` and then build in Xcode 15.1. Everything worked fine. The uncommented your line to activate the frame processor and still everything fine with the expected output:
```
 LOG  Frame: 1080x1920 (yuv)
 LOG  Frame: 1080x1920 (yuv)
 LOG  Frame: 1080x1920 (yuv)
 LOG  Frame: 1080x1920 (yuv)
 LOG  Frame: 1080x1920 (yuv)
 LOG  Frame: 1080x1920 (yuv)
 LOG  Frame: 1080x1920 (yuv)
```

mrousavy, 2024-02-06:
Thanks @MSchmidt for testing!!! ❤️  Sounds great, so this is a local environment issue then. @lucksp try clearing your cache and try again :)

closed 2024-02-06 by mrousavy

lucksp, 2024-02-06:
Thanks for everyone's input.  A few things I'd like to comment on even though it's closed:

> it is a compatibility issue with RN Worklets and Reanimated versions. Just use the latest reanimated and latest worklets core versions, use the plugin in babel config file, clear your cache and start metro again - then it will work.

@mrousavy there is no Reanimated in the sample repo I provided:  just `react-native-vision-camera`, babel & worklets.  I am sorry you didn't have time to look into it further, I tried to make it as simple as possible:  install the `dev build` image in the repo README & test it out.  

> This is quite a convoluted setup to demonstrate the crash. Does it still happen without all the hooks, context and routing? Also did you try to build your project locally with `expo prebuild` already?

@MSchmidt This sample repo is a reproduction of what my production app does, where I originally found the issue:  navigate to a route and it opens the camera.  It's not too far off from the example app in RNVC in terms of hooks.    

>  so this is a local environment issue then. @lucksp try clearing your cache and try again :)

I thought that since my original app issue was able to be replicated in a fresh new repo provided for this bug report, of just the camera functionality, that it would be more than just a local environment issue.   It is interesting that @Mnelemos was able to replicate the issue with straight react-native tho

MSchmidt, 2024-02-06:
@Mnelemos has an issue on Android. This got a bit mixed up here. I focused on iOS and your provided GitHub Repo to match your GitHub Issue specifically. There are many ways for frame processors to fail but this one seems to be a build issue. From what I understand you used EAS to create an expo dev client. This is quite specific, but might be worth following up with. We have no idea how EAS is building the app, that's why I used Xcode to build it myself.

mrousavy, 2024-02-06:
Oh you're right @MSchmidt, I mixed that up here - @Mnelemos your issue is not related to this afaik.

@lucksp is there any chance you can run the app through Xcode? That will show us a stacktrace, SIGABRT or even logs. We need those.

lucksp, 2024-02-10:
> @lucksp is there any chance you can run the app through Xcode? That will show us a stacktrace, SIGABRT or even logs. We need those.

@mrousavy got sidetracked last week...trying to run your example on XCode, but hitting roadblocks since it's not my primary environment...

> ❌ unable to to pen configuration settings file

I'll try to get back to you on this...

To give an update on previous discussion that it's my local environment, when I try running `prebuild` locally it fails:

> › Linking   IdentaFlyDev » IdentaFlyDev
> ❌  ld: building for iOS Simulator, but linking in object file built for iOS, file '/Users/lucksp/dev/myflyid/node_modules/react-native-fast-tflite/ios/TensorFlowLiteC.framework/TensorFlowLiteC' for architecture arm64

Any ideas what this is about?

lucksp, 2024-02-10:
> I've build your project locally. Prepared with `npx expo prebuild --platform ios` and then build in Xcode 15.1. Everything worked fine. The uncommented your line to activate the frame processor and still everything fine with the expected output:

@MSchmidt did you run `prebuild` on physical device, with the camera open?  That command is only for simulator, and camera can't run on simulator.

brandoncroberts, 2024-02-23:
@lucksp @mrousavy  FWIW I have been experimenting with vision-camera in Expo and I seem to get this error also when running dev-client, but not when I ran a non dev-client build. 
ie. `npx eas build --local` and then installing the generated apk on my android device, doesn't have the errors.

lucksp, 2024-02-23:
Thanks for the update @brandoncroberts 

what version of Expo & ReactNative?  

Also can you share moire about the problem when “it crashes”?  code samples etc?

brandoncroberts, 2024-02-25:
@lucksp Actually I missed the part where you said this was only an issue for you in iPhone, I had the issue on Android but I fixed it today. 
I don't have an iPhone for testing so I don't think I can help test anything for you on this.