margelo/react-native-vision-camera #2508: 🔧 Build Fail after adding frame processor

shivlisyn, 2024-02-05:
### How were you trying to build the app?

I am using react-native-vision-camera
after adding "react-native-worklets-core": "^0.3.0" build fails in android and ios both

### Full build logs

```tsx
Execution failed for task ':react-native-vision-camera:buildCMakeDebug[arm64-v8a]
```

### Project dependencies

```json
"dependencies": {
"react-native": "0.72.7",
"react-native-vision-camera": "3.8.2",
"react-native-worklets-core": "^0.3.0"
}
```

### VisionCamera Version

3.8.2

### Target platforms

Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-02-05:
[image]

Those are not full build logs at all.

closed 2024-02-05 by mrousavy

Bashar-Alziq, 2024-02-05:
https://react-native-vision-camera.com/docs/guides/troubleshooting