margelo/react-native-vision-camera #2511: 🔧 Build error: react-native-vision-camera:configureCMakeDebug[arm64-v8a] FAILED

Thomas-Vargas, 2024-02-05:
### How were you trying to build the app?

Building my app fails when installing react-native-vision camera, citing CMake issues in the package.

### Full build logs

```tsx
$ npm run android

> aaa_rn@0.0.1 android
> react-native run-android

info A dev server is already running for this project on port 8081.
info Installing the app...

> Configure project :react-native-vision-camera
[VisionCamera] react-native-worklets-core not found, Frame Processors disabled!

> Task :react-native-vision-camera:compileDebugKotlin
w: file:///C:/Users/thomas.vargas/Documents/aaa_rn/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraDevicesManager.kt:86:19 Parameter 'eventName' is never used
w: file:///C:/Users/thomas.vargas/Documents/aaa_rn/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraDevicesManager.kt:89:23 Parameter 'count' is never used
w: file:///C:/Users/thomas.vargas/Documents/aaa_rn/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt:7:44 'RCTEventEmitter' is deprecated. Deprecated in Java
w: file:///C:/Users/thomas.vargas/Documents/aaa_rn/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt:19:28 'RCTEventEmitter' is deprecated. Deprecated in Java
w: file:///C:/Users/thomas.vargas/Documents/aaa_rn/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt:19:57 'receiveEvent(Int, String!, WritableMap?): Unit' is deprecated. Deprecated in Java
w: file:///C:/Users/thomas.vargas/Documents/aaa_rn/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt:26:28 'RCTEventEmitter' is deprecated. Deprecated in Java
w: file:///C:/Users/thomas.vargas/Documents/aaa_rn/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt:26:57 'receiveEvent(Int, String!, WritableMap?): Unit' is deprecated. Deprecated in Java
w: file:///C:/Users/thomas.vargas/Documents/aaa_rn/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt:33:28 'RCTEventEmitter' is deprecated. Deprecated in Java
w: file:///C:/Users/thomas.vargas/Documents/aaa_rn/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt:33:57 'receiveEvent(Int, String!, WritableMap?): Unit' is deprecated. Deprecated in Java
w: file:///C:/Users/thomas.vargas/Documents/aaa_rn/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt:51:28 'RCTEventEmitter' is deprecated. Deprecated in Java
w: file:///C:/Users/thomas.vargas/Documents/aaa_rn/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt:51:57 'receiveEvent(Int, String!, WritableMap?): Unit' is deprecated. Deprecated in Java
w: file:///C:/Users/thomas.vargas/Documents/aaa_rn/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt:57:28 'RCTEventEmitter' is deprecated. Deprecated in Java
w: file:///C:/Users/thomas.vargas/Documents/aaa_rn/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt:57:57 'receiveEvent(Int, String!, WritableMap?): Unit' is deprecated. Deprecated in Java
w: file:///C:/Users/thomas.vargas/Documents/aaa_rn/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt:97:28 'RCTEventEmitter' is deprecated. Deprecated in Java
w: file:///C:/Users/thomas.vargas/Documents/aaa_rn/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt:97:57 'receiveEvent(Int, String!, WritableMap?): Unit' is deprecated. Deprecated in Java
w: file:///C:/Users/thomas.vargas/Documents/aaa_rn/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt:8:46 'ReactModule' is deprecated. Deprecated in Java
w: file:///C:/Users/thomas.vargas/Documents/aaa_rn/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt:23:2 'ReactModule' is deprecated. Deprecated in Java
w: file:///C:/Users/thomas.vargas/Documents/aaa_rn/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt:675:21 Parameter 'x' is never used
w: file:///C:/Users/thomas.vargas/Documents/aaa_rn/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt:675:29 Parameter 'y' is never used
w: file:///C:/Users/thomas.vargas/Documents/aaa_rn/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/RecordingSession.kt:49:96 'constructor MediaRecorder()' is deprecated. Deprecated in Java
w: file:///C:/Users/thomas.vargas/Documents/aaa_rn/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/CameraCharacteristics+getOutputSizes.kt:22:38 'get(Int, Int): CamcorderProfile!' is deprecated. Deprecated in Java    
w: file:///C:/Users/thomas.vargas/Documents/aaa_rn/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/CameraDevice+createCaptureSession.kt:61:12 'createCaptureSessionByOutputConfigurations((Mutable)List<OutputConfiguration!>!, CameraCaptureSession.StateCallback!, Handler?): Unit' is deprecated. Deprecated in Java
w: file:///C:/Users/thomas.vargas/Documents/aaa_rn/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/RecordingSession+getRecommendedBitRate.kt:52:24 'get(Int, Int): CamcorderProfile!' is deprecated. Deprecated in Java  
w: file:///C:/Users/thomas.vargas/Documents/aaa_rn/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/RecordingSession+getRecommendedBitRate.kt:54:24 'get(Int): CamcorderProfile!' is deprecated. Deprecated in Java       

> Task :react-native-vision-camera:configureCMakeDebug[arm64-v8a] FAILED
C/C++: CMake Error at CMakeLists.txt:11 (find_package):
C/C++:   Could not find a package configuration file provided by "ReactAndroid" with
C/C++:   any of the following names:
C/C++:     ReactAndroidConfig.cmake
C/C++:     reactandroid-config.cmake
C/C++:   Add the installation prefix of "ReactAndroid" to CMAKE_PREFIX_PATH or set
C/C++:   "ReactAndroid_DIR" to a directory containing one of the above files.  If
C/C++:   "ReactAndroid" provides a separate development package or SDK, be sure it
C/C++:   has been installed.

Deprecated Gradle features were used in this build, making it incompatible with Gradle 9.0.

You can use '--warning-mode all' to show the individual deprecation warnings and determine if they come from your own scripts or plugins.

For more on this, please refer to https://docs.gradle.org/8.3/userguide/command_line_interface.html#sec:command_line_warnings in the Gradle documentation.
40 actionable tasks: 35 executed, 5 up-to-date

info 💡 Tip: Make sure that you have set up your development environment correctly, by running npx react-native doctor. To read more about doctor command visit: https://github.com/react-native-community/cli/blob/main/packages/cli-doctor/README.md#doctor 

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:configureCMakeDebug[arm64-v8a]'.
> [CXX1429] error when building with cmake using C:\Users\thomas.vargas\Documents\aaa_rn\node_modules\react-native-vision-camera\android\CMakeLists.txt: -- The C compiler identification is Clang 17.0.2
  -- The CXX compiler identification is Clang 17.0.2
[... 101 lines cut ...]
```

### VisionCamera Version

3.8.2

### Target platforms

Android

### Operating system

Windows

### Can you build the VisionCamera Example app?

Yes, I can successfully build the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

nikolablesic, 2024-02-06:
Same issue for me

mrousavy, 2024-02-06:
This error is related to react-native, maybe you are on an old react-native version - I don't know you didn't tell me your version.

Try upgrading to react-native 0.71 or higher, that's definitely supported.

nikolablesic, 2024-02-06:
Installing Microsoft Visual C++ Redistributable did the trick for me

Thomas-Vargas, 2024-02-06:
> Installing Microsoft Visual C++ Redistributable did the trick for me

I already had this installed. I uninstalled/reinstalled but this unfortunately did not do the trick.

samad-o16labs, 2024-02-15:
I am also facing the same issue, has this been fixed?

samad-o16labs, 2024-02-15:
Deleting android/.gradle folder. 
It working for me 🎉

Thank you.

mrousavy, 2024-02-15:
Awesome 💪

closed 2024-02-15 by mrousavy

joris1995, 2024-04-03:
I'm also experiencing this issue, but in my case removing the android/.gradle file triggers another error, but for react-native-screens:

`> Task :react-native-vision-camera:configureCMakeRelWithDebInfo[arm64-v8a] FAILED

> Task :react-native-vision-camera:compileReleaseKotlin
w: file:///root/repo/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView.kt:201:52 Elvis operator (?:) always returns the left operand of non-nullable type Boolean
w: file:///root/repo/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/RecordingSession.kt:51:96 'constructor MediaRecorder()' is deprecated. Deprecated in Java
w: file:///root/repo/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/CameraDevice+createCaptureSession.kt:61:12 'createCaptureSessionByOutputConfigurations((Mutable)List<OutputConfiguration!>!, CameraCaptureSession.StateCallback!, Handler?): Unit' is deprecated. Deprecated in Java
w: file:///root/repo/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/RecordingSession+getRecommendedBitRate.kt:52:24 'get(Int, Int): CamcorderProfile!' is deprecated. Deprecated in Java
w: file:///root/repo/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/RecordingSession+getRecommendedBitRate.kt:54:24 'get(Int): CamcorderProfile!' is deprecated. Deprecated in Java
w: file:///root/repo/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/utils/CamcorderProfileUtils.kt:66:42 'get(Int, Int): CamcorderProfile!' is deprecated. Deprecated in Java
w: file:///root/repo/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/utils/CamcorderProfileUtils.kt:90:42 'get(Int, Int): CamcorderProfile!' is deprecated. Deprecated in Java

FAILURE: Build completed with 2 failures.

1: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':react-native-vision-camera:configureCMakeRelWithDebInfo[arm64-v8a]'.
> Picked up _JAVA_OPTIONS: -Xmx4608m -XX:+UnlockExperimentalVMOptions -XX:ThreadStackSize=4096 -XX:CompilerThreadStackSize=4096 -XX:+UseContainerSupport -XX:+HeapDumpOnOutOfMemoryError -Dfile.encoding=UTF-8`

This is a build output from CircleCi. Do you know what I need to change in order to resolve this?