margelo/react-native-vision-camera #2512: useCameraDevice experiencing delay on Android device (Samsung galaxy s21)  💭

SocDario, 2024-02-05:
### Question

I'm using native stack navigation and I have created stack screen that is using Camera. I have experienced short delay when navigating to that screen, sometimes can take 1 second and can be a bit longer. I figured out that it's due useCameraDevice hook, then I went back to docs and found out that setting specific device can reduce that delay which didn't help, example:

```
const device = useCameraDevice('back', {
    physicalDevices: ['wide-angle-camera'],
  });
```

Then I have tried to call Camera device initialization in onFocusEffect on screen before, that fixed my issue of navigation delay, but seems like a hack and not great option. Also worth mentioning is once camera devices hook is initialized each time after that navigating to screen is without delay. So my question is what I'm doing wrong in this situation?

### What I tried

I tried to use:
```
const device = useCameraDevice('back', {
    physicalDevices: ['wide-angle-camera'],
  });
```

But I'm experiencing delay when navigating first time on that screen, then I have implemented hacky solution on screen from which I'm navigating to Camera:
```
useFocusEffect(
    useCallback(() => {
      Camera.getAvailableCameraDevices();
    }, []),
  );
```

  

### VisionCamera Version

3.8.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2024-02-06:
Hey - I cannot help you without any logs. You should've filed a bug report instead of a question.

I don't know what you mean with "delay", is the JS thread frozen? The UI Thread? Or is the camera just taking longer to initialize?

SocDario, 2024-02-08:
@mrousavy
Hey sorry for waiting.
Well I can't tell for sure if it's JS or UI thread, but on first I would tell it's JS tread most likely, since navigation get blocked until devices are fetched, but as I said, I'm not 100% sure it's JS thread. Camera init is pretty much instant. I noticed also delay on closing/navigating from Camera screen to another. I will try to create minimal repo example so you can check it.

mrousavy, 2024-02-13:
But `useCameraDevices` does not do anything apart from a JavaScript loop. There is no native call in there.

mrousavy, 2024-04-22:
Hey - I think this issue has been fixed in [VisionCamera 4.0.0](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0). 🥳 

Please try V4 and let me know if you still experience this issue;
- if not, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support the development of VisionCamera and thank me for my time spent on fixing bugs and building new features.
- if you still see this issue, please comment and we can re-open this. But please update your native logs with the native (Xcode/Android Studio) logs from running VisionCamera V4 so I can investigate this.

closed 2024-04-22 by mrousavy

gosbhs12, 2024-06-21:
did you find any solution?

SocDario, 2024-06-26:
> did you find any solution?

I have upgraded Vision Camera to the latest version and didn't have this issue anymore, but I showed example of hacky solution that worked for me before.

`useFocusEffect(
    useCallback(() => {
      Camera.getAvailableCameraDevices();
    }, []),
  );`
  
  I have moved this function into App initialization and also didn't have any issues later in the app.