margelo/react-native-vision-camera #2513: 🐛 Build Failure: ios/Frame Processor/JSINSObjectConversion.mm

DeveloperMCD, 2024-02-05:
### What's happening?

Trying to build app; worked fine before, copied Podfile from supposed working project, but perhaps he didn't have Frame Processors enabled? This is a build failure on a piece of Vision Camera IOS code.

### Reproduceable Code

```tsx
Podfile:
# Resolve react_native_pods.rb with node to allow for hoisting
require Pod::Executable.execute_command('node', ['-p',
  'require.resolve(
    "react-native/scripts/react_native_pods.rb",
    {paths: [process.argv[1]]},
  )', __dir__]).strip

platform :ios, min_ios_version_supported
prepare_react_native_project!

# If you are using a `react-native-flipper` your iOS build will fail when `NO_FLIPPER=1` is set.
# because `react-native-flipper` depends on (FlipperKit,...) that will be excluded
#
# To fix this you can also exclude `react-native-flipper` using a `react-native.config.js`
# 
# module.exports = {
#   dependencies: {
#     ...(process.env.NO_FLIPPER ? { 'react-native-flipper': { platforms: { ios: null } } } : {}),
# 
flipper_config = ENV['NO_FLIPPER'] == "1" ? FlipperConfiguration.disabled : FlipperConfiguration.enabled

linkage = ENV['USE_FRAMEWORKS']
if linkage != nil
  Pod::UI.puts "Configuring Pod with #{linkage}ally linked Frameworks".green
  use_frameworks! :linkage => linkage.to_sym
end

post_install do |installer|
  react_native_post_install(
    installer,
    # Set `mac_catalyst_enabled` to `true` in order to apply patches
    # necessary for Mac Catalyst builds
    :mac_catalyst_enabled => false
  )
  __apply_Xcode_12_5_M1_post_install_workaround(installer)
  installer.pods_project.build_configurations.each do |config|
    config.build_settings["EXCLUDED_ARCHS[sdk=iphonesimulator*]"] = "arm64"
  end
end

def shared_pods
  config = use_native_modules!
  use_frameworks! :linkage => :static
  $RNFirebaseAsStaticFramework = true

  # Flags change depending on the env values.
  flags = get_default_flags()

  use_react_native!(
    :path => config[:reactNativePath],
    # Hermes is now enabled by default. Disable by setting this flag to false.
    # Upcoming versions of React Native may rely on get_default_flags(), but
    # we make it explicit here to aid in the React Native upgrade process.
    :hermes_enabled => flags[:hermes_enabled],
    :fabric_enabled => flags[:fabric_enabled],
    # Enables Flipper.
    #
    # Note that if you have use_frameworks! enabled, Flipper will not work and
    # you should disable the next line.
    # :flipper_configuration => flipper_config,
    # An absolute path to your application root.
    :app_path => "#{Pod::Config.instance.installation_root}/.."
  )
end

target 'stackbay' do
  shared_pods
end

React Native Vision Camera: 3.8.2
React Native Worklets Core: 0.3.0
React Native Firebase: 18.8.0
```

### Relevant log output

```shell
Details:  createItemModels creation requirements should not create capability item model for a capability item model that already exists.
Function: createItemModels(for:itemModelSource:)
Thread:   <_NSMainThread: 0x600002fa42c0>{number = 1, name = main}
Please file a bug at https://feedbackassistant.apple.com with this warning message and any useful information you can provide.
2024-02-05 17:38:09.490 xcodebuild[14976:5131509] [MT] DVTAssertions: Warning in /System/Volumes/Data/SWE/Apps/DT/BuildRoots/BuildRoot11/ActiveBuildRoot/Library/Caches/com.apple.xbs/Sources/IDEFrameworks/IDEFrameworks-22267/IDEFoundation/Provisioning/Capabilities Infrastructure/IDECapabilityQuerySelection.swift:103
Details:  createItemModels creation requirements should not create capability item model for a capability item model that already exists.
Function: createItemModels(for:itemModelSource:)
Thread:   <_NSMainThread: 0x600002fa42c0>{number = 1, name = main}
Please file a bug at https://feedbackassistant.apple.com with this warning message and any useful information you can provide.
2024-02-05 17:38:09.490 xcodebuild[14976:5131509] [MT] DVTAssertions: Warning in /System/Volumes/Data/SWE/Apps/DT/BuildRoots/BuildRoot11/ActiveBuildRoot/Library/Caches/com.apple.xbs/Sources/IDEFrameworks/IDEFrameworks-22267/IDEFoundation/Provisioning/Capabilities Infrastructure/IDECapabilityQuerySelection.swift:103
Details:  createItemModels creation requirements should not create capability item model for a capability item model that already exists.
Function: createItemModels(for:itemModelSource:)
Thread:   <_NSMainThread: 0x600002fa42c0>{number = 1, name = main}
Please file a bug at https://feedbackassistant.apple.com with this warning message and any useful information you can provide.
** BUILD FAILED **

The following build commands failed:
	CompileC /Users/developer/Library/Developer/Xcode/DerivedData/stackbay-hkdvgolugzmnjnbykqfllfpapwnp/Build/Intermediates.noindex/Pods.build/Debug-iphoneos/VisionCamera.build/Objects-normal/arm64/JSINSObjectConversion.o /Users/developer/Documents/ios/stackbayapp/node_modules/react-native-vision-camera/ios/Frame\ Processor/JSINSObjectConversion.mm normal arm64 objective-c++ com.apple.compilers.llvm.clang.1_0.compiler (in target 'VisionCamera' from project 'Pods')
(1 failure)
```

### Camera Device

```json
N/A
```

### Device

iPhone 13

### VisionCamera Version

3.8.2

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

DeveloperMCD, 2024-02-05:
If you need any other information, or want me to try anything, please let me know.

mrousavy, 2024-02-06:
If you can build without Firebase (and wihtout static linkage), I think this issue is not directly related to VisionCamera

DeveloperMCD, 2024-02-06:
I just want to be able to use Firebase and Vision Camera. Firebase might require static linkage, which seems to have caused an error during compiling. The error is in JSINSObjectConversion.mm, according to the output. Isn't that something that can be fixed, theoretically?

mrousavy, 2024-02-06:
> The error is in JSINSObjectConversion.mm, according to the output. Isn't that something that can be fixed, theoretically?

Well I don't know, I never tried to link it statically and in the error logs you showed, it doesn't seem like there is any specific error happening, so this would definitely require me to investigate this thoroughly (couple of hours of effort).

Maybe try clearing all your cache instead, as I think you mentioned this not working on your machine while it worked fine on someone else's machine.

```
rm -rf Pods Podfile.lock build
pod deintegrate
pod install
pod update
```

tonymatheus, 2024-03-21:
+ 1

eduardorfreitas93, 2024-03-28:
In your Podfile, add this line.

```
installer.pods_project.targets.each do |target|
  target.build_configurations.each do |config|
    ...
    config.build_settings['HEADER_SEARCH_PATHS'] ||= '$(inherited) '
    config.build_settings['HEADER_SEARCH_PATHS'] << '"${PODS_ROOT}/../../node_modules/react-native/ReactCommon" '
  end
end
```

eduardorfreitas93, 2024-03-28:
Or add to the library's dependencies, you can use this line:

VisionCamera.podspec

```
...
s.dependency "React"
s.dependency "React-Core"
s.dependency "React-callinvoker"
s.dependency "ReactCommon" <- here
...
```

mrousavy, 2024-04-22:
Hey - I think this issue has been fixed in [VisionCamera 4.0.0](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0). 🥳 

Please try V4 and let me know if you still experience this issue;
- if not, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support the development of VisionCamera and thank me for my time spent on fixing bugs and building new features.
- if you still see this issue, please comment and we can re-open this. But please update your native logs with the native (Xcode/Android Studio) logs from running VisionCamera V4 so I can investigate this.

closed 2024-04-22 by mrousavy

Benv547, 2024-04-23:
J'ai encore le problème en 4.0.1 avec react-native-worklets-core en 1.1.1 (exactement le même problème que OP)

`CompileC /Users/benv547/Library/Developer/Xcode/DerivedData/poccodeocr-bspdfsfthhwuaycpogdnuoythlxc/Build/Intermediates.noindex/Pods.build/Debug-iphoneos/VisionCamera.build/Objects-normal/arm64/FrameProcessor.o /Users/benv547/Documents/Projects/ReactNative/poccodeocr/node_modules/react-native-vision-camera/ios/FrameProcessors/FrameProcessor.mm normal arm64 objective-c++ com.apple.compilers.llvm.clang.1_0.compiler (in target 'VisionCamera' from project 'Pods')
    cd /Users/benv547/Documents/Projects/ReactNative/poccodeocr/ios/Pods
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang -x objective-c++ -ivfsstatcache /Users/benv547/Library/Developer/Xcode/DerivedData/SDKStatCaches.noindex/iphoneos17.4-21E210-99a79c02cfa501706f5b564e46d84afe.sdkstatcache -fmessage-length\=0 -fdiagnostics-show-note-include-stack -fmacro-backtrace-limit\=0 -fno-color-diagnostics -fmodules-prune-interval\=86400 -fmodules-prune-after\=345600 -fbuild-session-file\=/Users/benv547/Library/Developer/Xcode/DerivedData/ModuleCache.noindex/Session.modulevalidation -fmodules-validate-once-per-build-session -Wnon-modular-include-in-framework-module -Werror\=non-modular-include-in-framework-module -Wno-trigraphs -Wno-missing-field-initializers -Wno-missing-prototypes -Werror\=return-type -Wdocumentation -Wunreachable-code -Wno-implicit-atomic-properties -Werror\=deprecated-objc-isa-usage -Wno-objc-interface-ivars -Werror\=objc-root-class -Wno-arc-repeated-use-of-weak -Wimplicit-retain-self -Wno-non-virtual-dtor -Wno-overloaded-virtual -Wno-exit-time-destructors -Wduplicate-method-match -Wno-missing-braces -Wparentheses -Wswitch -Wunused-function -Wno-unused-label -Wno-unused-parameter -Wunused-variable -Wunused-value -Wempty-body -Wuninitialized -Wconditional-uninitialized -Wno-unknown-pragmas -Wno-shadow -Wno-four-char-constants -Wno-conversion -Wconstant-conversion -Wint-conversion -Wbool-conversion -Wenum-conversion -Wno-float-conversion -Wnon-literal-null-conversion -Wobjc-literal-conversion -Wshorten-64-to-32 -Wno-newline-eof -Wno-selector -Wno-strict-selector-match -Wundeclared-selector -Wdeprecated-implementations -Wno-c++11-extensions -Wno-implicit-fallthrough -fstrict-aliasing -Wprotocol -Wdeprecated-declarations -Winvalid-offsetof -Wno-sign-conversion -Winfinite-recursion -Wmove -Wcomma -Wblock-capture-autoreleasing -Wstrict-prototypes -Wrange-loop-analysis -Wno-semicolon-before-method-body -Wunguarded-availability -index-store-path /Users/benv547/Library/Developer/Xcode/DerivedData/poccodeocr-bspdfsfthhwuaycpogdnuoythlxc/Index.noindex/DataStore @/Users/benv547/Library/Developer/Xcode/DerivedData/poccodeocr-bspdfsfthhwuaycpogdnuoythlxc/Build/Intermediates.noindex/Pods.build/Debug-iphoneos/VisionCamera.build/Objects-normal/arm64/af3fcb34312c57c0f52879cdce924b91-common-args.resp -include /Users/benv547/Documents/Projects/ReactNative/poccodeocr/ios/Pods/Target\ Support\ Files/VisionCamera/VisionCamera-prefix.pch -MMD -MT dependencies -MF /Users/benv547/Library/Developer/Xcode/DerivedData/poccodeocr-bspdfsfthhwuaycpogdnuoythlxc/Build/Intermediates.noindex/Pods.build/Debug-iphoneos/VisionCamera.build/Objects-normal/arm64/FrameProcessor.d --serialize-diagnostics /Users/benv547/Library/Developer/Xcode/DerivedData/poccodeocr-bspdfsfthhwuaycpogdnuoythlxc/Build/Intermediates.noindex/Pods.build/Debug-iphoneos/VisionCamera.build/Objects-normal/arm64/FrameProcessor.dia -c /Users/benv547/Documents/Projects/ReactNative/poccodeocr/node_modules/react-native-vision-camera/ios/FrameProcessors/FrameProcessor.mm -o /Users/benv547/Library/Developer/Xcode/DerivedData/poccodeocr-bspdfsfthhwuaycpogdnuoythlxc/Build/Intermediates.noindex/Pods.build/Debug-iphoneos/VisionCamera.build/Objects-normal/arm64/FrameProcessor.o -index-unit-output-path /Pods.build/Debug-iphoneos/VisionCamera.build/Objects-normal/arm64/FrameProcessor.o

SwiftDriverJobDiscovery normal arm64 Compiling CameraSessionDelegate.swift, CameraView.swift, CameraView+Focus.swift, CameraView+RecordVideo.swift, CameraView+TakePhoto.swift, CameraView+TakeSnapshot.swift, CameraView+Zoom.swift, CameraViewManager.swift (in target 'VisionCamera' from project 'Pods')

warning: Run script build phase 'Bundle React Native code and images' will be run during every build because it does not specify any outputs. To address this warning, either add output dependencies to the script phase, or configure it to run in every build by unchecking "Based on dependency analysis" in the script phase. (in target 'poccodeocr' from project 'poccodeocr')
warning: Run script build phase '[CP-User] [Hermes] Replace Hermes for the right configuration, if needed' will be run during every build because it does not specify any outputs. To address this warning, either add output dependencies to the script phase, or configure it to run in every build by unchecking "Based on dependency analysis" in the script phase. (in target 'hermes-engine' from project 'Pods')
warning: Run script build phase '[CP-User] [RN]Check rncore' will be run during every build because it does not specify any outputs. To address this warning, either add output dependencies to the script phase, or configure it to run in every build by unchecking "Based on dependency analysis" in the script phase. (in target 'React-Fabric' from project 'Pods')

** BUILD FAILED **

[... cut ...]
	CompileC /Users/benv547/Library/Developer/Xcode/DerivedData/poccodeocr-bspdfsfthhwuaycpogdnuoythlxc/Build/Intermediates.noindex/Pods.build/Debug-iphoneos/VisionCamera.build/Objects-normal/arm64/JSINSObjectConversion.o /Users/benv547/Documents/Projects/ReactNative/poccodeocr/node_modules/react-native-vision-camera/ios/FrameProcessors/JSINSObjectConversion.mm normal arm64 objective-c++ com.apple.compilers.llvm.clang.1_0.compiler (in target 'VisionCamera' from project 'Pods')
(1 failure)`

I've tried eduardorfreitas93's resolutions but nothing changes.
Available if you need more information!

mrousavy, 2024-04-23:
fixed on latest main.
didn't do a release yet, am on vacation in italy

RutvinBhanderi, 2024-05-04:
Still getting same isuue in version 4.0.1

BarbieriMatheus-T10, 2024-05-06:
"@react-native-firebase/analytics": "^18.3.0",
    "@react-native-firebase/app": "^18.3.0",
    "@react-native-firebase/remote-config": "^18.3.0",
    "react": "18.2.0",
    "react-native": "0.72.13",

RutvinBhanderi, 2024-05-07:
The issue is with the ios/frame processor/JSINSObjectConversion file. I Downloaded the latest code and replaced the frame processor folder which is inside node_modules/react-native-vision-camera/ios/ with the downloaded code. then it is working fine.

dowatugkins, 2024-05-09:
I just installed 4.0.3 - I'm not sure exactly what version @mrousavy was talking about while on vacation as I was running 4.0.1 before I just updated - but it totally worked. Thank you for your effort getting this fixed.