margelo/react-native-vision-camera #2515: 🐛 Width & height of taken photo is incorrect on Android

retyui, 2024-02-06:
### What's happening?

I use Samsung s10 device with Android 10 to take photo and save it to the gallery 

the result  of  `.takePhoto()`:

```java
file.height: 3024
file.width: 4032
file.orientation: "landscape-right"
```

but when I checked the photo in the gallery I see a different information `width:3024 x height:4032`

[image]

---

I think need to take in account an exif orientation and swap a dimension like here: https://github.com/react-native-image-picker/react-native-image-picker/commit/13cdc7c1a19fa00cae17a85560126f2e1a18e4e3

### Reproduceable Code

```tsx
const file = await camera.current.takePhoto()
await CameraRoll.save(`file://${file.path}`, {
  type: 'photo',
})
```

### Relevant log output

```shell
no
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-right",
  "hardwareLevel": "full",
  "maxZoom": 8,
  "minZoom": 1,
  "maxExposure": 20,
  "supportsLowLightBoost": true,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": false,
  "minFocusDistance": 10,
  "minExposure": -20,
  "name": "BACK (0)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Samsung s10

### VisionCamera Version

3.8.2

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-02-06:
Orientation is tracked here: https://github.com/mrousavy/react-native-vision-camera/issues/1891

You're getting width and height in natural sensor orientation. At the moment, proper orientation is not yet implemented.

closed 2024-02-06 by mrousavy

retyui, 2024-02-06:
well I funded 69$, I hope it might be fixed soon

mrousavy, 2024-02-06:
Thank you! Yea, stability issues are first priority now, then we'll see about orientation :)