margelo/react-native-vision-camera #2520: Use `AsyncExec` every time it finishes executing 💭

tomerh2001, 2024-02-06:
### Question

I have a document detection model I want to run but it's slow, I don't want to hard-lock to running it once every second (with `runAtTargetFps(1, ...)`) but rather want it to execute sequentially in a different thread.

I.E. the next call to my callback should be immediately after the previous call has finished, independent of the FPS or what frame it is  

### What I tried

I know I can technically wrap some logic into `runAsync` that'll check if the previous call has finished - or use a recursive call in a one-off thing

But institutively this utility seems (to me at least) useful when dealing with frame processing, so maybe a sister function to `runAsync` and `runAtTargetFps` could be introduced?

### VisionCamera Version

8.0.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2024-02-07:
Also, I'm not sure if I get your question. You want to run something on another Thread, then use `runAsync`. That's what I built it for. Did you try using `runAsync`?

oscar-b, 2024-02-08:
I assume he means it should only process 1 frame sequentially, and drop any frames that are aquired whilst still processing.

mrousavy, 2024-02-08:
Yep, this is exactly what `runAsync` does 👍

closed 2024-02-08 by mrousavy

tomerh2001, 2024-02-08:
> Also, I'm not sure if I get your question. You want to run something on another Thread, then use `runAsync`. That's what I built it for. Did you try using `runAsync`?

Since `runAsync` gets a `frame` I assumed it does something like pushing every frame to a queue to be processed. So this means if I use `runAsync` and my `detect` function takes 1.5s than it does:
1. `detect` on the frame at 0s
2. `detect` on the frame at 1.5s
3. `detect` on the frame at 3s
4. so on...
?

Also, when I try to use `runAsync` it just crashes without any error, this is my code:
```
	const frameProcessor = useFrameProcessor(frame => {
		'worklet';

		if (!model || !isCameraInitialized || !isCameraActive) {
			return;
		}

		runAsync(frame, () => {
			'worklet';

			const resizedFrame = resize(frame, {
				scale: {
					height: 384,
					width: 384,
				},
				dataType: 'uint8',
				pixelFormat: 'rgb',
			});

			/**
			* A normalized version of the frame where each pixel is a value between 0 and 1.
			* @type {Float32Array}
			*/
			const normalizedData = new Float32Array(resizedFrame.length);
			for (let i = 0; i < resizedFrame.length; i++) {
				normalizedData[i] = resizedFrame[i] / 255.0;
			}

			/**
			* The output of the document detection model, it is a flattened 1x348x348x2 tensor.
			* The channels are as follows:
			* - The first channel is the not_document channel.
			* - The second channel is the document channel.
			* @type {Float32Array}
			*/
			const output = model.runSync([normalizedData])[0];

			// Calculate the document confidence by iterating over the output
			// This is done by summing the values of the document channel and dividing by the total number of pixels in the document channel
			let documentConfidence = 0;
			for (let i = 0; i < output.length; i += 2) {
				const y = Math.floor(i / 348);

                                // skip x% of the height
				if (y < 348 * 0.2) {
					continue;
				}

				documentConfidence += output[i + 1];
			}

			documentConfidence /= output.length / 2;
			console.log({documentConfidence});

			// If the confidence is high enough, grab the image
			if (documentConfidence > 2) {
				grabImageWorklet(true);
			}
		});
	},	[model, isCameraInitialized, isCameraActive]);
```

tomerh2001, 2024-02-09:
Update:
this works:
```
runAsync(frame, Worklets.createRunInJsFn(() => {
    console.log(frame.width);
});
```
I get the width printed endlessly

But when I try to do something with the frame like:
```
runAsync(frame, Worklets.createRunInJsFn(() => {
	const resizedFrame = resize(frame, {
		scale: {
			height: 384,
			width: 384,
		},
		dataType: 'uint8',
		pixelFormat: 'rgb',
	});

	console.log(resizedFrame.length);
});
```

Nothing happens - it doesn't even crash, just nothing happens

mrousavy, 2024-02-09:
Ah yeah then no, you dont need reanimated.

But you shouldn't use runOnJS, just runAsync:

```ts
runAsync(frame, () => {
  'worklet'
  // your code
})
```

tomerh2001, 2024-02-09:
> Ah yeah then no, you dont need reanimated.
> 
> But you shouldn't use runOnJS, just runAsync:
> 
> ```ts
> runAsync(frame, () => {
>   'worklet'
>   // your code
> })
> ```

That doesn't work, and it straight up crashes... this is my crash report from flipper:

```
*** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***

Unknown

*** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***

Build fingerprint: 'google/cheetah/cheetah:14/UQ1A.240105.004/11206848:user/release-keys'
Revision: 'MP1.0'
ABI: 'arm64'
Timestamp: 2024-02-09 19:49:09.838956168+0200
Process uptime: 19s
Cmdline: com.gamapaymobile
pid: 22746, tid: 22956, name: mqt_js  >>> com.gamapaymobile <<<
uid: 10398
tagged_addr_ctrl: 0000000000000001 (PR_TAGGED_ADDR_ENABLE)
signal 11 (SIGSEGV), code 2 (SEGV_ACCERR), fault addr 0x0000006ebd02d42d
    x0  0000000000000001  x1  00000074e5511248  x2  0000000000000000  x3  0000000000000000
    x4  00000074e7805f10  x5  b400007537dc59e8  x6  0000000000000000  x7  00000073a34d88e3
    x8  0000006e7e02c4e8  x9  0000006ebd02d408  x10 0000000000000003  x11 00000000dd000000
    x12 00000074e78061c8  x13 0000000000000051  x14 0000000000000001  x15 0000006e7e022570
    x16 0000000000000051  x17 0000000000000600  x18 00000074e769c000  x19 0000006e7dc00000
    x20 00000074e7805d68  x21 00000074e7809000  x22 00000074e5511280  x23 fffa000000000000
    x24 ffff006e7e02c4e8  x25 0000000000bfef50  x26 0000000000000000  x27 b400007537dc5990
    x28 00000074e5511138  x29 00000074e7805d40
    lr  0000007448c13480  sp  00000074e7805d40  pc  0000007448c1349c  pst 0000000060001000
72 total frames
backtrace:
      #00 pc 000000000016149c  /data/app/~~FSywDNfZ_bfstefKi9tnAw==/com.gamapaymobile-atp5BqxQzqvZiGGhJ5L2LQ==/base.apk!libhermes.so (offset 0x2649000) (BuildId: db6fd99979b1c632a8aa7fc3fb1374981dc2c7c4)
      #01 pc 00000000000ce538  /data/app/~~FSywDNfZ_bfstefKi9tnAw==/com.gamapaymobile-atp5BqxQzqvZiGGhJ5L2LQ==/base.apk!libhermes.so (offset 0x2649000) (BuildId: db6fd99979b1c632a8aa7fc3fb1374981dc2c7c4)
      #02 pc 00000000000cc7c0  /data/app/~~FSywDNfZ_bfstefKi9tnAw==/com.gamapaymobile-atp5BqxQzqvZiGGhJ5L2LQ==/base.apk!libhermes.so (offset 0x2649000) (BuildId: db6fd99979b1c632a8aa7fc3fb1374981dc2c7c4)
      #03 pc 0000000000106888  /data/app/~~FSywDNfZ_bfstefKi9tnAw==/com.gamapaymobile-atp5BqxQzqvZiGGhJ5L2LQ==/base.apk!libhermes.so (offset 0x2649000) (BuildId: db6fd99979b1c632a8aa7fc3fb1374981dc2c7c4)
      #04 pc 00000000000ef104  /data/app/~~FSywDNfZ_bfstefKi9tnAw==/com.gamapaymobile-atp5BqxQzqvZiGGhJ5L2LQ==/base.apk!libhermes.so (offset 0x2649000) (BuildId: db6fd99979b1c632a8aa7fc3fb1374981dc2c7c4)
      #05 pc 00000000000ec554  /data/app/~~FSywDNfZ_bfstefKi9tnAw==/com.gamapaymobile-atp5BqxQzqvZiGGhJ5L2LQ==/base.apk!libhermes.so (offset 0x2649000) (BuildId: db6fd99979b1c632a8aa7fc3fb1374981dc2c7c4)
      #06 pc 00000000000ce830  /data/app/~~FSywDNfZ_bfstefKi9tnAw==/com.gamapaymobile-atp5BqxQzqvZiGGhJ5L2LQ==/base.apk!libhermes.so (offset 0x2649000) (BuildId: db6fd99979b1c632a8aa7fc3fb1374981dc2c7c4)
      #07 pc 0000000000095060  /data/app/~~FSywDNfZ_bfstefKi9tnAw==/com.gamapaymobile-atp5BqxQzqvZiGGhJ5L2LQ==/base.apk!libhermes.so (offset 0x2649000) (BuildId: db6fd99979b1c632a8aa7fc3fb1374981dc2c7c4)
      #08 pc 000000000005a850  /data/app/~~FSywDNfZ_bfstefKi9tnAw==/com.gamapaymobile-atp5BqxQzqvZiGGhJ5L2LQ==/base.apk!libVisionCamera.so (offset 0x11b52000) (RNWorklet::JsiWorklet::call(std::__ndk1::shared_ptr<facebook::jsi::Function>, facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long)+332) (BuildId: 4024460c6f4361b8a3019c1cb6706c086a77f83f)
      #09 pc 0000000000059dd8  /data/app/~~FSywDNfZ_bfstefKi9tnAw==/com.gamapaymobile-atp5BqxQzqvZiGGhJ5L2LQ==/base.apk!libVisionCamera.so (offset 0x11b52000) (RNWorklet::WorkletInvoker::call(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long)+140) (BuildId: 4024460c6f4361b8a3019c1cb6706c086a77f83f)
      #10 pc 00000000001a02a4  /data/app/~~FSywDNfZ_bfstefKi9tnAw==/com.gamapaymobile-atp5BqxQzqvZiGGhJ5L2LQ==/base.apk!librnworklets.so (offset 0x5315000) (RNWorklet::JsiObjectWrapper::setFunctionValue(facebook::jsi::Runtime&, facebook::jsi::Value const&)::'lambda'(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long)::operator()(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long) const+68) (BuildId: 371ddf5f821b2248e04c5e7ad2fab66b8f369f28)
      #11 pc 00000000001a023c  /data/app/~~FSywDNfZ_bfstefKi9tnAw==/com.gamapaymobile-atp5BqxQzqvZiGGhJ5L2LQ==/base.apk!librnworklets.so (offset 0x5315000) (BuildId: 371ddf5f821b2248e04c5e7ad2fab66b8f369f28)
      #12 pc 00000000001a0188  /data/app/~~FSywDNfZ_bfstefKi9tnAw==/com.gamapaymobile-atp5BqxQzqvZiGGhJ5L2LQ==/base.apk!librnworklets.so (offset 0x5315000) (facebook::jsi::Value std::__ndk1::__invoke_void_return_wrapper<facebook::jsi::Value>::__call<RNWorklet::JsiObjectWrapper::setFunctionValue(facebook::jsi::Runtime&, facebook::jsi::Value const&)::'lambda'(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long)&, facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long>(RNWorklet::JsiObjectWrapper::setFunctionValue(facebook::jsi::Runtime&, facebook::jsi::Value const&)::'lambda'(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, un [... cut ...]
      #69 pc 000000000017919c  /data/app/~~FSywDNfZ_bfstefKi9tnAw==/com.gamapaymobile-atp5BqxQzqvZiGGhJ5L2LQ==/base.apk!librnworklets.so (offset 0x5315000) (BuildId: 371ddf5f821b2248e04c5e7ad2fab66b8f369f28)
      #70 pc 00000000000c9ccc  /apex/com.android.runtime/lib64/bionic/libc.so (__pthread_start(void*)+204) (BuildId: 19c32900d9d702c303d2b4164fbba76c)
      #71 pc 000000000005db00  /apex/com.android.runtime/lib64/bionic/libc.so (__start_thread+64) (BuildId: 19c32900d9d702c303d2b4164fbba76c)
```

jslok, 2024-02-10:
runAsync gives me an error: `Frame Processor Error: Regular javascript function '' cannot be shared. Try decorating the function with the 'worklet' keyword to allow the javascript function to be used as a worklet., js engine: VisionCamera`

```
  const frameProcessor = useFrameProcessor((frame) => {
    'worklet'

    runAsync(frame, () => {
      'worklet'
      console.log("I'm running asynchronously, possibly at a lower FPS rate!")
    })
  }, [])
  ```

mrousavy, 2024-02-13:
Did you guys try `runAsync` in the example app? It works for me there.

Just compare your versions with the example app, maybe it is a REA compatibility issue, a Babel issue, or a RN Worklets issue?

mattijsf, 2024-02-14:
> runAsync gives me an error: `Frame Processor Error: Regular javascript function '' cannot be shared. Try decorating the function with the 'worklet' keyword to allow the javascript function to be used as a worklet., js engine: VisionCamera`
> 
> ```
>   const frameProcessor = useFrameProcessor((frame) => {
>     'worklet'
> 
>     runAsync(frame, () => {
>       'worklet'
>       console.log("I'm running asynchronously, possibly at a lower FPS rate!")
>     })
>   }, [])
> ```
> 
> Using 3.9.0 beta 3. runAtTargetFps works well.

I also have this exact problem and `runAtTargetFps` works (but I get an occasional stutter so I'd like to try `runAsync`).
- react-native-reanimated@3.6.1
- react-native-vision-camera@3.8.2
- react-native-worklets-core@0.3.0

jslok, 2024-02-17:
`yarn start --reset cache` solved the worklet error for me.

TheGhostFish, 2024-04-24:
I think I found the issue.

The current behavior is:
The resize plugin returns its data as a SharedArray.
The SharedArray runtime is retrieved from the "VisionCamera" worklet context, where the frame processor runs on.
Hence, it can only be used in frame processor function directly, or in runAtTargetFps function as it uses the same context.

I did a dirty workaround that fixed the issue on my side, by removing the "VisionCamera.async" context, and making both SharedArray and runAsync use the default worklet context.
A clean solution would be to automatically detect the runtime on SharedArray creation.

```
diff --git a/node_modules/react-native-vision-camera/android/src/main/cpp/frameprocessors/java-bindings/JSharedArray.cpp b/node_modules/react-native-vision-camera/android/src/main/cpp/frameprocessors/java-bindings/JSharedArray.cpp
index 46f94f8..bdef5b7 100644
--- a/node_modules/react-native-vision-camera/android/src/main/cpp/frameprocessors/java-bindings/JSharedArray.cpp
+++ b/node_modules/react-native-vision-camera/android/src/main/cpp/frameprocessors/java-bindings/JSharedArray.cpp
@@ -30,7 +30,7 @@ JSharedArray::JSharedArray(const jni::alias_ref<jhybridobject>& javaThis, const
   _javaPart = jni::make_global(javaThis);
 
 #if VISION_CAMERA_ENABLE_FRAME_PROCESSORS
-  jsi::Runtime& runtime = proxy->cthis()->getWorkletRuntime();
+  jsi::Runtime& runtime = RNWorklet::JsiWorkletContext::getDefaultInstance()->getWorkletRuntime();
 #else
   jsi::Runtime& runtime = *proxy->cthis()->getJSRuntime();
 #endif
diff --git a/node_modules/react-native-vision-camera/src/FrameProcessorPlugins.ts b/node_modules/react-native-vision-camera/src/FrameProcessorPlugins.ts
index c8bd29d..e44be10 100644
--- a/node_modules/react-native-vision-camera/src/FrameProcessorPlugins.ts
+++ b/node_modules/react-native-vision-camera/src/FrameProcessorPlugins.ts
@@ -110,8 +110,7 @@ try {
   }
 
   isAsyncContextBusy = Worklets.createSharedValue(false)
-  const asyncContext = Worklets.createContext('VisionCamera.async')
-  runOnAsyncContext = asyncContext.createRunAsync((frame: Frame, func: () => void) => {
+  runOnAsyncContext = Worklets.defaultContext.createRunAsync((frame: Frame, func: () => void) => {
     'worklet'
     try {
       // Call long-running function
```

@mrousavy Please let me know if I'm missing something. Thanks.

mrousavy, 2024-04-29:
Hey @TheGhostFish - unfortunately this isn't as simple as it sounds. Moving SharedArrays between contexts is really dangerous as the data is just not thread safe. So the real solution is to implement SharedArray proxying/copying.

Your patch might work as a temporary solution though - where it previously was bound to FP runtime only, you can not use it in Worklets default context only.

willadamskeane, 2024-05-07:
@mrousavy Any other ideas to workaround here? I'm running into the same issue, trying to use the resize plugin inside runAsync. The plugin runs fine but trying to access the returned value crashes the app.

jslok, 2025-04-25:
I used the patch-package from @TheGhostFish and from what I can tell, it works for my case with no issues that I can see.

jslok, 2025-09-19:
Just updating this issue still exists on 0.81.4 for runAsync and the solution above still works though the files have changed.