margelo/react-native-vision-camera #2521: Frame Processor threw an error: Exception in HostObject::get for prop 'bytesPerRow': java.lang.ArrayIndexOutOfBoundsException: length=0; index=0🐛

vemarav, 2024-02-06:
### What's happening?

```js
 const frameProcessor = useFrameProcessor(frame => {
    'worklet';
    const inputBytesRow = frame.bytesPerRow // --> throws error  Frame Processor threw an error: Exception in HostObject::get for prop 'bytesPerRow': java.lang.ArrayIndexOutOfBoundsException: length=0; index=0🐛 
    // ...
 }, [])
```

### Reproduceable Code

```tsx
<Camera
        device={device}
        format={format}
        isActive
        resizeMode="cover"
        style={StyleSheet.absoluteFill}
        frameProcessor={frameProcessor}
/>
```

### Relevant log output

```shell
LOG  Running "fitbot" with {"rootTag":61}
 LOG  Installing bindings...
 LOG  Successfully installed!
 LOG  Loading react-native-worklets-core...
 LOG  Worklets loaded successfully
 LOG  Loading Tensorflow Lite Model 1
 LOG  Resolved Model path: http://localhost:8081/assets/pose-detection-fast.tflite?platform=android&hash=361a28992b649751d6d66cd8cfb1e2a6
 LOG  Model loaded!
 LOG  Model: uint8 [1,192,192,3] -> float32 [1,1,17,3]

ERROR Frame Processor threw an error: Exception in HostObject::get for prop 'bytesPerRow': java.lang.ArrayIndexOutOfBoundsException: length=0; index=0🐛
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "limited",
  "maxZoom": 4,
  "minZoom": 1,
  "maxExposure": 200,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minExposure": -200,
  "name": "FRONT (1)",
  "hasFlash": false,
  "hasTorch": false,
  "position": "front",
  "id": "1"
}
```

### Device

Samsung GALAXY A31

### VisionCamera Version

3.8.2

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-02-07:
You are probably in format `"native"`. Try setting `pixelFormat="yuv"` on the Camera.

mrousavy, 2024-02-13:
Fixed in 3.9.x 👍

svjatoslavtt, 2024-02-14:
Hi, I had the same problem and `[Release 3.9.0-beta.3]` fixed it, but now I have another problem with `Android`. It reboots when I open the camera. In my application, we use `react-native-vision-camera` combined with `@ismaelmoreiraa/vision-camera-ocr` to read text in real time.
Could you please give me possible solutions to fix this?

mrousavy, 2024-02-15:
@svjatoslavtt did 3.9.0-beta.5 fix the issue for you? :)