margelo/react-native-vision-camera #2524: 🐛 IOS camera frozen when app goes to background or slide down menu bar more than one

e-vural, 2024-02-07:
### What's happening?

- Start video recording
- App goes to background and open again its working fine but not all time. 
- Then second time goes to background (Most of times camera frozen at that step. But sometimes need third times goes to background) 

I only once slide down menu and camera , buttons ,back swipe disabled. Only countdown active.

https://github.com/mrousavy/react-native-vision-camera/assets/11753784/2940035a-7f11-4322-a89f-97eebb7d1468

https://github.com/mrousavy/react-native-vision-camera/assets/11753784/9d6d9da6-35e3-4b64-bcb2-f1e6260a736d

### Reproduceable Code

```tsx
/**I used app state. */
const appState = useAppState()
const isFocused = useIsFocused()
const isActive = isFocused && appState === "active"

<Camera
    onError={(er) => {
        console.log("camera error ", er)
    }}
    onStopped={(e) => {
        console.log("onStopped", e)
    }}
    exposure={0}
    isActive={isActive}
    device={device}
    ref={camera}
    format={format}
    video={true}
    audio={true}
    enableBufferCompression={true}
    resizeMode={"contain"}
    style={{
        aspectRatio: ASPECT_RATIO,
        zIndex: -1,
        width: "100%",
        height: "100%",
    }}
/>
```

### Relevant log output

```shell
2024-02-07 15:47:37.561941+0300 [1019:279157] [native] VisionCamera.didSetProps(_:): Updating 1 props: [isActive]
2024-02-07 15:47:37.561976+0300 [1019:279157] [native] VisionCamera.configure(_:): configure { ... }: Waiting for lock...
2024-02-07 15:47:37.563281+0300 [1019:279658] [native] VisionCamera.configure(_:): configure { ... }: Updating CameraSession Configuration... Difference(inputChanged: false, outputsChanged: false, videoStabilizationChanged: false, orientationChanged: false, formatChanged: false, sidePropsChanged: false, torchChanged: false, zoomChanged: false, exposureChanged: false, audioSessionChanged: false)
2024-02-07 15:47:37.750399+0300 1019:279658] [native] VisionCamera.onCameraStopped(): Camera stopped!
2024-02-07 15:47:38.755402+0300[1019:279652] [Common] Snapshot request 0x2808b35a0 complete with error: <NSError: 0x2817761c0; domain: FBSSceneSnapshotErrorDomain; code: 3; reason: "the request was canceled">
```

### Camera Device

```json
{
  "position": "back",
  "minZoom": 1,
  "hasTorch": true,
  "hardwareLevel": "full",
  "isMultiCam": false,
  "sensorOrientation": "landscape-right",
  "maxExposure": 8,
  "supportsLowLightBoost": false,
  "supportsRawCapture": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "formats": [],
  "name": "Back Camera",
  "hasFlash": true,
  "minExposure": -8,
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:0",
  "maxZoom": 16,
  "minFocusDistance": 12
}
```

### Device

iPhone XS, iPhone 15 Pro Max, iPhone 13 Pro

### VisionCamera Version

3.9.0-beta.1

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

e-vural, 2024-02-07:
I guess old iphones have that. problem. Iphone 15 promax works fine.

mrousavy, 2024-02-13:
You need to post more logs. 4 lines of log output is not enough for me to see what's going on.

closed 2024-03-01 by mrousavy

pjcape, 2024-08-11:
Hi @e-vural , how did u solve this issue ? I have tried other libraries like tesla camera kit and I face the same issue. Currently tested on iPhone 13 and iPhone 6.