margelo/react-native-vision-camera #2527: Task ':react-native-vision-camera:generateCodegenSchemaFromJavaScript' uses this output of task ':react-native-vision-camera:prepareHeaders' without declaring an explicit or implicit dependency. This can lead to incorrect results being produced, depending on what order the tasks are executed.

mertavcioglu, 2024-02-08:
### How were you trying to build the app?

My teammate worked on this library on macOS and I have to work on afterwards, my app does not builds.

[image]

I tried to build with yarn start or in android folder: ./gradlew build, both of them did not work.

I tried to add

`<uses-permission android:name="android.permission.CAMERA" />`

to android > app > src > main > AndroidManifest.xml, but it does not worked.

### Full build logs

```tsx
Reason: Task ':react-native-vision-camera:generateCodegenSchemaFromJavaScript' uses this output of task ':react-native-vision-camera:prepareHeaders' without declaring an explicit or implicit dependency. This can lead to incorrect results being produced, depending on what order the tasks are executed.

    Possible solutions:
      1. Declare task ':react-native-vision-camera:prepareHeaders' as an input of ':react-native-vision-camera:generateCodegenSchemaFromJavaScript'.
      2. Declare an explicit dependency on ':react-native-vision-camera:prepareHeaders' from ':react-native-vision-camera:generateCodegenSchemaFromJavaScript' using Task#dependsOn.
      3. Declare an explicit dependency on ':react-native-vision-camera:prepareHeaders' from ':react-native-vision-camera:generateCodegenSchemaFromJavaScript' using Task#mustRunAfter.
```

### Project dependencies

```json
"dependencies": {
    "@bam.tech/react-native-image-resizer": "^3.0.7",
    "@georstat/react-native-image-cache": "^3.1.0",
    "@react-native-async-storage/async-storage": "^1.21.0",
    "@react-native-community/blur": "^4.3.2",
    "@react-native-picker/picker": "^2.6.1",
    "@react-navigation/bottom-tabs": "6.5.8",
    "@react-navigation/native": "^6.1.7",
    "@react-navigation/native-stack": "^6.9.13",
    "@rnmapbox/maps": "10.1.0-beta.23",
    "axios": "^1.5.0",
    "dayjs": "^1.11.9",
    "native-base": "^3.4.28",
    "patch-package": "^8.0.0",
    "postinstall-postinstall": "^2.1.0",
    "react": "18.2.0",
    "react-native": "0.72.4",
    "react-native-3dcube-navigation": "^1.0.5",
    "react-native-animatable": "^1.3.3",
    "react-native-collapsible": "^1.6.1",
    "react-native-date-picker": "^4.3.3",
    "react-native-face-detection": "^0.1.1",
    "react-native-file-access": "^3.0.4",
    "react-native-fs": "^2.20.0",
    "react-native-gesture-handler": "^2.13.3",
    "react-native-image-picker": "^7.1.0",
    "react-native-insta-story": "^1.1.9",
    "react-native-keyboard-aware-scroll-view": "^0.9.5",
    "react-native-linear-gradient": "^2.8.3",
    "react-native-paper": "^5.11.1",
    "react-native-phone-number-input": "^2.1.0",
    "react-native-reanimated": "^3.6.2",
    "react-native-safe-area-context": "^4.7.2",
    "react-native-screens": "^3.26.0",
    "react-native-select-dropdown": "^3.4.0",
    "react-native-shadow-2": "^7.0.8",
    "react-native-simple-accordion": "^1.17.0",
    "react-native-snap-carousel": "^3.9.1",
    "react-native-status-bar-height": "^2.6.0",
    "react-native-svg": "^13.13.0",
    "react-native-swipe-list-view": "^3.2.9",
    "react-native-vector-icons": "^10.0.3",
    "react-native-view-shot": "^3.8.0",
    "react-native-vision-camera": "^3.8.2",
    "react-native-webview": "^13.6.4",
    "react-native-worklets-core": "^0.3.0",
    "turkey-yl-district": "^1.1.0"
  },
```

### VisionCamera Version

3.8.2

### Target platforms

Android

### Operating system

Windows

### Can you build the VisionCamera Example app?

Yes, I can successfully build the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-02-08:
If it builds for your teammate but does not build for you then it is an issue with your development machine.

Maybe try using a shorter path or something, I don't know what happens on Windows.

closed 2024-02-08 by mrousavy

metinaltinbas, 2024-02-10:
Having the same issue on android right now.

gh-pro, 2024-03-01:
Having same issue on android in mac m2

gh-pro, 2024-03-01:
"dependencies": {
    "@bam.tech/react-native-image-resizer": "^3.0.7",
    "@gorhom/bottom-sheet": "^4",
    "@react-native-async-storage/async-storage": "^1.22.1",
    "@react-native-community/datetimepicker": "^7.6.2",
    "@react-native-firebase/app": "^18.8.0",
    "@react-native-firebase/auth": "^18.8.0",
    "@react-native-firebase/firestore": "^18.8.0",
    "@react-native-firebase/messaging": "^18.8.0",
    "@react-native-firebase/storage": "^18.8.0",
    "@react-navigation/bottom-tabs": "^6.5.12",
    "@react-navigation/native": "^6.1.10",
    "@react-navigation/native-stack": "^6.9.18",
    "@reduxjs/toolkit": "^2.2.1",
    "axios": "^1.6.7",
    "i18next": "^23.9.0",
    "lodash": "^4.17.21",
    "moment": "^2.30.1",
    "native-base": "^3.4.28",
    "react": "18.2.0",
    "react-i18next": "^14.0.5",
    "react-native": "0.72.6",
    "react-native-device-info": "^10.12.0",
    "react-native-gesture-handler": "^2.15.0",
    "react-native-image-picker": "^7.1.0",
    "react-native-localize": "^3.0.6",
    "react-native-qrcode-svg": "^6.3.0",
    "react-native-reanimated": "^3.7.1",
    "react-native-safe-area-context": "^4.9.0",
    "react-native-screens": "^3.29.0",
    "react-native-svg": "^14.1.0",
    "react-native-vector-icons": "^10.0.3",
    "react-native-vision-camera": "^3.9.0",
    "react-redux": "^9.1.0"
  },
  "devDependencies": {
    "@babel/core": "^7.20.0",
    "@babel/preset-env": "^7.20.0",
    "@babel/runtime": "^7.20.0",
    "@react-native/eslint-config": "^0.72.2",
    "@react-native/metro-config": "^0.72.11",
    "@tsconfig/react-native": "^3.0.0",
    "@types/react": "^18.0.24",
    "@types/react-test-renderer": "^18.0.0",
    "babel-jest": "^29.2.1",
    "eslint": "^8.19.0",
    "jest": "^29.2.1",
    "metro-react-native-babel-preset": "0.76.8",
    "prettier": "^2.4.1",
    "react-test-renderer": "18.2.0",
    "typescript": "4.8.4"
  },