margelo/react-native-vision-camera #2531: Vision Camera 2, Error With React Native 0.73.4 on Android🔧

edritech93, 2024-02-09:
### How were you trying to build the app?

npm run android

### Full build logs

```tsx
Build file '/Users/yudiedrialviska/projects/nascance/upgrade/haermes3-mobile/node_modules/react-native-vision-camera/android/build.gradle' line: 354

A problem occurred evaluating project ':react-native-vision-camera'.
> Cannot invoke method replace() on null object

* Try:
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

* Exception is:
org.gradle.api.GradleScriptException: A problem occurred evaluating project ':react-native-vision-camera'.
	at org.gradle.groovy.scripts.internal.DefaultScriptRunnerFactory$ScriptRunnerImpl.run(DefaultScriptRunnerFactory.java:93)
	at org.gradle.configuration.DefaultScriptPluginFactory$ScriptPluginImpl.lambda$apply$0(DefaultScriptPluginFactory.java:135)
	at org.gradle.configuration.ProjectScriptTarget.addConfiguration(ProjectScriptTarget.java:79)
	at org.gradle.configuration.DefaultScriptPluginFactory$ScriptPluginImpl.apply(DefaultScriptPluginFactory.java:138)
	at org.gradle.configuration.BuildOperationScriptPlugin$1.run(BuildOperationScriptPlugin.java:65)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$1.execute(DefaultBuildOperationRunner.java:29)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$1.execute(DefaultBuildOperationRunner.java:26)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$2.execute(DefaultBuildOperationRunner.java:66)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$2.execute(DefaultBuildOperationRunner.java:59)
	at org.gradle.internal.operations.DefaultBuildOperationRunner.execute(DefaultBuildOperationRunner.java:157)
	at org.gradle.internal.operations.DefaultBuildOperationRunner.execute(DefaultBuildOperationRunner.java:59)
	at org.gradle.internal.operations.DefaultBuildOperationRunner.run(DefaultBuildOperationRunner.java:47)
	at org.gradle.internal.operations.DefaultBuildOperationExecutor.run(DefaultBuildOperationExecutor.java:68)
	at org.gradle.configuration.BuildOperationScriptPlugin.lambda$apply$0(BuildOperationScriptPlugin.java:62)
	at org.gradle.configuration.internal.DefaultUserCodeApplicationContext.apply(DefaultUserCodeApplicationContext.java:44)
	at org.gradle.configuration.BuildOperationScriptPlugin.apply(BuildOperationScriptPlugin.java:62)
	at org.gradle.api.internal.project.DefaultProjectStateRegistry$ProjectStateImpl.lambda$applyToMutableState$1(DefaultProjectStateRegistry.java:395)
	at org.gradle.api.internal.project.DefaultProjectStateRegistry$ProjectStateImpl.fromMutableState(DefaultProjectStateRegistry.java:413)
	at org.gradle.api.internal.project.DefaultProjectStateRegistry$ProjectStateImpl.applyToMutableState(DefaultProjectStateRegistry.java:394)
	at org.gradle.configuration.project.BuildScriptProcessor.execute(BuildScriptProcessor.java:46)
	at org.gradle.configuration.project.BuildScriptProcessor.execute(BuildScriptProcessor.java:27)
	at org.gradle.configuration.project.ConfigureActionsProjectEvaluator.evaluate(ConfigureActionsProjectEvaluator.java:35)
	at org.gradle.configuration.project.LifecycleProjectEvaluator$EvaluateProject.lambda$run$0(LifecycleProjectEvaluator.java:109)
	at org.gradle.api.internal.project.DefaultProjectStateRegistry$ProjectStateImpl.lambda$applyToMutableState$1(DefaultProjectStateRegistry.java:395)
	at org.gradle.api.internal.project.DefaultProjectStateRegistry$ProjectStateImpl.lambda$fromMutableState$2(DefaultProjectStateRegistry.java:418)
	at org.gradle.internal.work.DefaultWorkerLeaseService.withReplacedLocks(DefaultWorkerLeaseService.java:360)
	at org.gradle.api.internal.project.DefaultProjectStateRegistry$ProjectStateImpl.fromMutableState(DefaultProjectStateRegistry.java:418)
	at org.gradle.api.internal.project.DefaultProjectStateRegistry$ProjectStateImpl.applyToMutableState(DefaultProjectStateRegistry.java:394)
	at org.gradle.configuration.project.LifecycleProjectEvaluator$EvaluateProject.run(LifecycleProjectEvaluator.java:100)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$1.execute(DefaultBuildOperationRunner.java:29)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$1.execute(DefaultBuildOperationRunner.java:26)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$2.execute(DefaultBuildOperationRunner.java:66)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$2.execute(DefaultBuildOperationRunner.java:59)
	at org.gradle.internal.operations.DefaultBuildOperationRunner.execute(DefaultBuildOperationRunner.java:157)
	at org.gradle.internal.operations.DefaultBuildOperationRunner.execute(DefaultBuildOperationRunner.java:59)
	at org.gradle.internal.operations.DefaultBuildOperationRunner.run(DefaultBuildOperationRunner.java:47)
	at org.gradle.internal.operations.DefaultBuildOperationExecutor.run(DefaultBuildOperationExecutor.java:68)
	at org.gradle.configuration.project.LifecycleProjectEvaluator.evaluate(LifecycleProjectEvaluator.java:72)
	at org.gradle.api.internal.project.DefaultProject.evaluate(DefaultProject.java:788)
	at org.gradle.api.internal.project.DefaultProject.evaluate(DefaultProject.java:156)
	at org.gradle.api.internal.project.ProjectLifecycleController.lambda$ensureSelfConfigured$2(ProjectLifecycleController.java:84)
	at org.gradle.internal.model.StateTransitionController.lambda$doTransition$14(StateTransitionController.java:255)
	at org.gradle.internal.model.StateTransitionController.doTransition(StateTransitionController.java:266)
	at org.gradle.internal.model.StateTransitionController.doTransition(StateTransitionController.java:254)
	at org.gradle.internal.model.StateTransitionController.lambda$maybeTransitionIfNotCurrentlyTransitioning$10(StateTransitionController.java:199)
	at org.gradle.internal.work.DefaultSynchronizer.withLock(DefaultSynchronizer.java:34)
	at org.gradle.internal.model.StateTransitionController.maybeTransitionIfNotCurrentlyTransitioning(StateTransitionController.java:195)
	at org.gradle.api.internal.project.ProjectLifecycleController.ensureSelfConfigured(ProjectLifecycleController.java:84)
	at org.gradle.api.internal.project.DefaultProjectStateRegistry$ProjectStateImpl.ensureConfigured(DefaultProjectStateRegistry.java:369)
	at org.gradle.execution.TaskPathProjectEvaluator.configure(TaskPathProjectEvaluator.java:33)
	at org.gradle.execution.TaskPathProjectEvaluator.configureHierarchy(TaskPathProjectEvaluator.java:49)
[... 167 lines cut ...]
```

### VisionCamera Version

2.16.8

### Target platforms

Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

pnogier, 2024-02-13:
Same issue here
React Native `0.73.2`
React Native Vision Camera `2.16.6` (Need if for plugins available only on v2)

```
* Where:
Build file '/Users/pnogier/xxxx/xxxx/xxxx/node_modules/react-native-vision-camera/android/build.gradle' line: 354

* What went wrong:
A problem occurred evaluating project ':react-native-vision-camera'.
> Cannot invoke method replace() on null object
```

Where you able to fix or patch this ?

mikebouwmans, 2024-02-14:
Seems like this is only an issue when frame processors are enabled. I don't use them, so I added ```disableFrameProcessors=true```  to my  ```gradle.properties``` file.

nhatnguyen95, 2024-02-19:
I have same issue on `react-native 0.73.4` and `react-native-vision-camera: 2.16.8`. I can't disableFrameProcessors because I am using qr code and face detector

boiboif, 2024-02-19:
same issue

tasmto, 2024-03-04:
Thank you @mikebouwmans you are a life saver!
Just to add onto your answer for people using expo prebuild; you may need to create a custom expo plugin ([see this StackOverflow answer](https://stackoverflow.com/questions/72748846/modify-android-gradle-properties-in-expo-managed-app)) to make sure this property gets added when your gradle.properties file gets generated.

Here is what mine looks like:
 `const {withGradleProperties} = require('@expo/config-plugins');

// This is the plugin that will modify the gradle.properties file
module.exports = config => {
  const newGraddleProperties = [
    // To get React Native Vision Camera V2 working with expo prebuild.
    {
      type: 'property',
      key: 'disableFrameProcessors',
      value: true,
    },
  ];

  return withGradleProperties(config, config => {
    newGraddleProperties.map(gradleProperty =>
      config.modResults.push(gradleProperty),
    );
    return config;
  });
};`

felginep, 2024-03-21:
Any updates on this ? I have the same issue and can't disable frame processors...

AndyRC6, 2024-03-29:
Bump. Having the same issue and am needing frame processors.

pnogier, 2024-03-30:
Hi there, 
For everyone still having issues with this, I finally ended up going with `expo-camer`, `expo-barcode-scanner` and `expo-face-detector`.
It suits well for my use case, so I would recommend it to everyone needing a face detector and a QR code scanner.

shan007g, 2024-04-03:
I found that the frameprocessors are dependent on **react-native-reanimated** & I didn't use this lib so I removed it and it worked for me 
I'm able to use vision camera v2 on react native 0.73 with **react-native-reanimated** removed

ruanitto, 2024-04-17:
> ### How were you trying to build the app?
> npm run android
> 
> ### Full build logs
> ```tsx
> Build file '/Users/yudiedrialviska/projects/nascance/upgrade/haermes3-mobile/node_modules/react-native-vision-camera/android/build.gradle' line: 354
> 

The error occurs due to the react-native version not having the BOOST_VERSION variable in gradle.properties (in **_node_modules/react-native/ReactAndroid/gradle.properties_**), check the version of the react-native package used.

In my case, I changed line 337 in _**node_modules/react-native-vision-camera/android/build.gradle**_

for:

```gradle 
   // line 337
   def BOOST_VERSION = "1_85_0" //thirdPartyVersions["BOOST_VERSION"]
```

I suggest after the change, create a patch for the file.

  def BOOST_VERSION = "1_85_0" //thirdPartyVersions["BOOST_VERSION"]

mrousavy, 2024-04-22:
Hey - please upgrade to VisionCamera V4. V2 is no longer maintained.

closed 2024-04-22 by mrousavy

boiboif, 2024-04-23:
> Hey - please upgrade to VisionCamera V4. V2 is no longer maintained.

Hello, Will VisionCamera V4 still experience preview stretching issues in landscape mode?

Jamal-ReachFirst, 2024-07-26:
> ```groovy-gradle
> def BOOST_VERSION = "1_85_0"
> ```

Fixed my issue