margelo/react-native-vision-camera #2532: 🐛 when switching from back to front camera freezes, when trying to take pic the phone reboots itself

efstathiosntonas, 2024-02-09:
### What's happening?

When trying to change between front -> back camera the camera view is flipping and then is freezing, when I tap the button to take the picture the device reboots it self.

Tried on Samsung S22 Android 14 and Xiaomi Note 8 on Android 12, on Xiaomi works as a charm. I downgraded to 3.8.2 and the issue is still there. 

react native@0.73.4, reanimated@3.7.0, gesture handler@2.15.0

On example app, it won't even flip the camera, it will crash

ps. had to truncate the logs by a lot, it exceeded the 65K limit of github comment.

### Reproduceable Code

```tsx
camera: 

import React, { ReactElement, useCallback, useEffect, useRef, useState } from "react";
import {
  GestureResponderEvent,
  Platform,
  StatusBar,
  StyleSheet,
  View
} from "react-native";
import { useIsFocused } from "@react-navigation/native";
import {
  PinchGestureHandler,
  PinchGestureHandlerGestureEvent,
  TapGestureHandler
} from "react-native-gesture-handler";
import Reanimated, {
  Extrapolate,
  interpolate,
  useAnimatedGestureHandler,
  useAnimatedProps,
  useSharedValue
} from "react-native-reanimated";
import {
  Camera as RNCamera,
  CameraRuntimeError,
  PhotoFile,
  useCameraDevice,
  useCameraFormat,
  VideoFile
} from "react-native-vision-camera";
import FontAwesome6Pro from "react-native-vector-icons/FontAwesome6Pro";
import mime from "mime";
import { openSettings } from "react-native-permissions";
import { PressableOpacity } from "react-native-pressable-opacity";
import { setRootViewBackgroundColor } from "@pnthach95/react-native-root-view-background";
import { useStyles } from "react-native-unistyles";
import { useTranslation } from "react-i18next";

import { CaptureButton } from "./views/CaptureButton";
import { CONTENT_SPACING, MAX_ZOOM_FACTOR, SAFE_AREA_PADDING } from "./Constants";
import { dimensions, notify, showErrorAlert, sleep } from "@utils/utils";
import { fullScreenAlert } from "@components/FullScreenAlert";
import { StatusBarBlurBackground } from "./views/StatusBarBlurBackground";
import { useIsForeground } from "@hooks/useIsForeground";
import { checkCameraPermission } from "@functions/image";

const ReanimatedCamera = Reanimated.createAnimatedComponent(RNCamera);
Reanimated.addWhitelistedNativeProps({
  zoom: true
});

const SCALE_FULL_ZOOM = 3;
const BUTTON_SIZE = 50;

[... 952 lines cut ...]
  "hasTorch": false,
  "position": "front",
  "id": "1"
}
```

### Device

Samsung S22 Android 14

### VisionCamera Version

3.9.0-beta.3 and 3.8.2

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

efstathiosntonas, 2024-02-09:
when I removed the `fps` prop the issue went away.

mrousavy, 2024-02-13:
Thanks for the bug report, but the reproduceable code is way too much. Can you please try to reproduce it in the example app instead?

efstathiosntonas, 2024-02-13:
hi, it's happening on example too, I've tested it against Samsung S22 Android 14.

mrousavy, 2024-02-13:
Okay yea I just read your comment about `fps` - I know about this issue, and I honestly don't know what to do here. Samsung is saying their phones can do 60 FPS in the API, but when you try to use 60 FPS it crashes.
They are messing it up here, there's nothing I can do except explicitly disabling 60 FPS on Samsung devices.

Just to clarify; does it also crash if just set `fps={30}`, or only something higher than 30?

efstathiosntonas, 2024-02-13:
thanks for the heads up, will check with 30 in a while but it should be the default if the prop isn't provided, right?

mrousavy, 2024-02-13:
Technically not, `null` means it won't set `AE_EXPOSURE_DURATION`, whereas `30` means it will set it to `30`.

efstathiosntonas, 2024-02-13:
`fps={30}` won't crash it.

mrousavy, 2024-02-13:
Yea okay, then it definitely is the same issue that I am already aware of. It's so lame that Samsung simply lies in [`CONTROL_AE_AVAILABLE_TARGET_FPS_RANGES`](https://developer.android.com/reference/android/hardware/camera2/CameraCharacteristics#CONTROL_AE_AVAILABLE_TARGET_FPS_RANGES) - it contains 60 FPS, but doesn't actually support 60 FPS.

efstathiosntonas, 2024-02-13:
I wished they were lame only on that part lol!

I think the issue title should be ...in front camera

mrousavy, 2024-02-13:
Just to double check if I am understanding this correctly - all values listed here are supported, right? Or only values here, that are also in `CamcorderProfile.videoFrameRate`?

[image]

mrousavy, 2024-02-13:
Can you try with a really low res format and see if that works at 60 FPS?

```ts
const format = useCameraFormat(device, [
  { videoResolution: { width: 100, height: 100 }
])
```

efstathiosntonas, 2024-02-13:
on back camera it throwed this: 

`fps={60}`

```
 ERROR  [format/invalid-fps: [format/invalid-fps] The given format cannot run at 60 FPS! Make sure your FPS is lower than `format.maxFps` but higher than `format.minFps`.]
```

when switching to front no error but the screen is black, tried setting format to 400x400, same thing.

efstathiosntonas, 2024-02-14:
it wont crash after setting fps to 60. Thanks Marc, keep it up!

closed 2024-02-14 by efstathiosntonas

thegreatzeus, 2024-04-03:
@mrousavy I can confirm this issue still exists. 
I have tested v3.9.0-beta.4 and the latest release v3.9.2 on Galaxy S23 Ultra and Galaxy S22 both running Android 14.
60 FPS is available on both devices' front and back cameras, but VisionCamera is limited to 30 FPS.

Example code:
```jsx
import { View, Text } from "react-native";
import { useEffect } from 'react';
import { useCameraFormat, useCameraDevice } from "react-native-vision-camera";

export default function App() {
  const device = useCameraDevice("front");
  const format = useCameraFormat(device, [
    { fps: 60 }
  ]);

  useEffect(() => {
    console.log(format)
  }, [format]);

  return (
    <View>
      <Text>Test</Text>
    </View>
  );
}
```

Outputs:
```json
{
  "autoFocusSystem": "contrast-detection",
  "fieldOfView": 80.25568970571118,
  "maxFps": 30,
  "maxISO": 3200,
  "maxZoom": 8,
  "minFps": 1,
  "minISO": 50,
  "photoHeight": 2736,
  "photoWidth": 3648,
  "pixelFormats": [
    "yuv",
    "native"
  ],
  "supportsDepthCapture": false,
  "supportsPhotoHdr": false,
  "supportsVideoHdr": false,
  "videoHeight": 1440,
  "videoStabilizationModes": [
    "off",
    "off"
  ],
  "videoWidth": 1920
}

```

mrousavy, 2024-04-03:
Again, this is because of Android.
Not because of VisionCamera.
Samsung locks 60 FPS for only their own app.

laxminarayana-capgemini, 2024-06-22:
Hi @mrousavy, Is there any solutions for this. I trying to run in several Android mobiles. Most of the mobiles are giving maxFps as 60 only. Please can you suggest any solutions?
Thanks.

mrousavy, 2024-06-24:
Dude did you read my comment directly above your comment?

Dat-Mobile, 2025-01-14:
Sadly android device limited the 60fps, for now I can only test 60fps on iphone