margelo/react-native-vision-camera #2533: 🐛 ANDROID: 3.9.0-beta.3 stop recording with flash -  app shutdown

MilanRadon, 2024-02-09:
### What's happening?

[image]
[image]

### Reproduceable Code

```tsx
demo app
```

### Relevant log output

```shell
nothing
```

### Camera Device

```json
nothing
```

### Device

android 14

### VisionCamera Version

3.9.0-beta.3 

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

MilanRadon, 2024-02-09:
maybe is problem on ios..but i dont know..

mrousavy, 2024-02-13:
[image]

Author did not post required information, closing issue.

closed 2024-02-13 by mrousavy

mrousavy, 2024-02-13:
Created a PR for this anyways sicne I know how to fix it: https://github.com/mrousavy/react-native-vision-camera/pull/2548