margelo/react-native-vision-camera #2535: 🐛Cannot convert lass com.google.mlkit.vision.face.Face to jsi::Value

eladattias1996, 2024-02-10:
### What's happening?

I tried to write a face detection frame processor plugin and I currently focus on Android using MLKit Vision, I'm able to get faces from MLkit (I log them in the Logcat successfully) but when I return them over the bridge I get the following error: `Frame Processor Error: Exception in HostFunction: Received unknown JNI type "class com.google.mlkit.vision.face.Face"! Cannot convert to jsi::Value., js engine: VisionCamera`.

According to the docs: _Return values will automatically be converted to JS values, assuming they are representable in the ["Types" table](https://react-native-vision-camera.com/docs/guides/frame-processors-plugins-overview#types). So the following Java Frame Processor Plugin_ so I guess there is a bug there for some reason.

### Reproduceable Code

```tsx
The android code:

class FaceDetector {
    companion object {
        private val TAG: String = "FaceDetector"
        private val FACE_DETECTION_OPTIONS = FaceDetectorOptions.Builder().setLandmarkMode(FaceDetectorOptions.LANDMARK_MODE_ALL).build()

        fun detectFaces(image: InputImage): Task<List<Face>> {
            Log.d(TAG, "in detect faces")

            val detector = FaceDetection.getClient(FACE_DETECTION_OPTIONS)
            val taskCompletionSource = TaskCompletionSource<List<Face>>()

            Log.d(TAG, "after getting detector $detector")

            detector.process(image)
                .addOnSuccessListener { faces ->
                    faces.forEach { face -> Log.d(TAG, "face object: $face") }
                    taskCompletionSource.setResult(faces)
                }
                .addOnFailureListener { error -> taskCompletionSource.setException(error) }
                .addOnCompleteListener { Log.d(TAG, "done process") }

            return taskCompletionSource.task
        }
    }
}
```
```
class FaceDetectorFrameProcessorPlugin(proxy: VisionCameraProxy, options: Map<String, Any>?): FrameProcessorPlugin() {
  companion object {
    private const val TAG = "FaceDetectorFrameProcessorPlugin"
  }

  override fun callback(frame: Frame, arguments: Map<String, Any>?): Any? {
    return try {
      Tasks.await(FaceDetector.detectFaces(getInputImageFromFrame(frame)))
    } catch (e: Exception) {
      null
    }
  }

  private fun getInputImageFromFrame(frame: Frame): InputImage {
    val image = frame.image
    val rotationDegrees = 0 // TODO: calculate it later

    return InputImage.fromMediaImage(image, rotationDegrees)
  }
}
```

The react native code:
```
const plugin = VisionCameraProxy.initFrameProcessorPlugin('detectFaces');

// TODO: put type here!!!
export function detectFaces(frame: Frame) {
  'worklet'
  if (plugin == null) {
    throw new Error("Failed to load Frame Processor Plugin!");
  }
  return plugin.call(frame);
}
```
```
export const CameraScreen: FC = () => {
  const device = useCameraDevice('front');

  const frameProcessor = useFrameProcessor(frame => {
    'worklet';

    const result = detectFaces(frame);

    console.log('the result is', result);
  }, []);

  if (!device) return <DeviceNotFound />;

  return (
    <Camera
      isActive
      style={styles.fullScreen}
      device={device}
      frameProcessor={frameProcessor}
      pixelFormat="yuv"
    />
  );
};

export const styles = StyleSheet.create({
  fullScreen: {
    flex: 1,
  },
});
```
In the `babel.config.js`:
```
module.exports = {
  presets: ['module:@react-native/babel-preset'],
  plugins: ['react-native-worklets-core/plugin']
};
```
```

### Relevant log output

```shell
Android:
2024-02-10 09:51:29.079 30315-30419 FaceDetect...ssorPlugin com.cameratry                        D  Got face from the model - Face{boundingBox=Rect(685, 245 - 1113, 680), trackingId=-1, rightEyeOpenProbability=-1.0, leftEyeOpenProbability=-1.0, smileProbability=-1.0, eulerX=3.0604274, eulerY=83.78829, eulerZ=88.48733, landmarks=Landmarks{landmark_0=FaceLandmark{type=0, position=PointF(1014.8454, 392.491)}, landmark_1=FaceLandmark{type=1, position=PointF(934.7279, 444.18445)}, landmark_3=FaceLandmark{type=3, position=PointF(855.9822, 597.48425)}, landmark_4=FaceLandmark{type=4, position=PointF(862.48254, 425.0186)}, landmark_5=FaceLandmark{type=5, position=PointF(971.47156, 416.07233)}, landmark_6=FaceLandmark{type=6, position=PointF(940.79266, 377.3722)}, landmark_7=FaceLandmark{type=7, position=PointF(939.3011, 409.4529)}, landmark_9=FaceLandmark{type=9, position=PointF(901.4224, 576.6362)}, landmark_10=FaceLandmark{type=10, position=PointF(859.82404, 412.73965)}, landmark_11=FaceLandmark{type=11, position=PointF(990.2341, 424.16638)}}, contours=Contours{Contour_1=null, Contour_2=null, Contour_3=null, Contour_4=null, Contour_5=null, Contour_6=null, Contour_7=null, Contour_8=null, Contour_9=null, Contour_10=null, Contour_11=null, Contour_12=null, Contour_13=null, Contour_14=null, Contour_15=null}}

React Native:
Frame Processor Error: Exception in HostFunction: Received unknown JNI type "class com.google.mlkit.vision.face.Face"! Cannot convert to jsi::Value., js engine: VisionCamera
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "full",
  "maxZoom": 8,
  "minZoom": 1,
  "maxExposure": 20,
  "supportsLowLightBoost": true,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": false,
  "minFocusDistance": 10,
  "minExposure": -20,
  "name": "FRONT (1)",
  "hasFlash": false,
  "hasTorch": false,
  "position": "front",
  "id": "1"
}
```

### Device

Samsung S22 (Android 12)

### VisionCamera Version

3.8.2

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-02-13:
The error message say

```
Received unknown JNI type "class com.google.mlkit.vision.face.Face"! Cannot convert to jsi::Value.
```

Which tells you that a `Face` cannot automatically be converted to a JSI Value.

The docs say:

> Return values will automatically be converted to JS values, assuming they are representable in the ["Types" table](https://react-native-vision-camera.com/docs/guides/frame-processors-plugins-overview#types).

Which is also pretty self explanatory in my opinion as the "Types" table tells you exactly which types you can use.

[image]

There is no mention of a `WritableArray` here. If you want to return an array, look into the 4th row of the "Types" table which tells you to use `List<Object>`.

In my opinion I made those docs very self explanatory, but if you think they aren't clear enough please open a PR to make the docs even clearer.

closed 2024-02-13 by mrousavy

mrousavy, 2024-02-13:
haha no worries